aws/amazon-freertos #2934: Cmake error because IDF_PATH and IDF_PYTHON_ENV_PATH

monhack, 2021-01-13:
Hi @all, 

how do you solve this problem?
[image]
I have tried exporting with the appropriate path but it still doesn't work

shubhamkulkarni97, 2021-01-13:
Hi @monhack,

Python requirements for ESP-IDF are not satisfied on your machine, this is the main cause of build failure. There is no need to set IDF_PATH and IDF_PYTHON_ENV_PATH.

Please follow these steps to install python requirements:
* `cd vendors/espressif/esp-idf`
* `./install.sh`
* `. ./export.sh`
* `cd <amazon-freertos-root>`
* `unset IDF_PATH`

Building application after running these commands should work.

Thanks,
Shubham

monhack, 2021-01-13:
Thanks, @shubhamkulkarni97 

I have done step by step but it doesn't work either, the error is still the same

```
sudo cmake -DVENDOR=espressif -DBOARD=esp32_devkitc -DCOMPILER=xtensa-esp32 -DCMAKE_TOOLCHAIN_FILE='/home/monhack/Documents/Magang/freertos/control/ESP32-GG-6/tools/cmake/toolchains/xtensa-esp32.cmake' -DCMAKE_C_COMPILER='~/esp/xtensa-esp32-elf/bin/xtensa-esp32-elf-gcc' -DCMAKE_CXX_COMPILER='~/esp/xtensa-esp32-elf/bin/xtensa-esp32-elf-g++' -DCMAKE_ASM_COMPILER='~/esp/xtensa-esp32-elf/bin/xtensa-esp32-elf-gcc' -DAFR_COMPILER_CC='~/esp/xtensa-esp32-elf/bin/xtensa-esp32-elf-gcc' -DAFR_COMPILER_CXX='~/esp/xtensa-esp32-elf/bin/xtensa-esp32-elf-g++' -DCMAKE_MAKE_PROGRAM='/home/monhack/.espressif/tools/ninja' -DAFR_BOARD='espressif.esp32_devkitc' -S . -B esp32-GG-6
[sudo] password for monhack: 
-- Submodule update
Skipping submodule '../../../libraries/abstractions/backoff_algorithm/'
Skipping submodule '../../../../libraries/abstractions/pkcs11/corePKCS11/'
Skipping submodule '../../../libraries/abstractions/pkcs11/corePKCS11/'
Skipping submodule '../../../libraries/abstractions/pkcs11/corePKCS11/'
Skipping submodule '../../../libraries/abstractions/pkcs11/corePKCS11/'
Skipping submodule '../../../libraries/coreHTTP/'
Skipping submodule '../../../libraries/coreHTTP/'
Skipping submodule '../../../libraries/coreHTTP/'
Skipping submodule '../../../libraries/coreJSON/'
Skipping submodule '../../../libraries/coreJSON/'
Skipping submodule '../../../libraries/coreJSON/'
Skipping submodule '../../../libraries/coreMQTT/'
Skipping submodule '../../../libraries/coreMQTT/'
Skipping submodule '../../../libraries/coreMQTT/'
Skipping submodule '../../../libraries/device_defender_for_aws/'
Skipping submodule '../../../libraries/device_defender_for_aws/'
Skipping submodule '../../../libraries/device_defender_for_aws/'
Skipping submodule '../../../libraries/device_shadow_for_aws/'
Skipping submodule '../../../libraries/device_shadow_for_aws/'
Skipping submodule '../../../libraries/device_shadow_for_aws/'
Skipping submodule '../../libraries/freertos_plus/standard/freertos_plus_tcp/'
Skipping submodule '../../libraries/freertos_plus/standard/freertos_plus_tcp/'
Skipping submodule '../../libraries/freertos_plus/standard/freertos_plus_tcp/'
Skipping submodule '../../../libraries/jobs_for_aws/'
Skipping submodule '../../../libraries/jobs_for_aws/'
Skipping submodule '../../../libraries/jobs_for_aws/'
CMake Warning at vendors/espressif/esp-idf/tools/cmake/idf.cmake:24 (message):
  IDF_PATH environment variable not found.  Setting IDF_PATH to
  '/home/monhack/Documents/Magang/freertos/control/ESP32-GG-6/vendors/espressif/esp-idf'.
Call Stack (most recent call first):
  vendors/espressif/boards/esp32/CMakeLists.txt:38 (include)
  CMakeLists.txt:72 (include)

-- Component directory /home/monhack/Documents/Magang/freertos/control/ESP32-GG-6/vendors/espressif/esp-idf/components/mbedtls does not contain a CMakeLists.txt file. No component will be added
-- Component directory /home/monhack/Documents/Magang/freertos/control/ESP32-GG-6/vendors/espressif/esp-idf/components/unity does not contain a CMakeLists.txt file. No component will be added
-- Checking Python dependencies...
The following Python requirements are not satisfied:
gdbgui==0.13.2.0
pygdbmi<=0.9.0.2
reedsolo>=1.5.3,<=1.5.4
bitstring>=3.1.6
ecdsa>=0.16.0
Please follow the instructions found in the "Set up the tools" section of ESP-IDF Getting Started Guide
Diagnostic information:
    IDF_PYTHON_ENV_PATH: (not set)
    Python interpreter used: /usr/bin/python
    Warning: python interpreter not running from IDF_PYTHON_ENV_PATH
    PATH: /usr/local/sbin:/usr/local/bin:/usr/sbin:/usr/bin:/sbin:/bin:/snap/bin
CMake Error at vendors/espressif/esp-idf/tools/cmake/build.cmake:271 (message):
  Some Python dependencies must be installed.  Check above message for
  details.
Call Stack (most recent call first):
  vendors/espressif/esp-idf/tools/cmake/build.cmake:397 (__build_check_python)
  vendors/espressif/boards/esp32/CMakeLists.txt:542 (idf_build_process)
  CMakeLists.txt:72 (include)

-- Configuring incomplete, errors occurred!
See also "/home/monhack/Documents/Magang/freertos/control/ESP32-GG-6/esp32-GG-6/CMakeFiles/CMakeOutput.log".
See also "/home/monhack/Documents/Magang/freertos/control/ESP32-GG-6/esp32-GG-6/CMakeFiles/CMakeError.log".
```

whether the setting must be superuser (root)?

romaindequidt, 2021-01-13:
> cd vendors/espressif/esp-idf
> pip install -r requirements.txt

monhack, 2021-01-18:
Thanks, @romaindequidt

I have tried the steps you gave, but it still gets an error

```
root@f-f:/home/monhack/Documents/Magang/freertos/control/ESP32-GG-6/vendors/espressif/esp-idf# cmake -DVENDOR=espressif -DBOARD=esp32_devkitc -DCOMPILER=xtensa-esp32 -S . -B esp32-GG-8
-- The C compiler identification is GNU 7.5.0
-- The CXX compiler identification is GNU 7.5.0
-- The ASM compiler identification is GNU
-- Found assembler: /usr/bin/cc
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Check for working C compiler: /usr/bin/cc - skipped
-- Detecting C compile features
-- Detecting C compile features - done
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Check for working CXX compiler: /usr/bin/c++ - skipped
-- Detecting CXX compile features
-- Detecting CXX compile features - done
CMake Error at CMakeLists.txt:5 (message):
  Current directory
  '/home/monhack/Documents/Magang/freertos/control/ESP32-GG-6/vendors/espressif/esp-idf'
  is not buildable.  Change directories to one of the example projects in
  '/home/monhack/Documents/Magang/freertos/control/ESP32-GG-6/vendors/espressif/esp-idf/examples'
  and try again.

-- Configuring incomplete, errors occurred!
See also "/home/monhack/Documents/Magang/freertos/control/ESP32-GG-6/vendors/espressif/esp-idf/esp32-GG-8/CMakeFiles/CMakeOutput.log".
```

shubhamkulkarni97, 2021-01-19:
@monhack Please check build, flash and run instructions in [getting started guide](https://docs.aws.amazon.com/freertos/latest/userguide/getting_started_espressif.html#build-and-run-example-espressif).

aggarw13, 2021-01-25:
@monhack, were you able to resolve the build command issue?
From your logs, you need to run the `cmake -DVENDOR=espressif ...` command from the root directory of the repository.

monhack, 2021-01-29:
oke thanks @shubhamkulkarni97 @aggarw13

closed 2021-01-29 by monhack