aws/amazon-freertos #2936: OTA update on ESP fails

kumarfirmware, 2021-01-14:
Hi Team,

I am using AWS FreeRTOS version 202007 and ESP32.

I am facing an issue in OTA Update.

If  the  parameter "otaconfigMAX_NUM_BLOCKS_REQUEST"  is set to 8, means for every data request(prvRequestDataHandler), there is a response of 8 packets is generated.

How are the 8 packets received? Are all the 8 packets received on the device from cloud at once and sent for process one by one?  what is the timing between each packet? Is there any acknowledgment sent back to the cloud after each packet? Can we control the timing between these 8 packets?

After receiving the first packet, I need to process the data of the first packet received, then I need to receive the second packet and then process the same. How is the timing between these packets controlled? Is there a way where I can notify first packet is processed, then receive the second packet.

Can you please explain this process in detail? How is the synchronization achieved?

Also how should I notify to receive next set of 8 packets as soon as possible after I receive first set of 8 packets? 

Thanks.

pvyawaha, 2021-01-15:
Hello, 

The OTA agent when configured to request more than 1 file block for example 8 then the agent will wait for the timeout as configured in [otaconfigFILE_REQUEST_WAIT_MS](https://github.com/aws/amazon-freertos/blob/master/vendors/espressif/boards/esp32/aws_demos/config_files/aws_ota_agent_config.h#L57 ). This timeout is used only when the number of blocks requested are not received otherwise the next set of blocks are requested without waiting for this timeout to expire.

When the packets are received they are queued and processed in order as they were inserted in the queue i.e written to the flash area which is in OTA pal of the platform. [ESP OTA Port](https://github.com/aws/amazon-freertos/tree/master/vendors/espressif/boards/ports/ota)

The otaconfigFILE_REQUEST_WAIT_MS must be set according to the rate at which data is received and processed by the platform. So when requesting 8 blocks of 4 KB i.e 32KB of data the time must be sufficient to receive the data from service and write it to the flash. The maximum data that can be requested is 128 KB per request so the number of data blocks can be set to 32 but the otaconfigFILE_REQUEST_WAIT_M must be sufficient to receive 128 KB and process it. 

Are you facing issue with default configuration or after changing it ? Can you please explain the issue and provide logs?

kumarfirmware, 2021-01-17:
Hi Team,

Thanks for the message. Thanks for your immediate response.

Still I did not understand the design,

1. can we control the timing between the 8 packets received from the cloud in the above mentioned case?

2. Because as soon as I receive first packet and process it, second packet comes very fast. Is there a way I can notify to the service once I process first packet only , second packet can be received from the cloud? Please explain this in detail

3. But if I make this otaconfigMAX_NUM_BLOCKS_REQUEST to 1, it sends request for every packet and the response takes couple of seconds and to receive 30 packets it takes close to 120 seconds, which is a lot and causes performance problem.

4. kindly suggest me how can I improve the performance. to receive a job(with 30 packets of size 4096 bytes) it should not take more than 30 seconds but currently it takes 90 to 120 seconds which is killing performance.

Please suggest best possible fast solution.

Thanks,
Kumar.

kumarfirmware, 2021-01-19:
Hi pvyawaha,

Hope you had an excellent weekend.

Can you please help me in this as soon as possible.

Thanks,
Kumar.

pvyawaha, 2021-01-19:
Hello,

> 1. can we control the timing between the 8 packets received from the cloud in the above mentioned case?

When OTA library is configured to request 8 file blocks , the service sends 8 file blocks to the device. 

> 2.Because as soon as I receive first packet and process it, second packet comes very fast. Is there a way I can notify to the service once I process first packet only , second packet can be received from the cloud? Please explain this in detail  

The way to control the flow of packets from the service is using the  [otaconfigMAX_NUM_BLOCKS_REQUEST ](https://github.com/aws/amazon-freertos/blob/master/vendors/espressif/boards/esp32/aws_demos/config_files/aws_ota_agent_config.h#L106). When the file blocks are received they are decoded and written to the flash. If this process is slow then packets can be dropped. In such case to avoid dropping packets the number of buffers [otaconfigMAX_NUM_OTA_DATA_BUFFERS](https://github.com/aws/amazon-freertos/blob/master/vendors/espressif/boards/esp32/aws_demos/config_files/aws_ota_agent_config.h#L12) must higher or atleaset equal to [otaconfigMAX_NUM_BLOCKS_REQUEST ](https://github.com/aws/amazon-freertos/blob/master/vendors/espressif/boards/esp32/aws_demos/config_files/aws_ota_agent_config.h#L106). 

> 3. But if I make this otaconfigMAX_NUM_BLOCKS_REQUEST to 1, it sends request for every packet and the response takes couple of seconds and to receive 30 packets it takes close to 120 seconds, which is a lot and causes performance problem.

This is correct if `otaconfigMAX_NUM_BLOCKS_REQUEST ` is set to 1 then device will send a request for next block and will be slower than requesting multiple blocks. The speed will also be depending on the network and region.

> 4. kindly suggest me how can I improve the performance. to receive a job(with 30 packets of size 4096 bytes) it should not take more than 30 seconds but currently it takes 90 to 120 seconds which is killing performance.

As recommended in the 2. above if the number of buffers can be increased it will help in speeding up the process.

kumarfirmware, 2021-01-19:
Hi pvyawaha,

Thanks for your immediate response. can you please confirm my understanding through the following as soon as possible.

1. When OTA library is configured to request 8 file blocks , the service sends 8 file blocks to the device. Time between each file block  with in the 8 files blocks is not configurable and not decided by the device.

2. As and when low level layers, receives one block from the cloud, it pushes to the upper layer or application layer for writing in to the flash or for processing. or all the 8 blocks are received by the low level layer at once from the cloud and pushed to the application layer or higher layer. I am not understanding where is the queuing happening here? how is the queuing happening here in between the 8 blocks.

3. I guess when ever the new block receives the old block over writes it, if the processing is not fast. Can you explain this architecture in detail?

4. If I understand correctly, if I request 8 blocks at once and if I allocate 8 buffers, so there is no chance of packets getting dropped, because packets are queued and proper processing can be done. **But how I can communicate to the queue, I received one packet and I processed one packet, send me the next packet, how should I do this?**

Because if spacing between each of these 8 packets is 150 milli seconds, but my processing of the packet takes 250 milli seconds, how can I communicate this to the queue? because I should not loose packet and at the same time I should process the packet.

5. After the 8 packets received, next request will be sent to fetch next 8 packets after otaconfigFILE_REQUEST_WAIT_MS or immediately?

6.Based on your experience, can you please tell me if the device is having good network and device and cloud are in the same region, what is the time it takes to get a response for a request in the Job?

7. I have 30 packets for OTA in the job of size 4096 bytes, if I fetch 8 packets at once with 8 buffers allocated and 4 requests sent to finish 30 packets download, can this finish in 10 seconds? Is it practically possible?

8. Are the jobs through MQTT are fast? Are the jobs through HTTP are fast? In general performance of jobs is very less because we need to request and get the packets each time?

9. If you say Jobs through HTTP is fast, can I have an MQTT connection to push messages to the cloud and at the same time OTA through HTTP possible or not?

Please let me know these things as soon as possible, so that I can test and report the performance.

Thanks,
Kumar.

kumarfirmware, 2021-01-21:
Hi pvyawaha,

Can you please help me with the answers of the above as soon as possible.

Thanks,
Kumar.

pvyawaha, 2021-01-22:
Hello,

> 1.When OTA library is configured to request 8 file blocks , the service sends 8 file blocks to the device. Time between each file block with in the 8 files blocks is not configurable and not decided by the device.

That is correct, for request of 8 blocks of 4KB each it is requesting 32KB of data which service will send.

> 2. As and when low level layers, receives one block from the cloud, it pushes to the upper layer or application layer for writing in to the flash or for processing. or all the 8 blocks are received [...]

The OTA library waits for the file block received event in a queue which is queued by the mqtt callback executing in the taskpool context. The data is read by netrecv task from the network stack which is processed by the taskpool to generate the mqtt callback for a file block. 

> 3. I guess when ever the new block receives the old block over writes it, if the processing is not fast. Can you explain this architecture in detail?

When a mqtt callback is generated for the file block the data is copied in OTA buffer and event is queued  in the OTA processing queue. The OTA library checks if the data is processed ( written to flash ) before the buffers are used again otherwise it drops the packet.

> 4. If I understand correctly, if I request 8 blocks at once and if I allocate 8 buffers, so there is no chance of packets getting dropped, because packets are queued and proper processing can be [...]

The file blocks are queued by the mqtt callback so what you can do is increase the priority of the OTA Agent in the OTA Agent config [OTA Agent Priority Config](https://github.com/aws/amazon-freertos/blob/master/vendors/espressif/boards/esp32/aws_demos/config_files/aws_ota_agent_config.h#L62) to 6. This way each block should be processed first before next is queued. 

> 5. After the 8 packets received, next request will be sent to fetch next 8 packets after otaconfigFILE_REQUEST_WAIT_MS or immediately?

Sent right after the last block from previous request is received.

> 6, 7, 8, 9 

Regarding HTTP the file download is done from S3 using the presigned url. So the performance will be different. We will try to run tests using the same configuration and release you mentioned to get some metrics. It will depend on other factors like I mentioned earlier.

Meanwhile please try to run test using the priority change I mention above in point 4.

kumarfirmware, 2021-01-25:
Hi pvyawaha,

Hope you had an excellent weekend. Thanks for the above detailed explanation.

After our discussion, I started testing. The following are my results.

Setup 1:- ESP connects to AWS by using stable Wi-Fi connection.

Time difference between request and response of OTA MQTT packets is 300 milli seconds. Each packet size is 4096 bytes. This seems to be reasonable. If you are seeing different timing you can let me know.

Setup 2:- ESP connects to AWS by using Mobile App by using MQTT over BLE.

1. Time difference between request and response of OTA MQTT packets is 3 to 5 seconds. This is worst performance. Each packet size is 4096 bytes.

2.  I reduced the packet size to 2048 bytes, then the time difference between request and response of OTA MQTT packets is 1 to 2 seconds. 

3. I further reduced the packet size to 1024 bytes, then the time difference between request and response of OTA MQTT packets is half a second.

4. Based on this pattern, I could clearly understand this could be problem with the BLE connection parameters between ESP and Mobile App.

5.I want to return to 4096 bytes and time difference between request and response of OTA MQTT packets to less than half a second.

6. So I want to update the BLE connection parameters on ESP32, to solve this problem. Do you also think this is the root cause of the problem? Can you please provide your thoughts on the same and suggest a way to move forward.

7. what is the function used to update BLE connection interval in ESP32? The following function seems to be empty and not implemented?

BTStatus_t IotBle_ConnParameterUpdateRequest( const BTBdaddr_t * pBdAddr,
                                              IotBleConnectionParam_t * pConnectionParam );

8.  what is the function used to update BLE connection interval in iOS SDK?  I do not see any function in iOS sdk which will allow me to set the connection interval in AWS FreeRTOS.

9. I made the following changes but no use. 
     //#define IOT_BLE_ADVERTISING_CONN_INTERVAL_MIN    0x20
    //#define IOT_BLE_ADVERTISING_CONN_INTERVAL_MAX    0x40
    #define IOT_BLE_ADVERTISING_CONN_INTERVAL_MIN    0x20
    #define IOT_BLE_ADVERTISING_CONN_INTERVAL_MAX    0x190

10. Can you please suggest what should be the BLE connection interval parameters to receive 4096 bytes with bare minimum delay.

Can you please help me in this as soon as possible. This is really urgent request.

Thanks,
Kumar.

ravibhagavandas, 2021-01-26:
Hi,

I am able to reproduce this with 4096 bytes OTA blocks size. Connection interval was 24ms between IOS and ESP32 device and the MTU was 512 bytes. From the test the whole block is transferred in under a second, but there is an initial delay of 3 seconds between when a block is received from cloud and the first MTU chunk (509 bytes) is send to the device. I will debug this more and will keep you posted.

Meanwhile as a work around to speed up OTA you can reduce the block size to `1024` and increase number of blocks fetched in one request to 3 or 4, by changing the following config in `aws_ota_agent_config.h`
```
#define otaconfigLOG2_FILE_BLOCK_SIZE           10UL
#define otaconfigMAX_NUM_BLOCKS_REQUEST        3U
#define otaconfigMAX_NUM_OTA_DATA_BUFFERS       4U
```

kumarfirmware, 2021-01-26:
Hi Ravi,

Thanks for confirming the issue. Good that you are able to reproduce the same issue on your end.

What function did you use to get connection interval?

I guess this problem is occurring because of very small connection interval.

Is there a function on iOS or ESP32 to set the connection interval to bigger value?

This is really urgent request, we need to solve this as soon as possible. Fetching 1K will increase number of packets and increase the total time. That's the reason I opted for 4K.

Thanks,
Kumar.

ravibhagavandas, 2021-01-27:
You can use ESP function `ble_gap_conn_find()` to get the details of connection parameters. Pass in the connection handle obtained from the connection callback.

The problem should not occur due to small connection interval. Smaller connection interval BLE can send packets quicker (as radio needs  to wait for one connection interval) before sending next packet. So it should lead to increased throughput.
> Fetching 1K will increase number of packets and increase the total time

This should not increase the total time as you also increase number of packets fetched in one request. 

What version of the amazon freertos IOS SDK are you using ? If not using the latest SDK can you upgrade to latest version and test ? Also what IOS device are you using?

kumarfirmware, 2021-01-27:
Hi Ravi,

I do not agree with the small connection interval.  

Small connection interval means we can send only less packets.Bigger connection interval means we can send more packets in one conncetion interval.

Say for example for 4096 bytes block, we need 9 packets of 512 bytes each. This cannot be fit in 24 miliseconds connection interval. If the connection interval is big it can fit all the packets at once, increasing the through put.

Can you please look in to this and explain in detail.

I am using amazon freertos IOS SDK - 1.0.0

I did pod update now and I got AWSIoT (2.22.0) and AWSCore (= 2.22.0) updated. But pod update did not update amazon freertos IOS SDK.

iOS version I am using is 13.7

Can you please help in this as soon as possible.

Thanks,
Kumar.

ravibhagavandas, 2021-01-27:
Hi,
Connection interval is the time interval between each data transfer events between a peripheral and central. The central and peripheral turn on radios and transmit at each connection interval time. So increasing the connection interval will save power (by increasing the time the radio is off), but will decrease throughput as it can transmit only less frequently. Connection interval is always set by central and peripheral can request for a range by providing min and max values. Central will try to set an interval within this range. You can read more about connection interval [here](https://www.novelbits.io/ble-connection-intervals/)

Regarding IOS SDK update, I think `pod update` will only update its dependencies, to get latest SDK can you pull from master branch in this repo: https://github.com/aws/amazon-freertos-ble-ios-sdk/tree/master

kumarfirmware, 2021-01-27:
Hi Ravi,

Thanks for the explanation.

Can you please confirm the issue is fixed in 1.1.0 release or it is fixed only in the master branch?

Because my app is in production, it is risk to pull the code from master. Please confirm as soon as possible.

Thanks,
Kumar.

ravibhagavandas, 2021-01-29:
Hi Kumar,

We found the issue to be with the Swift CBOR encoder used within the IOS SDK, taking close to a second to encode the block of 4k bytes, in our test. We are working on a fix and will update this post soon.

kumarfirmware, 2021-01-29:
Hi Ravi,

Thanks for the explanation.

Does that mean the issue is not fixed in 1.1.0 release and also in the master branch also? I should  not upgrade my code. Please confirm as soon as possible.

Also the time delay for 4K packets is 3 seconds. It is very consistent and easily reproduceable.

Please let me know the fix as soon as possible.

Thanks,
Kumar.

ravibhagavandas, 2021-02-01:
@kumarfirmware 
Yes, we are still working on the fix and confirming it by testing. The fix will be available in the latest master code once tested. I will keep you posted on the updates.

kumarfirmware, 2021-02-03:
Hi Ravi,

Any idea when the fix will be available? we are running short of time.

If we can quickly solve this, it will really help. Already it took good amount of time.

Where do you see the challenge? it is reproduceable very consistently right?

Thanks,
Kumar.

ravibhagavandas, 2021-02-03:
Hi @kumarfirmware 
We have created a pull request for the fixes to resolve the performance issue: https://github.com/aws/amazon-freertos-ble-ios-sdk/pull/20
The bulk of time spent was in encoding the 4k bytes chunks by the mobile SDK thirdy party CBOR library. The library is now changed to a more newer well maintained third party CBOR library and we saw performance improvement for OTA, encode latency reduced from 800ms to 2ms. With this fix, an 800KB OTA image takes close to 2 minutes in our tests with ESP32 device and IOS SDK. Could you verify at your end with this patch.

ravibhagavandas, 2021-02-09:
@kumarfirmware 
Did the above PR fix your issue ? For your information, have merged this PR to master branch of the SDK.

kumarfirmware, 2021-02-11:
Hi Ravi,

Thanks for the message. 

Concern from the team is, since it is a production App, they use pod install, while building the App to get the latest SDK. Now if they take the SDK code from GitHub master branch, will it not be a problem while they build the App again?

Can you please help in this with a solution?

Thanks, 
Kumar.

mingyue86010, 2021-02-17:
Hi @kumarfirmware,

We just packed the latest code and released as v1.2.0 on Github https://github.com/aws/amazon-freertos-ble-ios-sdk/releases . The v1.2.0 has also been published to pod https://cocoapods.org/pods/AmazonFreeRTOS .
The pod install should work now.

Thank you!
Ming

kumarfirmware, 2021-02-17:
Hi Ming,

Thanks a lot this really helps the team. Thanks a lot for your effort.

@ravibhagavandas The latest release fixed the issue of 3 to 5 seconds wait between the packets, i.e. MQTT Request and MQTT Response.

But still I see a wait of 750 milli seconds between MQTT request and Response on MQTT over BLE. 

I am not happy with this 750 milli seconds delay. It is a lot of delay still. Can you please look in to this immediately?

Why does it take 750 milli seconds to receive response?

Thanks,
Kumar.

ravibhagavandas, 2021-02-17:
@kumarfirmware 

Could you tell me, how is the delay of 750ms measured ? Is it delay between a request for block is made from device and a complete block is received at the device ? If so, the BLE connection interval is a contributing factor, as you are allowed to transmit only in a connection interval time. The interval is always chosen by the central device which connects to it. One MTU size chunk is sent within a connection interval as each chunks are send using `write with response` ATT command, so if connection interval is ~100 ms and MTU is 512 bytes it will take ~800ms.

kumarfirmware, 2021-02-17:
Hi Ravi,

Thanks for the message.

750 ms is the delay between a request for block is made from device and a complete block is received at the device.

In the same thread above we clearly discussed Connection interval was 24ms between IOS and ESP32 device and the MTU was 512 bytes.

Even if you take this case of connection interval 24ms it takes 216 ms to collect all 4096 bytes. Still 750 ms is a lot. Still there is a huge delay.

Can you test at your end and let me know where is the problem and how to reduce this 750 ms to the least possible value.

Can you please let me know this as soon as possible.

Thanks,
Kumar.

kumarfirmware, 2021-02-22:
Hi Ravi,

Any progress on this? Where do you see the challenge?

It is reproduceable very consistently right? 

Thanks,
Kumar.

ravibhagavandas, 2021-05-06:
Hi @kumarfirmware,

I have broken down the total latency for a 4232 byte OTA block sent from an Iphone8 (IOS 14.4) to an ESP32 device.

MTU size for the connection is 512 bytes. This means at a time max 509 bytes can be sent from the application layer through the BLE connection.

Here are the logs for the test. The logs are recorded on ESP32 as soon as a 509 byte chunk is received from the Nimble stack. Logs are recorded on the phone as soon as a 509 byte is sent to the IOS BLE stack.

Iphone (sends chunks using Write with response):
```
2021-05-05 16:59:55.326239-0700 AmazonFreeRTOSDemo[21546:6705133] [AFR][36552963-6C7A-F188-3996-54C96252499B][LOT] ↓ 4232 bytes - 1 in queue
2021-05-05 16:59:55.377680-0700 AmazonFreeRTOSDemo[21546:6705133] [AFR][36552963-6C7A-F188-3996-54C96252499B][LOT] ↓ 3723 bytes - 1 in queue
2021-05-05 16:59:55.436826-0700 AmazonFreeRTOSDemo[21546:6705133] [AFR][36552963-6C7A-F188-3996-54C96252499B][LOT] ↓ 3214 bytes - 1 in queue
2021-05-05 16:59:55.496902-0700 AmazonFreeRTOSDemo[21546:6705133] [AFR][36552963-6C7A-F188-3996-54C96252499B][LOT] ↓ 2705 bytes - 1 in queue
2021-05-05 16:59:55.559134-0700 AmazonFreeRTOSDemo[21546:6705133] [AFR][36552963-6C7A-F188-3996-54C96252499B][LOT] ↓ 2196 bytes - 1 in queue
2021-05-05 16:59:55.634449-0700 AmazonFreeRTOSDemo[21546:6705133] [AFR][36552963-6C7A-F188-3996-54C96252499B][LOT] ↓ 1687 bytes - 1 in queue
2021-05-05 16:59:55.706456-0700 AmazonFreeRTOSDemo[21546:6705133] [AFR][36552963-6C7A-F188-3996-54C96252499B][LOT] ↓ 1178 bytes - 1 in queue
2021-05-05 16:59:55.766619-0700 AmazonFreeRTOSDemo[21546:6705133] [AFR][36552963-6C7A-F188-3996-54C96252499B][LOT] ↓ 669 bytes - 1 in queue
2021-05-05 16:59:55.826631-0700 AmazonFreeRTOSDemo[21546:6705133] [AFR][36552963-6C7A-F188-3996-54C96252499B][LOT] LAST PART ↓ 160 bytes - 0 in queue
```

ESP32 (receives chunks and sends a response):
```
296 2903 [ble] Reecived  Chunk 1, length = 509

297 2903 [ble] Sent response for chunk 1

298 2909 [ble] Reecived  Chunk 2, length = 509

299 2909 [ble] Sent response for chunk 2

300 2915 [ble] Reecived  Chunk 3, length = 509

301 2915 [ble] Sent response for chunk 3

302 2921 [ble] Reecived  Chunk 4, length = 509

303 2921 [ble] Sent response for chunk 4

304 2927 [ble] Reecived  Chunk 5, length = 509

305 2927 [ble] Sent response for chunk 5

306 2936 [ble] Reecived  Chunk 6, length = 509

307 2936 [ble] Sent response for chunk 6

308 2942 [ble] Reecived  Chunk 7, length = 509

309 2942 [ble] Sent response for chunk 7

310 2948 [ble] Reecived  Chunk 8, length = 509

311 2948 [ble] Sent response for chunk 8

312 2954 [ble] Reecived  Chunk 9, length = 160

313 2954 [ble] Reecived  final chunk 

314 2954 [ble] Sent response for final chunk

315 2954 [OTA Agent Task] [prvIngestDataBlock] Received file block 9, size 4096

```

Tick rate for ESP32 is 100HZ, which means one tick is 10 ms. As you can see from logs that there is a difference of 60ms (6 ticks * 10 ms) between each chunk received on device and there is same 60 ms difference between when a response is received and next chunk is sent from the Iphone. So its clear that the 60ms is spent in BLE transmission between two BLE stacks.

I will try to break down this latency of 60ms. The final negotiated connection interval is  30ms (24*1.25) as seen from the logs:

```
GAP procedure initiated: connection parameter update; conn_handle=0 itvl_min=16 itvl_max=64 latency=0 supervision_timeout=100 min_ce_len=0 max_ce_len=0
6 279 [ble] [INFO ][DEMO][2790] BLE Connected to remote device, connId = 0

7 287 [ble] Connection parameter update interval = 24 latency = 0, timeout = 72
```

Here is a correction from what I said before, when I said a packet is sent in connection interval this actually means one link layer packet which is 251 bytes(with the Data Length Extension enabled). But Iphone can send upto 4 LL packet in  packet per interval. Splitting 509 bytes, this will end up in 3 packets per connection interval and will take an additional interval to get the response.  Hence its close to 60ms per 509 bytes which is seen in the logs.

To improve this further you could try reducing the minimum connection interval to max connection interval range, following  apple recommendations  [here](https://developer.apple.com/library/archive/qa/qa1931/_index.html). However the final negotiated connection interval  varies depending on the environment, and the devices used.

ravibhagavandas, 2021-05-14:
Closing this issue assuming the latency related queries are resolved. Please feel free to open a new issue if you have further questions.

closed 2021-05-14 by ravibhagavandas