aws/amazon-freertos #2938: Is it possible to use aws_iot_ota_update_demo.c for both WI-FI  and BLE with out using Network Manager?

kumarfirmware, 2021-01-14:
Hi Team,

I am using AWS FreeRTOS version 202012 and ESP32.

Is there a way to use aws_iot_ota_update_demo.c for BLE also?  I mean MQTT over BLE.

Is it possible to use aws_iot_ota_update_demo.c for both WI-FI  and BLE with out using Network Manager?

Can you provide me a sample how do you use  aws_iot_ota_update_demo.c for  MQTT over BLE with out using Network Manager?

I would like to use both BLE and WI-FI mutually exclusively.

Thanks
Kumar.

pvyawaha, 2021-01-15:
Hello, 

Please refer to the following blog post and documentation for running OTA on ESP32 platform using BLE -

https://aws.amazon.com/blogs/iot/perform-ota-firmware-updates-on-espressif-esp32-devices-using-amazon-freertos-bluetooth-low-energy-mqtt-proxy/
https://docs.aws.amazon.com/freertos/latest/userguide/ble-demo.html

We do not have a demo without network manager and using both BLE and WiFi at this time.

kumarfirmware, 2021-01-15:
Hi pvyawaha,

Thanks for your immediate response. I understood from you, with out Network Manager we cannot use both BLE and WI-FI for OTA.  Even though I need BLE and WI-FI mutually exclusive.

I can close this issue.

Thanks,
Kumar.

closed 2021-01-15 by kumarfirmware