aws/amazon-freertos #2941: add_executable cannot create target "my_app" because another target with   the same name already exists

romaindequidt, 2021-01-14:
**Describe the bug**
I tried to create my own CMake project for ESP32.

I have clone amazon-freertos ([master](https://github.com/aws/amazon-freertos/commit/2c143433d7a18391ce242c06856452c921239889)) and all its submodules.

Create my own CMakeLists.txt as expained here: 
[Using FreeRTOS in your own CMake project for ESP32](https://docs.aws.amazon.com/freertos/latest/userguide/getting_started_espressif.html)

And run te following CMake command:
```
cmake -S . -B build -DCMAKE_TOOLCHAIN_FILE=freertos/tools/cmake/toolchains/xtensa-esp32.cmake -GNinja
```

And got te following error:
```
CMake Error at freertos/vendors/espressif/boards/esp32/CMakeLists.txt:423 (add_executable):
  add_executable cannot create target "my_app" because another target with
  the same name already exists.  The existing target is an executable created
  in source directory "/home/rom/Documents/dev/iotercom/iotercom-freertos".
  See documentation for policy CMP0002 for more details.
Call Stack (most recent call first):
  freertos/CMakeLists.txt:70 (include)
```

**System information**
- Which hardware board or part numbers? esp32_devkitc
- Operating System Ubuntu 20.10 5.8.0-38-generic #43-
- Version of FreeRTOS: 202012.00-230-g2c143433d
- Configuration of FreeRTOS:
```
Target microcontroller:
  vendor:                  Espressif
  board:                   ESP32-DevKitC
  description:             Development board produced by Espressif that comes in two 
                           variants either with ESP-WROOM-32 or ESP32-WROVER module
  family:                  ESP32
  data ram size:           520KB
  program memory size:     4MB

Host platform:
  OS:                      Linux-5.8.0-38-generic
  Toolchain:               xtensa-esp32
  Toolchain path:          /home/user/.espressif/tools/xtensa-esp32-elf/esp-2020r3-8.4.0/xt
                           ensa-esp32-elf
  CMake generator:         Ninja
```
May you help me?

Thank you!

romaindequidt, 2021-01-14:
Ok. Nevermind. I found your working [CMakeLists.txt](https://github.com/shubhamkulkarni97/amazon-freertos-examples/blob/feature/idf_v4.2/CMakeLists.txt) and by adapting mine, i've made it worked!
Sorry for the disturb.
Note: be carreful, there is a bug in the "issue interface" there are created twice.

pvyawaha, 2021-01-15:
Thank you for sharing update. Can you please share what you mean by the bug ?

closed 2021-01-19 by pvyawaha