aws/amazon-freertos #2949: How to log error on OTA update with AWSIotLogsV2 ?

elieDaan, 2021-01-21:
Hello, 

I want to know when I launch an OTA on a group of objet why the OTA is failed on CloudWatch like what we've got on the monitor. For example, If I launch an ota on a device with the same software, the Ota will be refused with the reason "same softer at reboot". 

I've seen this [documentation on AWS IOT Logging](https://docs.aws.amazon.com/iot/latest/developerguide/cloud-watch-logs.html). 

So I have on CloudWatch the log AWSIotLogsV2 but it's not enough. I've just on log : 

{
    "timestamp": "2021-01-20 11:50:04.331",
    "logLevel": "ERROR",
    "accountId": "XXX",
    "status": "Success",
    "eventType": "UpdateJobExecution",
    "protocol": "MQTT",
    "clientId": "XXX",
    "jobId": "XXXXX",
    "topicName": "$aws/things/XXX/jobs/XXXX/update",
    "details": "The destination status is FAILED. The request status is SUCCESS."
}

Thanks for your help

pvyawaha, 2021-01-22:
Hello,

The detailed status codes are updated in the respective jobs for the things, please refer to the troubleshooting guide https://docs.aws.amazon.com/freertos/latest/userguide/ota-failure-codes.html

For example if the OTA update job failed due to same version the error code will be `0x23000000UL`

List of error codes - https://www.freertos.org/Documentation/api-ref/ota/ota_err_constants.html

elieDaan, 2021-01-25:
Thank you very much, I found my answer by using the aws cli describe-job-execution. 

Do you have an idea what can cause kOTA_Err_CommitFailed ? 

Also, this week end I have a lot of devices being refused to connect to AWS IOT.  I think they all wanted to see if an update was available at the same time and at the same time other devices has been updated so requesting a lot of packets.
Do you think which [AWS IOT Quotas](https://docs.aws.amazon.com/general/latest/gr/iot-core.html#limits_iot) could have been triggered ? 

Thanks for you help :

pvyawaha, 2021-01-26:
Hello,

When a new image is received it is executed in self test , after successful completion of self test i.e connecting to AWS IoT service and updating the status it is committed. This also requires to change a flag on the device side to mark image as committed  and kOTA_Err_CommitFailed is when updating this flag fails.

The connections should be done using exponential backoff and random delay. Can you share more information about which version of the libraries you are using , network and scale of devices deployed?

elieDaan, 2021-01-27:
Well I'm using a version around july 2019. I can't tell you the exact version because AFR is not anymore a submodule of my project but a repo. The network using it's MQTT and I've got around 7000 devices deployed. 

What do you mean about exponential backoff ?

aggarw13, 2021-01-28:
Hi @elieDaan, please refer to this [AWS documentation on Exponential Backoff And Jitter](https://aws.amazon.com/blogs/architecture/exponential-backoff-and-jitter/) delay when re-attempting connection requests to avoid network congestion and thereby, enable the server to service requests.
(With version 202012.00, we also provide a library for calculating the backoff delay in [FreeRTOS/backoffAlgorithm](https://github.com/FreeRTOS/backoffAlgorithm). [Here](https://github.com/aws/amazon-freertos/blob/master/demos/coreMQTT/mqtt_demo_mutual_auth.c#L671-L717) is an example in the latest MQTT mutual authentication demo of using the library.)

elieDaan, 2021-01-28:
Thanks for the doc it's really interesting ! I wasn't aware of this kind of issues.  Maybe I need to implement something similar in my system, the principle is to wait a random delay (can be optimized with the jitter) before retry the connection, right ? 

 
Have you got some advice to update the AFR repo when we have modified the AFR code ? I know it's a bad practice but some times there is no other options..

aggarw13, 2021-01-28:
Hi @elieDaan,

>Thanks for the doc it's really interesting ! I wasn't aware of this kind of issues. Maybe I need to implement something similar in my system, the principle is to wait a random delay (can be optimized with the jitter) before retry the connection, right ?

Happy to know that the document is helpful. 😃  
Yes, for network connection re-tries, we recommend having an exponential incrementing delay for every retry attempt (after the failed 1st attempt) along with randomness so that if multiple devices face connection failures at the same time, they are not utilizing the same delay period. You can reference the links I shared above for the algorithm to calculate the backoff delay and its use for connection retries in demo applications.

> Have you got some advice to update the AFR repo when we have modified the AFR code ? I know it's a bad practice but some times there is no other options..

It will be difficult for me to suggest a plan without knowing the amount of changes you have made from a release version. Can you confirm that you have based your project off of the [202007.00](https://github.com/aws/amazon-freertos/tree/202007.00/) version?
However, the general advice I can give is to create a separate patch of modifications you have made from the original Amazon FreeRTOS version you used, and then attempt to apply those patches on a clone of a recent release version (like [202012.00](https://github.com/aws/amazon-freertos/tree/202012.00/)). 
You can determine the changeset across recent releases from your base release version by viewing the [CHANGELOG.md](https://github.com/aws/amazon-freertos/blob/master/CHANGELOG.md) and evaluate the scope of work and risk involved in upgrading the FreeRTOS version in your codebase.

elieDaan, 2021-01-29:
Hi ,

Well my project is based on the 202002.00 version. It's a good idea to do some patch ! 
Thank you all for your amazing work and your support.

closed 2021-01-29 by elieDaan