aws/amazon-freertos #2952: Logging issues

MichalChromecNXP, 2021-01-25:
Hello,
I think the logs from the demo examples are confusing.
I have tried example: device_shadow_for_aws and I get this log:

```
.
.
.
30 39316 [iot_thread] [INFO] [MQTT] [...\rtos\freertos\libraries\coreMQTT\source\core_mqtt.c:1563] 31 39316 [iot_thread] Received MQTT CONNACK successfully from broker.32 39316 [iot_thread]
33 39316 [iot_thread] [INFO] [MQTT] [...\rtos\freertos\libraries\coreMQTT\source\core_mqtt.c:1829] 34 39316 [iot_thread] MQTT connection established with the broker.35 39316 [iot_thread]
36 39316 [iot_thread] [INFO] [MQTT] [...\rtos\freertos\demos\common\mqtt_demo_helpers\mqtt_demo_helpers.c:632] 40 39316 [iot_thread] A clean MQTT connection is established. Cleaning up all the stored outgoing publishes.

                                                    45 39500 [iot_thread] [INFO] [MQTT] [...\rtos\freertos\libraries\coreMQTT\source\core_mqtt.c:886] 46 39500 [iot_thread] Packet received. ReceivedBytes=3.47 39500 [iot_thread]
48 39500 [iot_thread] [INFO] [MQTT] [...\rtos\freertos\demos\common\mqtt_demo_helpers\mqtt_demo_helpers.c:463] 49 39500 [iot_thread] MQTT_PACKET_TYPE_SUBACK.

                                                                                                                                                                                           50 39500 [iot_thread]
51 39902 [iot_thread] [INFO] [MQTT] [...\rtos\freertos\demos\common\mqtt_demo_helpers\mqtt_demo_helpers.c:751] 52 39902 [iot_thread] SUBSCRIBE topic $aws/things/aws_rpr_michal/shadow/delete/rejected to broker.

                                          53 39902 [iot_thread]
.
.
.
```

I have found the new version of Amazon-FreeRTOS (202012.00) is using new logging in which compose the log in this way:
```
        #define LogInfo( message )     SdkLog( ( "[INFO] [%s] "LOG_METADATA_FORMAT, LIBRARY_LOG_NAME, LOG_METADATA_ARGS ) ); SdkLog( message ); SdkLog( ( "\r\n" ) )
```
I see the incompatibility with the logging function: vLoggingPrintf. If the macro: configLOGGING_INCLUDE_TIME_AND_TASK_NAME is set to 1, each call of the SdkLog adds to message timestamp and task name, so in one call of LogInfo we get them three times in one message log.
Now each log message contains path to file and line from which the log was called. Is it necessary? I think it is useful for error logs, so would be possible to add some macro for disabling this feature in info logs?

Could you please help fix these issues?

Thank you,
Michal Chromec

abhidixi11, 2021-01-27:
Hello @MichalChromecNXP 
Thanks for reaching out and pointing out the problem. We are taking a look and will get back to you soon.

aggarw13, 2021-01-29:
Hi @MichalChromecNXP, as you have rightfully pointed out, when the `configLOGGING_INCLUDE_TIME_AND_TASK_NAME` is enabled, the task name and tick count metadata are added 3 times for every call to the `LogInfo` macro. 
I am working on a solution ([in PR]()) to avoid the repetition of information that will improve the readability of the logs. 

Regarding your request for a macro to disable the file and line number metadata information from logs, there exist the `LOG_METADATA_FORMAT` and `LOG_METADATA_ARGS` macros in the [`logging_stack.h`](https://github.com/aws/amazon-freertos/blob/202012.00/libraries/logging/include/logging_stack.h#L51-L52) file which we can consider exposing for user configuration. These macros configure metadata for all logging levels. 
Do you think that exposing these macros meet your needs for disabling file and line metadata? Or are you looking for log-level specific granularity of configuration for disabling the line and file information?

MichalChromecNXP, 2021-02-01:
Hi @aggarw13 ,
I have built the shadow example in Windows OS (IAR), so the problem in my log is that I don't see only filename (which would be OK), but I see whole path to the file (sometime it could be very long). I think it is problem in these lines (logging_stack.h):
```
/* Macro to extract only the file name from file path to use for metadata in
 * log messages. */
#ifdef _MSC_VER
    #define FILENAME           ( strrchr( __FILE__, '\\' ) ? strrchr( __FILE__, '\\' ) + 1 : __FILE__ )
#else
    #define FILENAME           ( strrchr( __FILE__, '/' ) ? strrchr( __FILE__, '/' ) + 1 : __FILE__ )
#endif
```
The definition of FILENAME should depends on the operating system not the compiler.

Regarding the option to remove the file name from the log - I proposed it, because I was confused from the log - the whole path to file name, but it is probably bug in the FILENAME definition, so removing of filename from the log will not be probably necessary.

Thank you,
Michal

aggarw13, 2021-02-04:
Hi @MichalChromecNXP, thank you for the feedback. I will look into utilizing the host OS to update the logic for `FILENAME`. 
In the meanwhile, I have made updates to the [PR](#2953 ) (I had linked above for fixing the problem of repeated metadata in messages) to also make it simpler to remove extra metadata if not required for your needs. 
The changes I pushed provides implementation for both ISO C90 and ISO C99 standards, where the C90 implementation only adds the library name metadata whereas the C99 implementation adds the filename and line number metadata. 
If you would like to remove the filename and line number metadata, you can simply use the C90 based implementation from the PR.

aggarw13, 2021-02-09:
Hi @MichalChromecNXP, we have merged the changes for fixing the issue of repeated metadata in log messages. 
To fix the issue of filepath taking up log buffer space in your builds, we have merged another change ([PR](#2982)) that rectifies the logic of filename extraction from file-path.  
As mentioned in my previous post, with the new changes, the filename metadata is only added when the ISO C99 implementation is used (through the `LOGGING_ENABLE_METADATA_WITH_C99_AND_GNU_EXTENSION` config). If you do not want the filename metadata to be present in the logs, you can use the ISO C90 base implementation by setting the aforementioned macro to `0`.

Please let us know if your issues stand resolved with the changes or if you have anymore questions.

closed 2021-03-08 by aggarw13