aws/amazon-freertos #2954: "ERROR: SOCKETS_Connect call failed" in greengrass discovery Demo

anubhav1, 2021-01-26:
**Describe the bug**
I am using Ubuntu(WSL2) on Windows as a GGCore Device and a ESP32DevKitC as an IoT Thing. I ran MQTT_muth_auth Demo on ESP32DevKitC which worked well. But while testing GREENGRASS_DISCOVERY_DEMO, I am getting an error.
The Error is : `SOCKETS_Connect call failed` . Logs are below.

**System information**
- Which hardware board or part numbers? : **GG Core** : Ubuntu(WSL2) on Win 10 , **IoT Thing** : ESP32DevKitC V4
- Operating System: Windows
- Project: GREENGRASS_DISCOVERY_DEMO

**Expected behavior**
ESP32DevKitC should be connected to GGCore

**To reproduce**
Build and flash the GREENGRASS_DISCOVERY_DEMO on ESP32DevKitC.

**Logs**
```
I (490) esp_image: segment 5: paddr=0x0010db64 vaddr=0x40080400 size=0x1bc08 (113672) load
I (562) boot: Loaded app from partition at offset 0x20000
I (562) boot: Disabling RNG early entropy source...
I (563) cpu_start: Pro cpu up.
I (566) cpu_start: Application information:
I (571) cpu_start: Project name:     esp-idf
I (576) cpu_start: App version:      1
I (580) cpu_start: Compile time:     Jan 26 2021 12:10:08
I (586) cpu_start: ELF file SHA256:  1e70aacf991eb28b...
I (592) cpu_start: ESP-IDF:          HEAD-HASH-NOTFOUND
I (598) cpu_start: Single core mode
I (603) heap_init: Initializing. RAM available for dynamic allocation:
I (610) heap_init: At 3FFAFF10 len 000000F0 (0 KiB): DRAM
I (616) heap_init: At 3FFB6388 len 00001C78 (7 KiB): DRAM
I (622) heap_init: At 3FFB9A20 len 00004108 (16 KiB): DRAM
I (628) heap_init: At 3FFBDB5C len 00000004 (0 KiB): DRAM
I (634) heap_init: At 3FFCEA80 len 00011580 (69 KiB): DRAM
I (640) heap_init: At 3FFE0440 len 0001FBC0 (126 KiB): D/IRAM
I (647) heap_init: At 40078000 len 00008000 (32 KiB): IRAM
I (653) heap_init: At 4009C008 len 00003FF8 (15 KiB): IRAM
I (659) cpu_start: Pro cpu start user code
I (6) cpu_start: Starting scheduler on PRO CPU.
E (29) uart: uart_set_hw_flow_ctrl(268): rx flow thresh error
0 1 [main] Initializing lwIP TCP stack
I (29) PKCS11: Initializing NVS partition: "storage"
E (29) PKCS11: failed nvs get file size 4354 0
1 1 [main] Write certificate...
2 6 [iot_thread] [INFO ][DEMO][60] ---------STARTING DEMO---------

I (79) BTDM_INIT: BT controller compile version [0fff744]
3 6 [iot_thread] [INFO ][INIT][60] SDK successfully initialized.
I (89) system_api: Base MAC address is not set, read default base MAC address from BLK0 of EFUSE
I (189) phy: phy_version: 4180, cb3948e, Sep 12 2019, 16:39:13, 0, 0
GAP procedure initiated: advertise; disc_mode=2 adv_channel_map=0 own_addr_type=1 adv_filter_policy=0 adv_itvl_min=187 adv_itvl_max=375
I (499) wifi:wifi driver task: 3ffdbb4c, prio:23, stack:3584, core=0
I (499) system_api: Base MAC address is not set, read default base MAC address from BLK0 of EFUSE
I (499) system_api: Base MAC address is not set, read default base MAC address from BLK0 of EFUSE
I (519) wifi:wifi firmware version: 1afdae0
I (519) wifi:config NVS flash: enabled
I (519) wifi:config nano formating: disabled
I (519) wifi:Init dynamic tx buffer num: 32
I (519) wifi:Init data frame dynamic rx buffer num: 32
I (529) wifi:Init management frame dynamic rx buffer num: 32
I (539) wifi:Init management short buffer num: 32
I (539) wifi:Init static rx buffer size: 1600
I (539) wifi:Init static rx buffer num: 10
I (549) wifi:Init dynamic rx buffer num: 32
I (559) wifi:mode : sta (24:0a:c4:5f:9b:f0)
I (559) WIFI: SYSTEM_EVENT_STA_START
I (3219) wifi:new:<11,0>, old:<1,0>, ap:<255,255>, sta:<11,0>, prof:1
I (4199) wifi:state: init -> auth (b0)
I (4209) wifi:state: auth -> assoc (0)
I (4219) wifi:state: assoc -> run (10)
I (4229) wifi:connected with TP-LINK_539C, aid = 8, channel 11, BW20, bssid = 50:c7:bf:43:53:9c
I (4239) wifi:security type: 3, phy: bgn, rssi: -58
I (4239) wifi:pm start, type: 1

I (4239) WIFI: SYSTEM_EVENT_STA_CONNECTED
I (4259) wifi:AP's beacon interval = 102400 us, DTIM period = 1
I (6169) event: sta ip: 192.168.1.105, mask: 255.255.255.0, gw: 192.168.1.1
I (6169) WIFI: SYSTEM_EVENT_STA_GOT_IP
4 615 [eventTask] [INFO ][DEMO][6150] Connected to WiFi access point, ip address: 192.168.1.105.
5 616 [iot_thread] [INFO ][DEMO][6160] Successfully initialized the demo. Network type for the demo: 1
GAP procedure initiated: stop advertising.
6 616 [iot_thread] [INFO ][MQTT][6160] MQTT library successfully initialized.
7 616 [iot_thread] [INFO ][DEMO][6160] Attempting automated selection of Greengrass device

GAP procedure initiated: stop advertising.
E (6279) PKCS11: failed nvs get file size 4354 0
8 957 [iot_thread] JSON file retrieval completed
9 957 [iot_thread] About to close socket.
10 958 [iot_thread] Socket closed.
11 958 [iot_thread] Stack high watermark for discovery helper task: 5336.
12 2796 [iot_thread] LwIP connect fail -1 113
13 2796 [iot_thread] ERROR! SOCKETS_Connect call failed: ServerAddress=3729791660, Port=45858, ReturnCode=-1
14 2796 [iot_thread] About to close socket.
15 2796 [iot_thread] Socket closed.
16 2796 [iot_thread] Stack high watermark for discovery helper task: 4952.
17 2797 [iot_thread] ERROR! Failed to resolve host address: ServerHost=::118 4647 [iot_thread] LwIP connect fail -1 113
19 4647 [iot_thread] ERROR! SOCKETS_Connect call failed: ServerAddress=0, Port=45858, ReturnCode=-1
20 4647 [iot_thread] About to close socket.
21 4648 [iot_thread] Socket closed.
22 4648 [iot_thread] Stack high watermark for discovery helper task: 4952.
23 6499 [iot_thread] LwIP connect fail -1 113
24 6499 [iot_thread] ERROR! SOCKETS_Connect call failed: ServerAddress=33022, Port=45858, ReturnCode=-1
25 6499 [iot_thread] About to close socket.
26 6499 [iot_thread] Socket closed.
27 6499 [iot_thread] Stack high watermark for discovery helper task: 4952.
28 6499 [iot_thread] GGD - Can't connect to greengrass Core
29 6499 [iot_thread] [ERROR][DEMO][64990] Auto-connect: Failed to retrieve Greengrass address and certificate.
30 6499 [iot_thread] [INFO ][MQTT][64990] MQTT library cleanup done.
31 6499 [iot_thread] [INFO ][DEMO][64990] Cleaned up MQTT library.
32 6499 [iot_thread] [INFO ][DEMO][64990] memory_metrics::freertos_heap::before::bytes::193688
33 6499 [iot_thread] [INFO ][DEMO][64990] memory_metrics::freertos_heap::after::bytes::101644
34 6499 [iot_thread] [INFO ][DEMO][64990] memory_metrics::demo_task_stack::before::bytes::8920
35 6501 [iot_thread] [INFO ][DEMO][64990] memory_metrics::demo_task_stack::after::bytes::4952
I (66029) wifi:state: run -> init (0)
I (66029) wifi:pm stop, total sleep time: 55304074 us / 61788635 us

I (66029) wifi:new:<11,0>, old:<11,0>, ap:<255,255>, sta:<11,0>, prof:1
I (66029) WIFI: SYSTEM_EVENT_STA_DISCONNECTED: 8
E (66039) wifi:wifi deinit internal: wifi not stop
36 6601 [iot_thread] [ERROR][DEMO][66010] Error running demo.
37 6602 [eventTask] [INFO ][DEMO][66020] Disconnected from WiFi access point, reason code: 0.
I (66059) wifi:flush txq
I (66059) wifi:stop sw txq
I (66059) wifi:lmac stop hw txq
I (66059) wifi:Deinit lldesc rx mblock:10
38 6606 [iot_thread] [INFO ][INIT][66060] SDK cleanup done.
39 6606 [iot_thread] [INFO ][DEMO][66060] -------DEMO FINISHED-------

I (186029) event: station ip lost
```

anubhav1, 2021-01-26:
It seems like FreeRtos is converting my core's ip address (from json) to a host name 3729791660 using `SOCKETS_GetHostByName( pxHostAddressData->pcHostAddress );` but this host name is weird. Host name is supposed to my PC name.

aggarw13, 2021-01-28:
Hi @anubhav1, we do not officially support GreenGrass Core on WSL2. 
Thus, to determine whether the issue is with the FreeRTOS side or with setup of GreenGrass Core on WSL2, would you be perform the Discovery test using another SDK (like the Python sample for GreenGrass Discovery available [here](https://github.com/aws/aws-iot-device-sdk-python/blob/master/samples/greengrass/basicDiscovery.py)) and check if you can connect to GreenGrassCore?

yourslab, 2021-02-07:
The reason why the IP address looks weird is because it is encoded as a 32-bit integer. In your case, `3729791660` maps to `172.26.80.222.80`. I'm guessing this is the hostname of your GGC. Have you added it as a host system using the steps below?

1. From `AWS IoT` -> `Greengrass` -> `Cores` -> [`Name of Core`] -> `Connectivity` manually add the local IP address of the host system with port 8883. 
2. Launch the FreeRTOS GG discovery demo on the ESP32.

anubhav1, 2021-02-10:
I added the local ip address of the core device with port 8883 in **AWS IoT -> Greengrass -> Cores -> [Name of Core] -> Connectivity**. I found the core's local ip address using `hostname -I`. But still i am getting same error.  This time I am running core on ` ubuntu-18.04 on Hypervisor on win 10`.  Greengrass daemon is up and running on core. 
Error Logs:

```
4 371 [eventTask] [INFO ][DEMO][3710] Connected to WiFi access point, ip address: 192.168.1.103.
5 372 [iot_thread] [INFO ][DEMO][3710] Successfully initialized the demo. Network type for the demo: 1
6 372 [iot_thread] [INFO ][MQTT][3720] MQTT library successfully initialized.
7 372 [iot_thread] [INFO ][DEMO][3720] Attempting automated selection of Greengrass device

E (3877) PKCS11: failed nvs get file size 4354 0
8 729 [iot_thread] JSON file retrieval completed
9 729 [iot_thread] About to close socket.
10 730 [iot_thread] Socket closed.
11 730 [iot_thread] Stack high watermark for discovery helper task: 5160.
12 2555 [iot_thread] LwIP connect fail -1 113
13 2555 [iot_thread] ERROR! SOCKETS_Connect call failed: ServerAddress=32811200, Port=45858, ReturnCode=-1
14 2555 [iot_thread] About to close socket.
15 2555 [iot_thread] Socket closed.
16 2555 [iot_thread] Stack high watermark for discovery helper task: 4952.
17 2556 [iot_thread] ERROR! Failed to resolve host address: ServerHost=::118 4406 [iot_thread] LwIP connect fail -1 113
19 4406 [iot_thread] ERROR! SOCKETS_Connect call failed: ServerAddress=0, Port=45858, ReturnCode=-1
20 4406 [iot_thread] About to close socket.
21 4407 [iot_thread] Socket closed.
22 4407 [iot_thread] Stack high watermark for discovery helper task: 4952.
23 6258 [iot_thread] LwIP connect fail -1 113
24 6258 [iot_thread] ERROR! SOCKETS_Connect call failed: ServerAddress=33022, Port=45858, ReturnCode=-1
25 6258 [iot_thread] About to close socket.
26 6258 [iot_thread] Socket closed.
27 6259 [iot_thread] Stack high watermark for discovery helper task: 4808.
28 6259 [iot_thread] GGD - Can't connect to greengrass Core
29 6259 [iot_thread] [ERROR][DEMO][62590] Auto-connect: Failed to retrieve Greengrass address and certificate.
30 6259 [iot_thread] [INFO ][MQTT][62590] MQTT library cleanup done.
31 6259 [iot_thread] [INFO ][DEMO][62590] Cleaned up MQTT library.
32 6259 [iot_thread] [INFO ][DEMO][62590] memory_metrics::freertos_heap::before::bytes::193556
33 6259 [iot_thread] [INFO ][DEMO][62590] memory_metrics::freertos_heap::after::bytes::74808
34 6259 [iot_thread] [INFO ][DEMO][62590] memory_metrics::demo_task_stack::before::bytes::8920
35 6259 [iot_thread] [INFO ][DEMO][62590] memory_metrics::demo_task_stack::after::bytes::4808
I (63607) wifi:state: run -> init (0)
I (63607) wifi:pm stop, total sleep time: 52478428 us / 61043268 us

I (63607) wifi:new:<4,0>, old:<4,0>, ap:<255,255>, sta:<4,0>, prof:1
I (63607) WIFI: SYSTEM_EVENT_STA_DISCONNECTED: 8
E (63617) wifi:wifi deinit internal: wifi not stop
36 6359 [iot_thread] [ERROR][DEMO][63590] Error running demo.
37 6360 [eventTask] [INFO ][DEMO][63600] Disconnected from WiFi access point, reason code: 0.
I (63627) wifi:flush txq
I (63637) wifi:stop sw txq
I (63637) wifi:lmac stop hw txq
I (63637) wifi:Deinit lldesc rx mblock:10
GAP procedure initiated: stop advertising.
GAP procedure initiated: stop advertising.
38 6365 [iot_thread] [INFO ][INIT][63650] SDK cleanup done.
39 6365 [iot_thread] [INFO ][DEMO][63650] -------DEMO FINISHED-------

I (183607) event: station ip lost
```

anubhav1, 2021-02-10:
One more question : Is there any online tool to convert ip address to 32 bit integer? I couldnt find any.

yourslab, 2021-02-10:
Are you able to ping the IP address from another machine in the same network? It could be that the Windows host machine is blocking some network traffic to the VM, but I am not entirely certain. A packet capture would be very helpful. I like using the Windows host machine as a WiFi AP then utilizing Wireshark to monitor my network traffic.

This is the tool I used:
https://www.vultr.com/resources/ipv4-converter/?ip_address=32811200
Although, keep in mind that it prints each byte in reverse order.

anubhav1, 2021-02-10:
Sorry but I am not familiar with wireshark.  Is there any other way to check?
I am able to ping the Ip address of ubuntu-18.04 on Hypervisor from the same computer's cmd prompt.

yourslab, 2021-02-10:
I think Wireshark is the easiest way to do packet capture. Can you also ping Ubuntu from another computer? Maybe only the same computer is able to ping.

anubhav1, 2021-02-10:
I jused tried pinging from other pc. Not able to ping.

anubhav1, 2021-02-10:
One thing I also noticed that even if there is one ip address in  Core's Connectivity , I see 2 server address in logs everytime.

yourslab, 2021-02-10:
If you forward the port from Ubuntu to host with the public IP, then it should also work. It looks like the Greengrass Daemon is picking up those addresses, but since you are running from a VM, you may need to do more manual configuration. I don't think there's any harm in having multiple addresses though because only one of them need to succeed.

anubhav1, 2021-02-12:
After installing Ubuntu 18.04 on a physical hardware , i am able to connect to GreenGrass. Seems like , greengrass running on VM/WSL2 needs a lil bit research. As my problem solved, I am closing it for now.

closed 2021-02-12 by anubhav1