aws/amazon-freertos #2956: private key not found for HTTPS

surya-teja-p, 2021-01-28:
Hi ,
 I am using 202007.00 July 2020 version of source code I am trying to perform HTTPS post request it requires only ROOT CA, which I am providing and it is working with POSTMAN but when I am trying with https library in C_SDK I am facing private key not found error. I am setting ClientCert and PrivateKey to NULL.

below are the logs :
0 739 [iot_thread] ERROR: Private key not found. 
1 739 [iot_thread] TLS_Connect fail (0x7d4, URL)
2 739 [iot_thread] [ERROR][NET][7390] Failed to establish new connection. Socket status: -1.
3 740 [iot_thread] [ERROR][HTTPS Client][7400] Failed to connect to the server at URL on port 443 with error: 4
4 740 [iot_thread] [ERROR][HTTPS Client][7400] Error in IotHttpsClient_Connect(). Error code 203.
5 740 [iot_thread] [ERROR][HTTPS Client][7400] connHandle was NULL.
6 740 [iot_thread] [ERROR][HTTPS Client][7400] IotHttpsClient_SendSync() failed.

Thanks 
Surya

aggarw13, 2021-01-29:
Hi @surya-teja-p, I am investigating into whether our TLS stack in the 202007.00 release supports establishing TLS connection in server authentication mode only. 
I will get back soon once I have more information.

surya-teja-p, 2021-01-29:
Hi @aggarw13 ,

That would be lot helpful if you can info about the issue

Thanks 
Surya

lundinc2, 2021-01-29:
Hi @surya-teja-p,

I am unfamiliar with postman, but I have created this branch based off the 202007.00 release https://github.com/lundinc2/amazon-freertos/tree/basic-tls-http. I have modified the S3 download sync demo to perform a request using basic TLS to https://www.httpbin.org. I hope you find this example useful, and it should work with no modifications. 

Thanks,

Carl

surya-teja-p, 2021-01-29:
Hi @lundinc2 

I hope this is the link you want me to check, [https://github.com/lundinc2/amazon-freertos/blob/basic-tls-http/demos/https/iot_demo_https_s3_download_sync.c](url), In this source file you are assigning private key and clientcert but setting the ca-certificate to NULL, I need other way to perform the operation with only ca-cert and no clientcert and privatekey, if I am wrong please correct me in understanding the code you mentioned.

Thanks 
Surya

lundinc2, 2021-01-29:
Hi @surya-teja-p,

My credentials were set to NULL in the `aws_clientcredential_keys.h` file. I have updated my branch to set the client and private key to NULL explicitly. This sounds like it will fulfill the requirements you are looking for. Please let me know if I can help further.

Thanks,

Calr

surya-teja-p, 2021-01-29:
Hi @lundinc2 

I have followed your iot_demo_https_s3_download_sync.c file but still the mentioned error came up, can you mention the changes that I need to do. The server needs ROOT-CA(caCert) to authenticate the communication.

This is my code
void runhttp(const IotNetworkInterface_t *networkInterface)
{
    IotHttpsReturnCode_t returnCode = IOT_HTTPS_OK;
    IotHttpsConnectionHandle_t connHandle = IOT_HTTPS_CONNECTION_HANDLE_INITIALIZER;
    IotHttpsRequestHandle_t reqHandle = IOT_HTTPS_REQUEST_HANDLE_INITIALIZER;
    IotHttpsResponseHandle_t respHandle = IOT_HTTPS_RESPONSE_HANDLE_INITIALIZER;
    IotHttpsRequestInfo_t reqInfo = IOT_HTTPS_REQUEST_INFO_INITIALIZER;
    IotHttpsResponseInfo_t respInfo = IOT_HTTPS_RESPONSE_INFO_INITIALIZER;
    IotHttpsSyncInfo_t syncRespInfo = IOT_HTTPS_SYNC_INFO_INITIALIZER;
    IotHttpsSyncInfo_t syncReqInfo = IOT_HTTPS_SYNC_INFO_INITIALIZER;
    uint32_t contentLength = 0;
    uint16_t responseStatus = 0;
    uint16_t respStatus = IOT_HTTPS_STATUS_OK;

    static IotHttpsConnectionInfo_t _connInfo =
    {
        .pAddress = IOT_TEST_HTTPS_SERVER_HOST_NAME,
        .addressLen = sizeof(IOT_TEST_HTTPS_SERVER_HOST_NAME) - 1,
        .port = IOT_TEST_HTTPS_PORT,
        .flags = 0,
        .userBuffer.pBuffer = _pConnUserBuffer,
        .userBuffer.bufferLen = sizeof(_pConnUserBuffer),
        .pNetworkInterface = networkInterface,
        .pCaCert = IOT_ROOT_CA,
        .caCertLen = sizeof(IOT_ROOT_CA),
        .pClientCert = NULL,   
        .clientCertLen = NULL, 
        .pPrivateKey = NULL,   
        .privateKeyLen = NULL, 
        .pAlpnProtocols = HTTPS_TEST_ALPN_PROTOCOL,
        .alpnProtocolsLen = sizeof(HTTPS_TEST_ALPN_PROTOCOL),
    };
    IotHttpsMethod_t method = IOT_HTTPS_METHOD_POST;

    respSyncInfo.pBody = _pRespBodyBuffer;          
    respSyncInfo.bodyLen = sizeof(_pRespBodyBuffer);

    if (IotHttpsClient_Init() != IOT_HTTPS_OK)
    {
        printf("iot http client init failed\n");
    }
    printf("IotHttpsClient_Init done\n");

    syncReqInfo.pBody = (uint8_t *)http;
    syncReqInfo.bodyLen = strlen(http);

    reqInfo.pHost = IOT_TEST_HTTPS_SERVER_HOST_NAME;
    reqInfo.hostLen = sizeof(IOT_TEST_HTTPS_SERVER_HOST_NAME) - 1;
    reqInfo.method = method;
    reqInfo.isNonPersistent = true;
    reqInfo.userBuffer.pBuffer = _pReqUserBuffer;
    reqInfo.userBuffer.bufferLen = sizeof(_pReqUserBuffer);
    reqInfo.isAsync = false;
    reqInfo.u.pSyncInfo = &syncReqInfo;

    respInfo.userBuffer.pBuffer = _pRespUserBuffer;
    respInfo.userBuffer.bufferLen = sizeof(_pRespUserBuffer);
    respInfo.pSyncInfo = &syncRespInfo;

    reqInfo.pPath = HTTPS_TEST_POST_REQUEST_PATH;
    reqInfo.pathLen = sizeof(HTTPS_TEST_POST_REQUEST_PATH) - 1;

    /* Set the response configurations. */
    respConfig.userBuffer.pBuffer = _pRespUserBuffer;
    respConfig.userBuffer.bufferLen = sizeof(_pRespUserBuffer);
    respConfig.pSyncInfo = &respSyncInfo;

    /* Initialize the request. */
    returnCode = IotHttpsClient_InitializeRequest(&reqHandle, &reqInfo);

    if (returnCode == IOT_HTTPS_OK)
    {
        printf("IotHttpsClient_InitializeRequest is connected\n");
    }

    _connInfo.pNetworkInterface = networkInterface;

    returnCode = IotHttpsClient_Connect(&connHandle, &_connInfo);
    if (returnCode == IOT_HTTPS_OK)
    {
        printf("IotHttpsClient_Connect is connected\n");
    }
    if (connHandle)
    {
        printf("conn handle is not NULL\n");
    }

    returnCode = IotHttpsClient_SendSync(connHandle, reqHandle, &respHandle, &respConfig, IOT_DEMO_HTTPS_SYNC_TIMEOUT_MS);
    if (returnCode == IOT_HTTPS_OK)
    {
        printf("Successfully Got  the Response\n");
    }
    
    printf("res body : %s\n", _pRespBodyBuffer);
    returnCode = IotHttpsClient_ReadResponseStatus(respHandle, &respStatus);
    if (returnCode != IOT_HTTPS_OK)
    {
        printf("SUCEESSFULLY GOT STATUS\n");
    }
    printf("respstatus %d \n\n", respStatus);

    if (connHandle != NULL)
    {
        IotHttpsClient_Disconnect(connHandle);
    }
    printf("HTTP disconnect done\n");
    /* Clean up the library because we are done using it. */
    IotHttpsClient_Cleanup();
}

Error log
1 1108 [iot_thread] ERROR: Private key not found. 
2 1108 [iot_thread] TLS_Connect fail (0x7d4, *************.amazonaws.com)
3 1109 [iot_thread] [ERROR][NET][11090] Failed to establish new connection. Socket status: -1.
4 1110 [iot_thread] [ERROR][HTTPS Client][11100] Failed to connect to the server at ******************.amazonaws.com on port 443 with error: 4
5 1110 [iot_thread] [ERROR][HTTPS Client][11100] Error in IotHttpsClient_Connect(). Error code 203.

Please let me know your suggestions.

Thanks 
Surya

lundinc2, 2021-01-29:
My apologies @surya-teja-p there was some objects lingering in my workspace. I did not intend to mislead. I will post a patch shortly.

Thanks,

Carl

surya-teja-p, 2021-01-29:
Hi @lundinc2 

That will be lot helpful 

Thanks 
Surya

lundinc2, 2021-02-02:
Hi Surya,

I have pushed a commit to my branch that should address this issue. Can you please see if it resolves your issue? I have attached the patch as a txt file as well, due to GitHub not support patch files, you will have to modify the file extension to be a patch type.

Thanks,
[basic_tls.txt](https://github.com/aws/amazon-freertos/files/5907936/basic_tls.txt)

Carl

surya-teja-p, 2021-02-04:
Hi @lundinc2 

Thanks for your help, the issue got solved the code is working as required.

Thanks 
Surya

closed 2021-02-04 by surya-teja-p