aws/amazon-freertos #2957: Cmake Error: Board is not supported: vendor.board ESP-IDF

walkershanks, 2021-01-28:
**Describe the bug**
When following the ESP32-DevkitC getting started guide (https://docs.aws.amazon.com/freertos/latest/userguide/getting_started_espressif.html), CMake fails when I run idf.py build from the amazon-freertos directory. 

**System information**
- Espressif ESP32 DevkitC
- Linux
- Version of FreeRTOS : 202012.00-242-g441e02e37
- Project: MQTT demo

**Expected behavior**
As stated on in the getting started guide, I would expect running this command to build the project without errors and produce .bin files for the firmware. 

**Screenshots or console output**
```walkershanks@walker:~/Documents/PlantBusiness/ESP32-test/amazon-freertos$ idf.py build
Executing action: all (aliases: build)
Running cmake in directory /home/walkershanks/Documents/PlantBusiness/ESP32-test/amazon-freertos/build
Executing "cmake -G Ninja -DPYTHON_DEPS_CHECKED=1 -DESP_PLATFORM=1 -DCCACHE_ENABLE=0 /home/walkershanks/Documents/PlantBusiness/ESP32-test/amazon-freertos"...
-- The C compiler identification is GNU 9.3.0
-- The CXX compiler identification is GNU 9.3.0
-- Check for working C compiler: /bin/cc
-- Check for working C compiler: /bin/cc -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Check for working CXX compiler: /bin/c++
-- Check for working CXX compiler: /bin/c++ -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Found Git: /bin/git (found version "2.25.1") 
-- Submodule update
Skipping submodule '../../../libraries/abstractions/backoff_algorithm/'
Skipping submodule '../../../../libraries/abstractions/pkcs11/corePKCS11/'
Skipping submodule '../../../libraries/abstractions/pkcs11/corePKCS11/'
Skipping submodule '../../../libraries/abstractions/pkcs11/corePKCS11/'
Skipping submodule '../../../libraries/abstractions/pkcs11/corePKCS11/'
Skipping submodule '../../../libraries/coreHTTP/'
Skipping submodule '../../../libraries/coreHTTP/'
Skipping submodule '../../../libraries/coreHTTP/'
Skipping submodule '../../../libraries/coreJSON/'
Skipping submodule '../../../libraries/coreJSON/'
Skipping submodule '../../../libraries/coreJSON/'
Skipping submodule '../../../libraries/coreMQTT/'
Skipping submodule '../../../libraries/coreMQTT/'
Skipping submodule '../../../libraries/coreMQTT/'
Skipping submodule '../../../libraries/device_defender_for_aws/'
Skipping submodule '../../../libraries/device_defender_for_aws/'
Skipping submodule '../../../libraries/device_defender_for_aws/'
Skipping submodule '../../../libraries/device_shadow_for_aws/'
Skipping submodule '../../../libraries/device_shadow_for_aws/'
Skipping submodule '../../../libraries/device_shadow_for_aws/'
Skipping submodule '../../libraries/freertos_plus/standard/freertos_plus_tcp/'
Skipping submodule '../../libraries/freertos_plus/standard/freertos_plus_tcp/'
Skipping submodule '../../libraries/freertos_plus/standard/freertos_plus_tcp/'
Skipping submodule '../../../libraries/jobs_for_aws/'
Skipping submodule '../../../libraries/jobs_for_aws/'
Skipping submodule '../../../libraries/jobs_for_aws/'
CMake Error at CMakeLists.txt:53 (message):
  Board is not supported: vendor.board

-- Configuring incomplete, errors occurred!
See also "/home/walkershanks/Documents/PlantBusiness/ESP32-test/amazon-freertos/build/CMakeFiles/CMakeOutput.log".
cmake failed with exit code 1
```

**To reproduce**
Steps to reproduce the behavior:

1. Follow getting started guide for Espressif ESP32 DevKitC
2. Try building project using `idf.py build` 

**Additional context**
Using Python 3.8
There are a couple things that I think may be contributing and that confuse me: 
1. The getting started guide does not mention installing and exporting the tools inside the vendors/espressif/esp-idf folder but the only way I was able to get `idf.py build' to run properly at all was to run both `install.sh' and 'export.sh'. 
2. The "Standard Setup of Toolchain for Linux" page that is linked by the getting started guide says to install the prerequisites using `sudo apt-get install git wget flex bison gperf python python-pip python-setuptools cmake ninja-build ccache libffi-dev libssl-dev dfu-util'. However, to get this to run I have to change `python` to `python3`.
3. I actually had this demo working a few days ago using a downloaded version of FreeRTOS from the FreeRTOS console. (I think it uses version 4.0 or 4.1 of the ESP-IDF and an older version of the xtensa toolchain), but I noticed the getting started guide has just been updated to reflect the newest changes to ESP-IDF so I decided to redo running the demo. 

Thank you! Any help will be greatly appreciated :)

aggarw13, 2021-01-28:
Hi @walkershanks, thank you for reporting the problem. 

We recently upgraded the ESP-IDF version to v4.2 which requires slightly different build and flash steps. We are working to update the [Getting Started Guide](https://docs.aws.amazon.com/freertos/latest/userguide/getting_started_espressif.html) to reflect the instructions required for using the new ESP-IDF version.

For the issue you have reported, the problem is caused by an incomplete cmake command that is issued by the `idf.py` tool when running the `idf.py build` command. 
To resolve the issue, please run the following CMake configuration command from the repository root directory before running the `idf.py build` command:
`-DVENDOR=espressif -DBOARD=esp32_wrover_kit -DCOMPILER=xtensa-esp32 -S . -B build -GNinja` 
With the above command, the vendor and board configurations are passed which is missed by the `idf.py build` command.
 
There are more differences from the current version of the Getting Started Guide for building and flashing with the ESP-IDFv4.2 SDK. I have summarized them [in this reply of another issue post](https://github.com/aws/amazon-freertos/issues/2959#issuecomment-769305345). To re-iterate the information in the post, make sure that:
* You have installed `xtensa-esp32-elf-gcc 8.2.0` as the new IDF version does not support the older `xtensa-esp32-elf-gcc 5.2.0` version.
* Your environment contains the required python dependencies for the new IDF version by running `python -m pip install -r ./vendors/espressif/esp-idf/requirements.txt` command.
* You are using `build` as the build directory as the `idf.py` tool is not supporting the `-B build-directory` flag in the recent version.

>The getting started guide does not mention installing and exporting the tools inside the vendors/espressif/esp-idf folder but the only way I was able to get idf.py build' to run properly at all was to run both install.sh' and 'export.sh'.
 
Thank you for pointing it out. We will look to incorporate it in the Getting Started Guide document.

> The "Standard Setup of Toolchain for Linux" page that is linked by the getting started guide says to install the prerequisites using sudo apt-get install git wget flex bison gperf python python-pip [...]

As we have upgraded to using the ESP-IDF v4.2 version, we will update the link to point to https://docs.espressif.com/projects/esp-idf/en/v4.2/esp32/get-started/linux-setup.html which mentions about using Python3.

walkershanks, 2021-01-28:
Hi @aggarw13 , after running the cmake command you provided I was able to build the project properly. Thank you for your help!

aggarw13, 2021-01-29:
Glad that your issue has been resolved. 🙂 If you have any further questions, feel free to reopen the issue.

closed 2021-01-29 by aggarw13