aws/amazon-freertos #2959: Related Amazon-freertos

Romit188, 2021-01-28:
./vendors/espressif/esp-idf/tools/idf.py erase_flash flash monitor -p /dev/ttyUSB0 -B Build1
Usage: ./vendors/espressif/esp-idf/tools/idf.py monitor [OPTIONS]
Try './vendors/espressif/esp-idf/tools/idf.py monitor --help' for help.

Error: Invalid value for '--monitor-baud' / '-B': Build1 is not a valid integer

How to solve this error?

aggarw13, 2021-01-28:
Hi @Romit188, we recently had an ESP-IDF update to v4.2 (in this [PR](https://github.com/aws/amazon-freertos/pull/2893)) which has slightly changed the flash instructions with the `idf.py` tool.

To flash the Espressif board with the new ESP-IDF SDK, please run the following command from with your build directory (i.e. `Build1`)
```
cd <build-directory>
../vendors/espressif/esp-idf/tools/idf.py erase_flash flash monitor -p /dev/ttyUSB0
```
The `idf.py` tool does not support the `-B` option in v4.2 (which can be seen with the `'./vendors/espressif/esp-idf/tools/idf.py monitor --help` command that lists all the options supported by the tool)

There are 3 other things to be aware of when building with the new ESP-IDF version:
*  ESP-IDF v4.2 release requires updated toolchain (`xtensa-esp32-elf-gcc 8.2.0`) and is not compatible with older toolchain used earlier (i.e. `xtensa-esp32-elf-gcc 5.2.0`)
* The Python package dependencies have changed in ESP-IDF v4.2. Therefore, please make sure to install the dependencies with `python -m pip install -r ./vendors/espressif/esp-idf/requirements.txt` on your system.
* The `idf.py` tool uses **Ninja** as the build tool. Therefore, please make sure to configure cmake with the following command that specifies use of **Ninja** as the build framework

We are actively working on updating our [Getting Started Guide](https://docs.aws.amazon.com/freertos/latest/userguide/getting_started_espressif.html#build-and-run-example-espressif) to reflect these changes, and apologize for the inconvenience caused.

dachalco, 2021-01-29:
@Romit188 

If you are using amazon-freertos with ESP-IDF v4.2, can you please instead try the following:

### Building on Mac/Linux
Go to the repo root directory then run the following to install any remaining dependencies
```bash
cd vendors/espressif/esp-idf
./install.sh
. ./export.sh
cd ../../..
unset IDF_PATH
```

Build the application
```bash
idf.py -DVENDOR=espressif -DBOARD=esp32_wrover_kit -DCOMPILER=xtensa-esp32 build
```

I believe the Windows build instructions remain the same. See the [Getting Started Guide](https://docs.aws.amazon.com/freertos/latest/userguide/getting_started_espressif.html).

### Interfacing ESP32
```bash
idf.py erase_flash
idf.py flash
idf.py monitor
```

You can combine these too
```bash
idf.py erase_flash flash monitor
```

Romit188, 2021-01-29:
Okay thank you aggarw13 and dachalco to support me....

I follow the instructions of both. And when i flash the code in my board i got this type of error.

CMakeFiles/afr_ble_hal.dir/vendors/espressif/esp-idf/components/mbedtls/port/mbedtls_debug.c.obj 
in this file...
In file included from ../libraries/3rdparty/mbedtls/include/mbedtls/platform_time.h:77,
                 from ../libraries/3rdparty/mbedtls/include/mbedtls/platform.h:68,
                 from ../vendors/espressif/esp-idf/components/mbedtls/port/mbedtls_debug.c:18:
/home/ar-cpu-35/esp/esp-idf/components/newlib/platform_include/time.h:28:19: error: unknown type name 'clockid_t'; did you mean '_lock_t'?
 int clock_settime(clockid_t clock_id, const struct timespec *tp);
                   ^~~~~~~~~
                   _lock_t
/home/ar-cpu-35/esp/esp-idf/components/newlib/platform_include/time.h:29:19: error: unknown type name 'clockid_t'; did you mean '_lock_t'?
 int clock_gettime(clockid_t clock_id, struct timespec *tp);
                   ^~~~~~~~~
                   _lock_t
/home/ar-cpu-35/esp/esp-idf/components/newlib/platform_include/time.h:30:18: error: unknown type name 'clockid_t'; did you mean '_lock_t'?
 int clock_getres(clockid_t clock_id, struct timespec *res);
                  ^~~~~~~~~
                  _lock_t
In file included from ../libraries/3rdparty/mbedtls/include/mbedtls/platform.h:68,
                 from ../vendors/espressif/esp-idf/components/mbedtls/port/mbedtls_debug.c:18:
../libraries/3rdparty/mbedtls/include/mbedtls/platform_time.h:78:9: error: unknown type name 'time_t'
 typedef time_t mbedtls_time_t;
         ^~~~~~
so can you help me about this?

Romit188, 2021-01-29:
In file included from ../vendors/espressif/boards/ports/ble/iot_ble_hal_common_gap.c:29:
../vendors/espressif/boards/ports/ble/nimble/iot_ble_hal_common_gap.c: In function 'prvGAPeventHandler':
../vendors/espressif/boards/ports/ble/nimble/iot_ble_hal_common_gap.c:256:14: error: 'BLE_GAP_EVENT_PAIRING_REQUEST' undeclared (first use in this function); did you mean 'BLE_GAP_REPEAT_PAIRING_RETRY'?
         case BLE_GAP_EVENT_PAIRING_REQUEST:
              ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~
              BLE_GAP_REPEAT_PAIRING_RETRY
../vendors/espressif/boards/ports/ble/nimble/iot_ble_hal_common_gap.c:256:14: note: each undeclared identifier is reported only once for each function it appears in
../vendors/espressif/boards/ports/ble/nimble/iot_ble_hal_common_gap.c:260:46: error: 'struct ble_gap_event' has no member named 'pairing_req'
                 rc = ble_gap_conn_find( event->pairing_req.conn_handle, &desc );
                                              ^~
../vendors/espressif/boards/ports/ble/nimble/iot_ble_hal_common_gap.c:268:44: error: 'struct ble_gap_event' has no member named 'pairing_req'
             if( ble_hs_cfg.sm_sc && ( event->pairing_req.authreq & BLE_SM_PAIR_AUTHREQ_SC ) )
                                            ^~
../vendors/espressif/boards/ports/ble/nimble/iot_ble_hal_common_gap.c:270:65: error: 'struct ble_gap_event' has no member named 'pairing_req'
                 if( ble_sm_sc_ioa[ ble_hs_cfg.sm_io_cap ][ event->pairing_req.io_cap ] == BLE_SM_IOACT_NONE )
                                                                 ^~
../vendors/espressif/boards/ports/ble/nimble/iot_ble_hal_common_gap.c:284:31: error: 'struct ble_gap_event' has no member named 'pairing_req'
                 else if( event->pairing_req.max_enc_key_size < BLE_SM_PAIR_KEY_SZ_MAX )
                               ^~
[3/1327] Building C object CMakeFiles/afr_defender.dir/libraries/c_sdk/aws/defender/src/aws_iot_defender_api.c.obj
ninja: build stopped: subcommand failed.
ninja failed with exit code

How to solve this two types of error?

aggarw13, 2021-01-29:
Hi @Romit188, at which step do you face the above errors? 
What is the status of running the build command `idf.py -DVENDOR=espressif -DBOARD=esp32_wrover_kit -DCOMPILER=xtensa-esp32 build`?

Romit188, 2021-01-30:
Hello aggarw13 , the above error is occur at final stage , means i flash the code in the ESP32 module using below command.
./vendors/espressif/esp-idf/tools/idf.py erase_flash flash monitor -p /dev/ttyUSB0 build.
at this time error is occur.

idf.py -DVENDOR=espressif -DBOARD=esp32_wrover_kit -DCOMPILER=xtensa-esp32 build 
And when i run this command this command was perfectly work . And directory is perfectly build. But with one warning , related to IDF_PATH.

Romit188, 2021-01-30:
Hi aggarw13 , i Solved the errors which is mentioned above in comment. But after this i get another type of error was generated.

Running cmake in directory /home/ar-cpu-35/amazon-freertos/build
Executing "cmake -G Ninja -DPYTHON_DEPS_CHECKED=1 -DESP_PLATFORM=1 -DCCACHE_ENABLE=0 /home/ar-cpu-35/amazon-freertos"...
CMake Error: Error: generator : Ninja
Does not match the generator used previously: Unix Makefiles
Either remove the CMakeCache.txt file and CMakeFiles directory or choose a different binary directory.
cmake failed with exit code.

Can you help me to solve this problem?

Romit188, 2021-01-30:
Hi aggarw13 i solved the error which is mentioned above.
After that i have another type of error.

CMake Error at CMakeLists.txt:53 (message):
  Board is not supported: vendor.board

-- Configuring incomplete, errors occurred!
See also "/home/ar-cpu-35/amazon-freertos/build/CMakeFiles/CMakeOutput.log".
cmake failed with exit code 1

How to solve this error , can you help me?

aggarw13, 2021-01-30:
Hi @Romit188, I'm happy that you were able to solve the prior errors. 

>After that i have another type of error.
>
> CMake Error at CMakeLists.txt:53 (message):
Board is not supported: vendor.board
>
> -- Configuring incomplete, errors occurred!
> See also "/home/ar-cpu-35/amazon-freertos/build/CMak
> 

The cause of this error is insufficient or incorrect CMake configuration which can happen when just using the `idf.py build` command. 

To solve the error, I recommend you perform the following steps:
1. Delete the build directory with  `rm -rf ./build`
2. Re-configure Cmake in `build` directory with `cmake -DVENDOR=espressif -DBOARD=esp32_wrover_kit -DCOMPILER=xtensa-esp32 -S . -B build -GNinja`
3. Build the application with `idf.py build` command. (Alternatively, you can combine steps 3. and 4. with `idf.py -DVENDOR=espressif -DBOARD=esp32_wrover_kit -DCOMPILER=xtensa-esp32 -GNinja build`. )
4. Flash board with `./vendors/espressif/esp-idf/tools/idf.py erase_flash flash monitor -p </path/to/serial/port>`. Please replace </path/to/serial/port>` with the serial/USB port present on your system for the ESP board connection.

Hope these help

Romit188, 2021-02-01:
./vendors/espressif/esp-idf/tools/idf.py -DVENDOR=espressif -DBOARD=esp32_wrover_kit -DCOMPILER=xtensa-esp32 -GNinja build
 When i build using this command, i get below error.

/home/ar-cpu-35/esp/esp-idf/components/esp_common/include/esp_idf_version.h: Assembler messages:
/home/ar-cpu-35/esp/esp-idf/components/esp_common/include/esp_idf_version.h:54: Error: unknown opcode or format name 'const'
[2/1202] Performing build step for 'bootloader'
ninja: no work to do.
[3/1202] Building C object esp-idf/xtensa/CMakeFiles/__idf_xtensa.dir/expression_with_stack_xtensa.c.obj
ninja: build stopped: subcommand failed.
ninja failed with exit code 1

Can you help me ?

Romit188, 2021-02-01:
home/ar-cpu-35/esp/xtensa-esp32-elf/xtensa-esp32-elf/sys-include/machine/_default_types.h: Assembler messages:
/home/ar-cpu-35/esp/xtensa-esp32-elf/xtensa-esp32-elf/sys-include/machine/_default_types.h:40: Error: unknown opcode or format name 'typedef'
/home/ar-cpu-35/esp/xtensa-esp32-elf/xtensa-esp32-elf/sys-include/machine/_default_types.h:42: Error: unknown opcode or format name 'typedef'
/home/ar-cpu-35/esp/xtensa-esp32-elf/xtensa-esp32-elf/sys-include/machine/_default_types.h:54: Error: unknown opcode or format name 'typedef'
/home/ar-cpu-35/esp/xtensa-esp32-elf/xtensa-esp32-elf/sys-include/machine/_default_types.h:56: Error: unknown opcode or format name 'typedef'
/home/ar-cpu-35/esp/xtensa-esp32-elf/xtensa-esp32-elf/sys-include/machine/_default_types.h:76: Error: unknown opcode or format name 'typedef'
/home/ar-cpu-35/esp/xtensa-esp32-elf/xtensa-esp32-elf/sys-include/machine/_default_types.h:78: Error: unknown opcode or format name 'typedef'
/home/ar-cpu-35/esp/xtensa-esp32-elf/xtensa-esp32-elf/sys-include/machine/_default_types.h:102: Error: unknown opcode or format name 'typedef'
/home/ar-cpu-35/esp/xtensa-esp32-elf/xtensa-esp32-elf/sys-include/machine/_default_types.h:104: Error: unknown opcode or format name 'typedef'
/home/ar-cpu-35/esp/xtensa-esp32-elf/xtensa-esp32-elf/sys-include/machine/_default_types.h:133: Error: unknown opcode or format name 'typedef'
/home/ar-cpu-35/esp/xtensa-esp32-elf/xtensa-esp32-elf/sys-include/machine/_default_types.h:135: Error: unknown opcode or format name 'typedef'
/home/ar-cpu-35/esp/xtensa-esp32-elf/xtensa-esp32-elf/sys-include/machine/_default_types.h:159: Error: unknown opcode or format name 'typedef'
/home/ar-cpu-35/esp/xtensa-esp32-elf/xtensa-esp32-elf/sys-include/machine/_default_types.h:161: Error: unknown opcode or format name 'typedef'
/home/ar-cpu-35/esp/xtensa-esp32-elf/xtensa-esp32-elf/sys-include/machine/_default_types.h:181: Error: unknown opcode or format name 'typedef'
/home/ar-cpu-35/esp/xtensa-esp32-elf/xtensa-esp32-elf/sys-include/machine/_default_types.h:183: Error: unknown opcode or format name 'typedef'
/home/ar-cpu-35/esp/xtensa-esp32-elf/xtensa-esp32-elf/sys-include/machine/_default_types.h:199: Error: unknown opcode or format name 'typedef'
/home/ar-cpu-35/esp/xtensa-esp32-elf/xtensa-esp32-elf/sys-include/machine/_default_types.h:201: Error: unknown opcode or format name 'typedef'
/home/ar-cpu-35/esp/xtensa-esp32-elf/xtensa-esp32-elf/sys-include/machine/_default_types.h:213: Error: unknown opcode or format name 'typedef'
/home/ar-cpu-35/esp/xtensa-esp32-elf/xtensa-esp32-elf/sys-include/machine/_default_types.h:221: Error: unknown opcode or format name 'typedef'
/home/ar-cpu-35/esp/xtensa-esp32-elf/xtensa-esp32-elf/sys-include/machine/_default_types.h:229: Error: unknown opcode or format name 'typedef'
/home/ar-cpu-35/esp/xtensa-esp32-elf/xtensa-esp32-elf/sys-include/machine/_default_types.h:231: Error: unknown opcode or format name 'typedef'
/home/ar-cpu-35/esp/xtensa-esp32-elf/xtensa-esp32-elf/sys-include/sys/_stdint.h:20: Error: unknown opcode or format name 'typedef'
/home/ar-cpu-35/esp/xtensa-esp32-elf/xtensa-esp32-elf/sys-include/sys/_stdint.h:24: Error: unknown opcode or format name 'typedef'
/home/ar-cpu-35/esp/xtensa-esp32-elf/xtensa-esp32-elf/sys-include/sys/_stdint.h:32: Error: unknown opcode or format name 'typedef'
/home/ar-cpu-35/esp/xtensa-esp32-elf/xtensa-esp32-elf/sys-include/sys/_stdint.h:36: Error: unknown opcode or format name 'typedef'
/home/ar-cpu-35/esp/xtensa-esp32-elf/xtensa-esp32-elf/sys-include/sys/_stdint.h:44: Error: unknown opcode or format name 'typedef'
/home/ar-cpu-35/esp/xtensa-esp32-elf/xtensa-esp32-elf/sys-include/sys/_stdint.h:48: Error: unknown opcode or format name 'typedef'
/home/ar-cpu-35/esp/xtensa-esp32-elf/xtensa-esp32-elf/sys-include/sys/_stdint.h:56: Error: unknown opcode or format name 'typedef'
/home/ar-cpu-35/esp/xtensa-esp32-elf/xtensa-esp32-elf/sys-include/sys/_stdint.h:60: Error: unknown opcode or format name 'typedef'
/home/ar-cpu-35/esp/xtensa-esp32-elf/xtensa-esp32-elf/sys-include/sys/_stdint.h:67: Error: unknown opcode or format name 'typedef'
/home/ar-cpu-35/esp/xtensa-esp32-elf/xtensa-esp32-elf/sys-include/sys/_stdint.h:72: Error: unknown opcode or format name 'typedef'
/home/ar-cpu-35/esp/xtensa-esp32-elf/xtensa-esp32-elf/sys-include/sys/_stdint.h:77: Error: unknown opcode or format name 'typedef'
/home/ar-cpu-35/esp/xtensa-esp32-elf/xtensa-esp32-elf/sys-include/sys/_stdint.h:82: Error: unknown opcode or format name 'typedef'
/home/ar-cpu-35/esp/xtensa-esp32-elf/xtensa-esp32-elf/sys-include/stdint.h:21: Error: unknown opcode or format name 'typedef'
/home/ar-cpu-35/esp/xtensa-esp32-elf/xtensa-esp32-elf/sys-include/stdint.h:22: Error: unknown opcode or format name 'typedef'
/home/ar-cpu-35/esp/xtensa-esp32-elf/xtensa-esp32-elf/sys-include/stdint.h:27: Error: unknown opcode or format name 'typedef'
/home/ar-cpu-35/esp/xtensa-esp32-elf/xtensa-esp32-elf/sys-include/stdint.h:28: Error: unknown opcode or format name 'typedef'
/home/ar-cpu-35/esp/xtensa-esp32-elf/xtensa-esp32-elf/sys-include/stdint.h:33: Error: unknown opcode or format name 'typedef'
/home/ar-cpu-35/esp/xtensa-esp32-elf/xtensa-esp32-elf/sys-include/stdint.h:34: Error: unknown opcode or format name 'typedef'
/home/ar-cpu-35/esp/xtensa-esp32-elf/xtensa-esp32-elf/sys-include/stdint.h:39: Error: unknown opcode or format name 'typedef'
[... cut ...]
/home/ar-cpu-35/esp/xtensa-esp32-elf/xtensa-esp32-elf/sys-include/stdint.h:82: Error: unknown opcode or format name 'typedef'
/home/ar-cpu-35/esp/esp-idf/components/esp_common/include/esp_idf_version.h:53: Error: unknown opcode or format name 'const'
[443/1651] Building C object esp-idf/xtensa/CMakeFiles/__idf_xtensa.dir/debug_helpers.c.obj
ninja: build stopped: subcommand failed.
ninja failed with exit code 1

I found the solutions 
like add the stdio.h file as well as string.h file but it was not supported. 
So can you help me about this?

Romit188, 2021-02-01:
Hello,
home/ar-cpu-35/esp/esp-idf/components/esp_common/include/esp_idf_version.h:54: Error: unknown opcode or format name 'const'
I can not solve this error , Error is related to const keyword , which is in <stdint.h> header file , i add this file but it is not work.
so i aomment this part.
After that another type of error was generated, shown below.

/home/ar-cpu-35/amazon-freertos/build/../vendors/espressif/boards/ports/common_io/iot_uart.c:233: undefined reference to `uart_register_callback_with_isr'
/home/ar-cpu-35/esp/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld: CMakeFiles/aws_demos.dir/vendors/espressif/boards/ports/common_io/iot_uart.c.obj: in function `iot_uart_ioctl':
/home/ar-cpu-35/amazon-freertos/build/../vendors/espressif/boards/ports/common_io/iot_uart.c:341: undefined reference to `uart_register_callback_with_isr'
collect2: error: ld returned 1 exit status
ninja: build stopped: subcommand failed.
ninja failed with exit code 1

How to solve this error?

aggarw13, 2021-02-01:
Hi @Romit188, it seems like the ESP-IDF path being used in your build (i.e. `home/ar-cpu-35/esp/esp-idf/components/esp_common/include/esp_idf_version.h`) does not point to the ESP-IDF SDK within the `amazon-freertos` repository. It should be using the `amazon-freertos/vendors/espressif/esp-idf` path that is present within the repository folder for building the SDK.

If you have an `IDF_PATH` environment variable set in your system (that you can verify with `echo $IDF_PATH`), please unset the path with `unset IDF_PATH`.
After that, please follow the steps I mentioned in the previous comment [here](https://github.com/aws/amazon-freertos/issues/2959#issuecomment-770299512).

Romit188, 2021-02-02:
Hello aggarw13
Error is related to E (730) uart: "uart_register_callback_with_isr" function which is not define in this file.
Can you tell me what is the definition of this function?

Romit188, 2021-02-02:
I can not find out the definition of "uart_register_callback_with_isr" function , So can you help me about this what is the definition of this function?

aggarw13, 2021-02-02:
Hi @Romit188, as I mentioned in my [previous post](https://github.com/aws/amazon-freertos/issues/2959#issuecomment-771148031), it seems like your environment is building the ESP-IDF SDK outside of the IDF SDK present in the `amazon-freertos/vendors/espressif/esp-idf` path. 
Can you please confirm whether :
* You have attempted to follow the steps I mentioned [here](https://github.com/aws/amazon-freertos/issues/2959#issuecomment-770299512)?
* You are able to build the ESP-IDF SDK present in the `vendors/espressif/esp-idf` path? 

**If the steps I provided in the above link don't work**, I recommend that you start afresh (by re-cloning the `amazon-freertos` repository) and reach the successful build state you were able to reach with the `idf.py -DVENDOR=espressif -DBOARD=esp32_wrover_kit -DCOMPILER=xtensa-esp32 build` command as you mentioned in [your prior post](https://github.com/aws/amazon-freertos/issues/2959#issuecomment-770162591). After this command, you can attempt to flash the board with the `make -C build flash` command.

Romit188, 2021-02-03:
Yes aggarw12 i performed the all steps which you had mentioned in the previous post.
yes i able to build the esp-idf.

But i solved the problem related to the flash in to the esp32. So perfectly done this steps.
So Thank for your time and support.

But one last question i have asked that, If you know about this pls tell me.

How to interface the temp sensor with the ESP32? Means in which file i should write the code of freertos , as well as which header file of GPIO pin i should use?

closed 2021-02-06 by Romit188