aws/amazon-freertos #2961: vDevModeKeyProvisioning

surya-teja-p, 2021-01-30:
Hi 

I want to know at which memory location the privatekey and client certificate are written and how to verify if the private key and client certificate are being written ?

Thanks 
Surya

lundinc2, 2021-02-01:
Hi Surya,

I am not sure which ESP32 you are using, but I am assuming you are using the ESP-32 devkitc or ESP-32 wrover kit, and not the ESP-32 boards that have a secure element on board. The answer is different if that is the case.

The private key and public key are written to flash using NVS and placed in the `storage` partition. You can check if they are written by using the PKCS #11 interface to "find" the objects. See the example here: https://github.com/aws/amazon-freertos/blob/441e02e3776833b436c9d69859d28a01bd29fa84/libraries/abstractions/pkcs11/test/iot_test_pkcs11.c#L597.

Some documentation that would be helpful:
* https://docs.aws.amazon.com/embedded-csdk/latest/lib-ref/libraries/standard/corePKCS11/docs/doxygen/output/html/core__pkcs11_8c.html
* https://docs.aws.amazon.com/embedded-csdk/latest/lib-ref/libraries/standard/corePKCS11/docs/doxygen/output/html/core__pkcs11__pal_8h.html
* https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/storage/nvs_flash.html
* https://github.com/aws/amazon-freertos/blob/4e8219e0923d459900924ff760ac492224b131fb/vendors/espressif/boards/esp32/aws_demos/partition-table.csv

Thanks,

Carl

leegeth, 2021-02-20:
@surya-teja-p,

Did @lundinc2's answer help with your question? Are you looking for any further help from us?

lundinc2, 2021-03-01:
Hello @surya-teja-p,

I am closing this as a stale issue. Please re-open this if you have any further questions!

closed 2021-03-01 by lundinc2