aws/amazon-freertos #2968: How to store OTA certificate without using adding it manually in "aws_ota_codesigner_certificate.h" file

MrMelon1232, 2021-02-04:
I am currently using NVS to store the OTA certificate. I would like to retrieve it via NVS by using a function. Is there any way to override the variable ```static const char signingcredentialSIGNING_CERTIFICATE_PEM[]``` and assign it to the return value of the function or parse an argument when initializing OTA to have custom certificates ?

Any other approach and suggestions are appreciated as well.

yourslab, 2021-02-05:
Have you considered provisioning the OTA certificate through PKCS #11? There is a `pkcs11configOTA_SUPPORTED` macro in `core_pkcs11_config.h` that you can use for this.

leegeth, 2021-02-20:
@MrMelon1232,

Could you try the provisioning that @yourslab mentioned? Are you looking for any further help from us?

MrMelon1232, 2021-02-23:
Hi @leegeth @yourslab, I have found a workaround for this issue and will therefore close it.

closed 2021-02-23 by MrMelon1232