aws/amazon-freertos #2971: Related to Interface the Temperature sensor with ESP32 module.

Romit188, 2021-02-05:
How to interface Temperature sensor with ESP32 module , Means where we write to code for this, and which GPIO header file we can use ? And if anyone done this please share the code in FreeRTOS.

My system - Linux 18.04
OS of ESP32-WROOM-32 D - FreeRTOS.

yourslab, 2021-02-05:
The demos directory provides a set of examples that you can use as a base to build your application. To flash your ESP32 with a demo, please refer to the guide below:
https://docs.aws.amazon.com/freertos/latest/userguide/getting_started_espressif.html

The header file you would be using is provided under the Common I/O interface as `iot_board_gpio.h`.

Romit188, 2021-02-06:
Okay , I understand perfectly.
Thank you for your valuable response.

yourslab, 2021-02-06:
That's great. I will close this issue for now, but just feel free to open again if you have further questions.

closed 2021-02-06 by yourslab