aws/amazon-freertos #2972: Related to the flash the code in ESP32.

Romit188, 2021-02-05:
In aws_demo_config.h file i enabled the shadow demo.
And then i flash the code in to the ESP32 module i get below error, when i flash the code.
assertion "globalSubscribePacketIdentifier == usPacketIdentifier" failed: file "../demos/common/mqtt_demo_helpers/mqtt_demo_helpers.c", line 490, function: vHandleOtherIncomingPacket

Can anyone help me about this?How to solve this error?

yourslab, 2021-02-05:
It's possible that you are receiving the SUBACK for an earlier subscription attempt rather than what the current subscription attempt is expecting. This can happen depending on the client's latency with AWS IoT Core. Can you try increasing `mqttexamplePROCESS_LOOP_TIMEOUT_MS` in `demos\common\mqtt_demo_helpers\mqtt_demo_helpers.c` to fix this problem?

Romit188, 2021-02-06:
Now, It is okk , i solved the error successful,
Thank you for your valuable report.

yourslab, 2021-02-06:
That's great. I will close this issue for now, but just feel free to open again if you have further questions.

closed 2021-02-06 by yourslab