aws/amazon-freertos #2975: RISC-V - Kendryte K210 Support

FL140, 2021-02-06:
Since the RISC-V platform is a rising star, support for chips available in the real world are important.

One of the available chips is the Kendryte K210 it is quite powerful for its price/size and even has an AI unit integrated. There are also modules already available which integrate the K210 on a module which can be handled more easily (e.g. https://www.seeedstudio.com/Sipeed-MAIX-I-module-WiFi-version-1st-RISC-V-64-AI-Module-K210-inside-p-3206.html). 

The good thing is there already exists an initial port to FreeRTOS available at https://github.com/kendryte/kendryte-freertos-sdk

But I get the feeling the port is not complete and I think it would be a good choice to officially add this platform here to Amazon FreeRTOS.

Anyone interested in working together on this?
Preferable someone who has already experience with porting platforms to FreeRTOS?

I already got a bunch of the modules and kits available on the market with and K210 to try things out and test stuff (e.g. https://www.seeedstudio.com/Sipeed-Maixduino-Kit-for-RISC-V-AI-IoT-p-4047.html).

yourslab, 2021-02-06:
Normally, the boards that end up in amazon-freertos are qualified under the [AWS Partner Device Catalog](https://devices.amazonaws.com/search?page=1&sv=freertos). However, if you'd like to implement and share a port for FreeRTOS, you are always welcome to submit a pull request under the [FreeRTOS/FreeRTOS](https://github.com/FreeRTOS/FreeRTOS/tree/master/FreeRTOS/Demo) repository.

FL140, 2021-02-06:
@yourslab Is a complete port necessary to be accepted or is a basic version enough to get started, and then add modules step by step? So the procedure would be to follow https://github.com/awsdocs/aws-freertos-docs/blob/master/doc_source/portingguide/index.md correct?

yourslab, 2021-02-06:
A basic port should be enough so that it ends up in https://github.com/FreeRTOS/FreeRTOS/tree/master/FreeRTOS/Demo

The porting guide you sent me is for vendors to qualify their boards with Amazon.

yourslab, 2021-02-06:
For the basic kernel port, you can follow the guide here:
https://www.freertos.org/FreeRTOS-porting-guide.html

FL140, 2021-02-06:
The port you linked is for the standard FreeRTOS tough. This was already done (at least partially) in  https://github.com/kendryte/kendryte-freertos-sdk I need a port to the Amazon FreeRTOS variant.

So do I need to get the already existing port into the standard FreeRTOS port first or can I just focus on the Amazon FreeRTOS port?

yourslab, 2021-02-06:
Yes, you cannot have the Amazon FreeRTOS port without doing the standard port first. An Amazon FreeRTOS port is normally done by the vendor and will require a lot more work and testing than the standard port.

yourslab, 2021-02-06:
Also keep in mind that the work required for making a standard port is a subset of the work required for making an Amazon FreeRTOS port

FL140, 2021-02-07:
Yes, correct I had a deeper look into the standard FreeRTOS porting guide and the existing K210 FreeRTOS SDK after our conversation here and noticed it myself that the standard FreeRTOS port from FreeRTOS/FreeRTOS-Kernel is linked to from Amazon FreeRTOS aws/amazon-freertos/freertos_kernel. 

So the next steps are:
* getting the existing K210 FreeRTOS SDK implementation
* moving all the platform depended files `heap_4.c`, `port.c` and `portmacro.h` to there correct place
 `FreeRTOS/FreeRTOS-Kernel/FreeRTOS-Kernel/portable/GCC/platform_name
 * check if there are impacts because of different FreeRTOS versions used in the K210 FreeRTOS SDK and the current master branch
 * implement the basic example
 
 Am I correct so far?
 
 When I looked into this a few questions came up:
 1. The files in the K210 FreeRTOS SDK implementation use the Apache License, Version 2.0. Is this an issue as the FreeRTOS-Kernel repository uses the MIT License? I think not, since both are permissive licenses and work together so if it is not a problem to keep the original copyright information in the files this shouldn't be a problem. Am I correct on this?
 2. Normally I would use `RISC-V_K210` or just `K210` (are there any preferences on this?) for `platform_name` but I noticed there is already a `RISC-V` directory which hosts the common part of "RISC-V port is split between the the code that is common across all currently supported RISC-V chips". But this generates a few questions.
 2a. K210 is a RV64 platform and while the directory name is RISC-V there are a bunch of references to a `FreeRTOS\Source\portable\GCC\RISC-V-RV32` directory which simply does not exist. So I presume this directory was once called RISC-V-RV32 and the documentation just didn't get updated!? Which brings up the next question.
 2b. If the directory was once called RISC-V-RV32 is the code in it still just for 32-bit platforms? A quick look into `FreeRTOS-Kernel/portable/GCC/RISC-V/portASM.S` revealed that there are references to 64-bit platforms (`__riscv_xlen == 64`). So from my understanding this should be OK, correct? But then again a `FreeRTOS-Kernel/portable/GCC/RISC-V/chip_specific_extensions/RV64I_CLINT_no_extensions/freertos_risc_v_chip_specific_extensions.h` or something more special for the K210 (I need to check this out.) is needed, correct? 
 
Sorry the list got long, but there are a bunch of different things to look at and I think it is better to discuss this prior to porting, and getting it right the first time than having a discussion afterwards in the code review when submitting the pull request. ;-)

yourslab, 2021-02-07:
The next steps you provided look to be in the right direction. It is likely though that changes need to be made to the existing port you linked me since it is using a much older version of the kernel.

I also personally don't think that the Apache License will be a problem. However, you had other great questions that will require more feedback from the kernel team. I will reach out to you again once we have clear answers to the points you brought up.

FL140, 2021-02-07:
THX, great!

Yes, these where the possible needed modifications I ment with "check if there are impacts because of different FreeRTOS versions used in the K210 FreeRTOS SDK and the current master branch" any particular major things that come into your mind since you know the project much better than me? Or just start out with a diff and look where it goes? ;-)

yourslab, 2021-02-07:
I would suggest taking a look at the changelog here from V10.0.1 onwards:
https://www.freertos.org/History.txt
For example, you will see that in V10.2.1, the RISC-V port now automatically switches between 32-bit and 64-bit cores. You probably won't have to make any major changes but just have to keep it up to speed with the existing RISC-V ports.

One thing you will have to do is implement 'ParTest.c`. I noticed that a lot of the drivers (including GPIO) were ported from C to CPP in the Kendryte FreeRTOS SDK:
https://github.com/kendryte/kendryte-freertos-sdk/commit/bba479fb45603d9344c73fb588506a683500ff80

I'm not quite sure why this was done. If you take a look at the standalone SDK, the drivers are still written in C:
https://github.com/kendryte/kendryte-standalone-sdk/tree/develop/lib/drivers

For now, I believe we only allow ports that are written in C. It would be interesting to know why Kendryte decided to make the switch to CPP in the deprecated FreeRTOS SDK.

yourslab, 2021-02-08:
Also, I think you will find more people interested to help you with porting this board in the FreeRTOS forums
https://forums.freertos.org

dan4thewin, 2021-03-04:
I'll close this now and hope to hear more about it in the FreeRTOS forums.

closed 2021-03-04 by dan4thewin