aws/amazon-freertos #2976: Cellular Library Integration

FL140, 2021-02-06:
There already exists a cellular library for FreeFRTOS (https://github.com/FreeRTOS/Lab-Project-FreeRTOS-Cellular-Library) and also a new branch (https://github.com/aws/amazon-freertos/tree/feature/cellular) has already been created by others to merge the library here into Amazon FreeRTOS.

There already exists a porting guide in the documentation (https://github.com/awsdocs/aws-freertos-docs/blob/master/doc_source/portingguide/freertos-porting-cellular.md) and it looks like the branch has followed this porting guide.

Unfortunately this branch is getting out of sync with master, so can this branch please be merged into master? Or is there anything that is needed to be done?

yourslab, 2021-02-07:
I have reached out to the cellular team, and they will be able to provide more insight into the state of the branch.

DanielYEHsieh, 2021-02-07:
Hi @FL140 ,
Thanks for your interest, so far the cellular library is still in preview stage:
https://aws.amazon.com/about-aws/whats-new/2020/12/freertos-adds-cellular-lte-m-library-support-cellular-iot-based-applications/

As you've mentioned, we provide porting guide with steps how you port your own integration in this stage.
We are maintaining the cellular libraries here:
https://github.com/FreeRTOS/Lab-Project-FreeRTOS-Cellular-Library
https://github.com/FreeRTOS/Lab-Project-FreeRTOS-Cellular-Demo

And you are right, feature/cellular branch is an example to follow the porting guide.

However, merging cellular branch back to master is still under working process and will take time, we can't give you a straight answer how long it will take, and how it will turn out.

If you have encounter any trouble integrating your own port, you are more than welcome to contact us through mail (dhs@amazon.com) for more details,

Thank you!

FL140, 2021-02-07:
@DanielYEHsieh thank's for the quick reply!

Well I already worked my way through those links and while they are a great source of information the directory structure in the FreeRTOS Labs repositories don't fit the `aws/amazon-freertos` structure this is why I was happy to find the `feature/cellular` branch in the `aws/amazon-freertos` repository which honors the directory structure there. Unfortunately this branch is quite out of sync with master (507 commits behind).

So while I understand this is a new feature and there will be changes coming up I just don't understand why
1. The directory structure of the FreeRTOS Labs repository and the `aws/amazon-freertos` repository are that different.
2. Why the cellular library isn't some sort of submodule in the `aws/amazon-freertos` repository like the `FreeRTOS/FreeRTOS-Kernel` repository.
3. Why the `feature/cellular` branch hasn't been merged into master jet (or a staging branch which would fit quite well for an unfinished feature).
4. So this means the improvements need to be coded in two places (FreeRTOS Labs and Amazon FreeRTOS repositories). 
5. The repository of interest will be `aws/amazon-freertos` but the primary coding repository is somewhere else.

But probably I just don't see it, so it would be great if you could provide me with some information why this is setup this way to understand the process flow here?

DanielYEHsieh, 2021-02-08:
@FL140 Great to know that you've got it worked, and we appreciate your feedbacks!

I would love to take a look your port if you have a repo see how you are doing, we do have a separate port based on `202012.00 ` which looks pretty same as `feature/cellular` branch and verified with [IoT Device Tester 4.0.1 ](https://docs.aws.amazon.com/freertos/latest/userguide/dev-test-versions-afr.html)(I can provide it through email).

To answer your questions, replying inline:

1. The directory structure of the FreeRTOS Labs repository and the aws/amazon-freertos repository are that different.

[Daniel] Yes, since Cellular library is still in the FreeRTOS Labs, it will need to go through a few process to graduate from lab project to a sub-module library or a separate library. There are remaining works to get the library more mature, it would be great if you can share your use case with us to collect data points how users are adopting it. Today, the maintenance will be focused on  the lab repos
https://github.com/FreeRTOS/Lab-Project-FreeRTOS-Cellular-Library
https://github.com/FreeRTOS/Lab-Project-FreeRTOS-Cellular-Demo

2. Why the cellular library isn't some sort of submodule in the aws/amazon-freertos repository, like e.g. the FreeRTOS/FreeRTOS-Kernel repository.

[Daniel] Same as #1.

3. Why the feature/cellular branch hasn't been merged into master jet (or a staging branch which would fit quite well for an unfinished feature) and got pretty outdated that way.

[Daniel] As mentioned previously we do have ports based on 202012.00 release, integrated with Cellular Library.

4. The repository of interest will be aws/amazon-freertos but the primary coding repository is somewhere else.

[Daniel] Same as #1.

5. So this means the improvements need to be coded in two places (FreeRTOS Labs and Amazon FreeRTOS repositories).

[Daniel] Same as #1.

mingyue86010, 2021-02-15:
It looks the costomer's question has been answered. Close this ticket. If having any further question, please feel free to re-open this ticket or open a new ticket. Thanks.

closed 2021-02-15 by mingyue86010