aws/amazon-freertos #2977: FreeRTOS with ESP32 : what should be in my main program to publish my own custom readings to aws iot.

Johnnyjax, 2021-02-06:
I've setup amazon freertos and esp-idf and have been able to successfully run the demo program. I've also gone through the "Using FreeRTOS in your own CMake project for ESP32" in the getting started guide and I've set up the directory structure for a custom project with help from this github project - https://github.com/tgsong/amazon-freertos-examples.git. I ran it and it successfully ran the demo project. My plan is to get various sensor measurements and then publish the data to aws iot. I already know how to get the sensor data and the libraries to call. How do i go about publishing the data without using the demo projects. Do I have to use the demo projects? What should be in my main.c file besides the sensor reading program?

yourslab, 2021-02-06:
That's awesome. You don't necessarily have to use the demo projects, but you can use them as an example and copy snippets of code that suit your project. I would recommend using the [coreMQTT Mutual Auth Demo](https://github.com/aws/amazon-freertos/blob/master/demos/coreMQTT/mqtt_demo_mutual_auth.c) to send those sensor measurements to AWS IoT Core over MQTT. Keep in mind that you may need to copy and paste some configs from [config_files](https://github.com/aws/amazon-freertos/tree/master/vendors/espressif/boards/esp32/aws_demos/config_files) to [freertos-configs](https://github.com/tgsong/amazon-freertos-examples/tree/master/freertos-configs).

Johnnyjax, 2021-02-06:
I tried copying snippets of code from the coreMQTT Mutual Auth Demo but kept having errors when building the project. There isn't any documentation of how the main.c program should be. Plus I also need to handle all the wifi connection to a network and It isn't clear how I'd achieve that from the main.c file. The easiest way would definitely just be to edit the coreMQTT Mutual Auth Demo file and run the demo instead but what also happens if i need to select functions from multiple demo projects.

yourslab, 2021-02-06:
For the most part, you would be able to use the [functions for WiFi documented for the esp-idf](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/network/esp_wifi.html). I agree that there is quite a bit of work to doing that, so modifying the demos would likely be the easiest route. If you need functions from multiple demos, just modify the [CMakeLists.txt](https://github.com/aws/amazon-freertos/blob/master/demos/coreMQTT/CMakeLists.txt#L30-L42) file to include the module dependencies you need. For example, if you'd like to use the coreHTTP library, you would just add `AFR::core_http` to the list.

Johnnyjax, 2021-02-06:
Once I got my project set up like it was done here - https://github.com/tgsong/amazon-freertos-examples.git. The very first thing i did was cleared the main file and put the esp-idf blink example in the file and it built and blinked the LED. I then copied the esp-idf softAP example for Wifi into my main.c file but when I did that, It didn't build successfully. It gave these errors. 

./src/main.c: In function 'wifi_event_handler':
../src/main.c:36:21: error: 'WIFI_EVENT_AP_STACONNECTED' undeclared (first use in this function)
     if (event_id == WIFI_EVENT_AP_STACONNECTED) {
                     ^
../src/main.c:36:21: note: each undeclared identifier is reported only once for each function it appears in
../src/main.c:37:9: error: unknown type name 'wifi_event_ap_staconnected_t'
         wifi_event_ap_staconnected_t* event = (wifi_event_ap_staconnected_t*) event_data;
         ^
../src/main.c:37:48: error: 'wifi_event_ap_staconnected_t' undeclared (first use in this function)
         wifi_event_ap_staconnected_t* event = (wifi_event_ap_staconnected_t*) event_data;
                                                ^
../src/main.c:37:77: error: expected expression before ')' token
         wifi_event_ap_staconnected_t* event = (wifi_event_ap_staconnected_t*) event_data;
                                                                             ^
In file included from ../freertos/vendors/espressif/boards/esp32/aws_demos/config_files/FreeRTOSConfig.h:40:0,
                 from ../freertos/freertos_kernel/include/FreeRTOS.h:57,
                 from ../freertos/vendors/espressif/boards/esp32/components/freertos/include/freertos/FreeRTOS.h:1,
                 from ../src/main.c:10:
../src/main.c:39:31: error: request for member 'mac' in something not a structure or union
                  MAC2STR(event->mac), event->aid);
                               ^
../freertos/vendors/espressif/esp-idf/components/esp32/include/rom/ets_sys.h:623:21: note: in definition of macro 'MAC2STR'
 #define MAC2STR(a) (a)[0], (a)[1], (a)[2], (a)[3], (a)[4], (a)[5]
                     ^
../freertos/vendors/espressif/esp-idf/components/log/include/esp_log.h:320:41: note: in expansion of macro 'ESP_LOG_LEVEL'
         if ( LOG_LOCAL_LEVEL >= level ) ESP_LOG_LEVEL(level, tag, format, ##__VA_ARGS__); \
                                         ^
../freertos/vendors/espressif/esp-idf/components/log/include/esp_log.h:276:38: note: in expansion of macro 'ESP_LOG_LEVEL_LOCAL'
 #define ESP_LOGI( tag, format, ... ) ESP_LOG_LEVEL_LOCAL(ESP_LOG_INFO,    tag, format, ##__VA_ARGS__)
                                      ^
../src/main.c:38:9: note: in expansion of macro 'ESP_LOGI'
         ESP_LOGI(TAG, "station "MACSTR" join, AID=%d",
         ^
../src/main.c:39:31: error: request for member 'mac' in something not a structure or union
                  MAC2STR(event->mac), event->aid);
                               ^
../freertos/vendors/espressif/esp-idf/components/esp32/include/rom/ets_sys.h:623:29: note: in definition of macro 'MAC2STR'
 #define MAC2STR(a) (a)[0], (a)[1], (a)[2], (a)[3], (a)[4], (a)[5]
                             ^
../freertos/vendors/espressif/esp-idf/components/log/include/esp_log.h:320:41: note: in expansion of macro 'ESP_LOG_LEVEL'
         if ( LOG_LOCAL_LEVEL >= level ) ESP_LOG_LEVEL(level, tag, format, ##__VA_ARGS__); \
                                         ^
../freertos/vendors/espressif/esp-idf/components/log/include/esp_log.h:276:38: note: in expansion of macro 'ESP_LOG_LEVEL_LOCAL'
 #define ESP_LOGI( tag, format, ... ) ESP_LOG_LEVEL_LOCAL(ESP_LOG_INFO,    tag, format, ##__VA_ARGS__)
                                      ^
../src/main.c:38:9: note: in expansion of macro 'ESP_LOGI'
         ESP_LOGI(TAG, "station "MACSTR" join, AID=%d",
         ^
../src/main.c:39:31: error: request for member 'mac' in something not a structure or union
                  MAC2STR(event->mac), event->aid);
                               ^
../freertos/vendors/espressif/esp-idf/components/esp32/include/rom/ets_sys.h:623:37: note: in definition of macro 'MAC2STR'
 #define MAC2STR(a) (a)[0], (a)[1], (a)[2], (a)[3], (a)[4], (a)[5]
                                     ^
../freertos/vendors/espressif/esp-idf/components/log/include/esp_log.h:320:41: note: in expansion of macro 'ESP_LOG_LEVEL'
         if ( LOG_LOCAL_LEVEL >= level ) ESP_LOG_LEVEL(level, tag, format, ##__VA_ARGS__); \
                                         ^
../freertos/vendors/espressif/esp-idf/components/log/include/esp_log.h:276:38: note: in expansion of macro 'ESP_LOG_LEVEL_LOCAL'
 #define ESP_LOGI( tag, format, ... ) ESP_LOG_LEVEL_LOCAL(ESP_LOG_INFO,    tag, format, ##__VA_ARGS__)
                                      ^
../src/main.c:38:9: note: in expansion of macro 'ESP_LOGI'
         ESP_LOGI(TAG, "station "MACSTR" join, AID=%d",
         ^
../src/main.c:39:31: error: request for member 'mac' in something not a structure or union
                  MAC2STR(event->mac), event->aid);
                               ^
../freertos/vendors/espressif/esp-idf/components/esp32/include/rom/ets_sys.h:623:45: note: in definition of macro 'MAC2STR'
 #define MAC2STR(a) (a)[0], (a)[1], (a)[2], (a)[3], (a)[4], (a)[5]
[... 1017 lines cut ...]
                  MAC2STR(event->mac), event->aid);
                                            ^
../freertos/vendors/espressif/esp-idf/components/log/include/esp_log.h:312:137: note: in definition of macro 'ESP_LOG_LEVEL'
[... cut ...]
../src/main.c: In function 'wifi_init_softap':
../src/main.c:56:57: error: 'WIFI_EVENT' undeclared (first use in this function)
     ESP_ERROR_CHECK(esp_event_handler_instance_register(WIFI_EVENT,
                                                         ^
../freertos/vendors/espressif/esp-idf/components/esp32/include/esp_err.h:117:31: note: in definition of macro 'ESP_ERROR_CHECK'
         esp_err_t __err_rc = (x);                                       \
                               ^
ninja: build stopped: subcommand failed.

yourslab, 2021-02-06:
Keep in mind that the esp-idf in amazon-freertos is a different version of the original esp-idf. For example, `WIFI_EVENT_AP_STACONNECTED` is named as `SYSTEM_EVENT_AP_STACONNECTED`. The exact code that you can use to connect to WiFi for ESP32 can be found under https://github.com/aws/amazon-freertos/blob/master/vendors/espressif/boards/ports/wifi/iot_wifi.c

The way it works for the amazon-freertos demos is as follows
[iot_wifi.c](https://github.com/aws/amazon-freertos/blob/master/vendors/espressif/boards/ports/wifi/iot_wifi.c) ---> [aws_iot_network_manager.c](https://github.com/aws/amazon-freertos/blob/master/demos/network_manager/aws_iot_network_manager.c) ---> [iot_demo_freertos.c](https://github.com/aws/amazon-freertos/blob/master/demos/demo_runner/iot_demo_freertos.c)

Johnnyjax, 2021-02-06:
Oh, Thanks a lot. I really appreciate this. I'll try that. Another issue I had was I modified the mqtt_demo_mutual_auth demo to continuously publish to a topic. It published successfully for a while and then It started giving this error.

MQTT_MutualAuth_Demo] [mqtt_demo_mutual_auth.c:892] 142 3343 [iot_thread] Failed to send PUBLISH message to broker: Topic=custom_aws_project/example/topic, Error=MQTTNoMemory143 3343 [iot_thread]

I searched for the MQTTNoMemory error online and it returns MQTTNoMemory if pBuffer is too small to hold the MQTT packet. I don't understand what this means or how I can increase the buffer. How do i fix this? Thanks

yourslab, 2021-02-06:
The buffer to receive MQTT messages is too small for the message it is trying to receive. The fix for this is to simply increase `democonfigNETWORK_BUFFER_SIZE`:
https://github.com/aws/amazon-freertos/blob/master/vendors/espressif/boards/esp32/aws_demos/config_files/mqtt_demo_mutual_auth_config.h#L119

By default, it is only 1 KB.

Johnnyjax, 2021-02-06:
Thanks. Does this mean I can't indefinitely publish to a topic? because all I did was basically modify the mqtt_demo_mutual_auth file so that it indefinitely publishes "helloWorld" to a topic with a delay in a while(1) loop. The first few publish were successful but It continuously failed after that. Even If i increase the democonfigNETWORK_BUFFER_SIZE size, I'd still face this issue as long as i continuously publish. Is there a workaround?

yourslab, 2021-02-06:
No, the mutual auth demo is indefinitely doing the same thing too. Could it be an issue with the change you made?

Johnnyjax, 2021-02-06:
All I basically did was change this
        /* Publish messages with QoS1, send and process Keep alive messages. */
        for( ulPublishCount = 0;
             ( ( xDemoStatus == pdPASS ) && ( ulPublishCount < ulMaxPublishCount ) );
             ulPublishCount++ )
        {
            LogInfo( ( "Publish to the MQTT topic %s.", mqttexampleTOPIC ) );
            xDemoStatus = prvMQTTPublishToTopic( &xMQTTContext );

            if( xDemoStatus == pdPASS )
            {
                /* Process incoming publish echo, since application subscribed to the same
                 * topic, the broker will send publish message back to the application.
                 * #prvWaitForPacket will try to receive an incoming PUBLISH packet from broker.
                 * Please note that PUBACK for the outgoing PUBLISH may also be received before
                 * receiving an incoming PUBLISH. */
                LogInfo( ( "Attempt to receive publish message from broker." ) );
                xMQTTStatus = prvWaitForPacket( &xMQTTContext, MQTT_PACKET_TYPE_PUBLISH );

                if( xMQTTStatus != MQTTSuccess )
                {
                    xDemoStatus = pdFAIL;
                }
            }

            /* Leave Connection Idle for some time. */
            LogInfo( ( "Keeping Connection Idle..." ) );
            vTaskDelay( mqttexampleDELAY_BETWEEN_PUBLISHES_TICKS );
        }

to this
        while(1)
        {
            LogInfo( ( "Publish to the MQTT topic %s.", mqttexampleTOPIC ) );
            xDemoStatus = prvMQTTPublishToTopic( &xMQTTContext );

            /* Leave Connection Idle for some time. */
            LogInfo( ( "Keeping Connection Idle..." ) );
            vTaskDelay( mqttexampleDELAY_BETWEEN_PUBLISHES_TICKS );
        }

Johnnyjax, 2021-02-06:
It doesn't fail too if all i do is change the for loop to a while loop.

yourslab, 2021-02-06:
Okay, but you are continuously publishing messages without receiving them. Eventually, the MQTT broker will cut you off because the buffer its using to send the ESP32 messages gets full. To fix this, you can call `prvWaitForPacket` after calling `prvMQTTPublishToTopic`. However, a better solution is to not subscribe to the topic at all. You probably just want to send those sensor readings anyway without the ESP32 having to receive them again. To do this, just remove the call to `prvMQTTSubscribeWithBackoffRetries`.

Johnnyjax, 2021-02-06:
Thanks. I removed the call to prvMQTTSubscribeWithBackoffRetries but the same error still occurs. It publishes for a couple of seconds and then generates the MQTTNoMemory error.

yourslab, 2021-02-06:
```
    xMQTTPublishInfo.pPayload = mqttexampleMESSAGE;
    xMQTTPublishInfo.payloadLength = strlen( mqttexampleMESSAGE );
```
Did you change this to point to some memory location and if so, can you print out what their contents are? Changes are the length is getting larger and larger and eventually `payloadLength` >`democonfigNETWORK_BUFFER_SIZE` .

Johnnyjax, 2021-02-06:
Nope.
#define mqttexampleMESSAGE        "Hello World!"
The only lines i changed were the ones i sent previously.  Everything else is the same as the original.

yourslab, 2021-02-06:
My mistake. You will still need the `prvWaitForPacket` in order to receive the PUBACK. Please add the call back inside your while loop, and it should work.

Johnnyjax, 2021-02-06:
I need to add that even if i'm not subscribed to the topic?

yourslab, 2021-02-06:
Yes, the broker will still need to send the PUBACK for any publish that you send out. `prvWaitForPacket` is for receiving any data from the broker, not just publishes.

Johnnyjax, 2021-02-06:
Oh... Thanks a lot. I now understand it perfectly and it works. I really appreciate the help 👍

yourslab, 2021-02-06:
Of course, my pleasure. I will close this issue for now, but feel free to open again if you encounter other issues with your project. Thank you.

Johnnyjax, 2021-02-08:
Hi. So I tried your advice about using the functions for wifi documented for the esp idf and it worked. I placed the station example of the esp-idf in my main.c file and it built and worked. I then tried combining the station code with the mqtt_demo_mutual_auth.c example and placing them in my main file with the plan that the esp would first connect to the wifi network specified, connect to aws iot and then start publishing. My main code is [here](https://github.com/Johnnyjax/my_custom_aws_project/blob/main/src/main.c). When I tried running it on my esp, It basically continuiously connect and disconnects to the wifi network repeatedly and never runs the mqtt_demo_mutual_auth code section I placed there.

yourslab, 2021-02-08:
Hey @Johnnyjax. I've seen this problem before when working on my own ESP32 projects as well. It's likely that one of your tasks is running out of stack space. Can you show a sample of the serial output? Another thing I've done to resolve this issue is to erase program memory before flashing again.

Johnnyjax, 2021-02-08:
I erased the program memory before flashing this time but It still did the same thing. It just gets stuck in the loop of connecting and disconnecting.

C:\Users\John\Documents\AWS\my_custom_aws_project>idf.py -B build -p COM5 monitor
Note: You are using Python 3.8.7. Python 3 support is new, please report any problems you encounter. Search for 'Setting the Python Interpreter' in the ESP-IDF docs if you want to use Python 2.7.
Checking Python dependencies...
Python requirements from C:\Users\John\Documents\AWS\my_custom_aws_project\freertos\vendors\espressif\esp-idf\requirements.txt are satisfied.
Running idf_monitor in directory C:\Users\John\Documents\AWS\my_custom_aws_project
Executing "C:\Users\John\.espressif\python_env\idf3.3_py3.8_env\Scripts\python.exe C:\Users\John\Documents\AWS\my_custom_aws_project\freertos\vendors\espressif\esp-idf\tools/idf_monitor.py -p COM5 -b 115200 C:\Users\John\Documents\AWS\my_custom_aws_project\build\afr_demo -m 'C:\Users\John\.espressif\python_env\idf3.3_py3.8_env\Scripts\python.exe' 'C:\Users\John\Documents\AWS\my_custom_aws_project\freertos\vendors\espressif\esp-idf\tools\idf.py' '-B'"...
--- idf_monitor on COM5 115200 ---
--- Quit: Ctrl+] | Menu: Ctrl+T | Help: Ctrl+T followed by Ctrl+H ---
ets Jun  8 2016 00:22:57

rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0018,len:4
load:0x3fff001c,len:7260
load:0x40078000,len:12652
ho 0 tail 12 room 4
load:0x40080400,len:6708
entry 0x40080778
I (73) boot: Chip Revision: 1
I (73) boot_comm: chip revision: 1, min. bootloader chip revision: 0
I (41) boot: ESP-IDF v3.3.2-241-g1d2d93acd 2nd stage bootloader
I (41) boot: compile time 08:29:35
I (41) boot: Enabling RNG early entropy source...
I (47) boot: SPI Speed      : 40MHz
I (51) boot: SPI Mode       : DIO
I (55) boot: SPI Flash Size : 4MB
I (59) boot: Partition Table:
I (63) boot: ## Label            Usage          Type ST Offset   Length
I (70) boot:  0 nvs              WiFi data        01 02 00010000 00006000
I (78) boot:  1 otadata          OTA data         01 00 00016000 00002000
I (85) boot:  2 phy_init         RF data          01 01 00018000 00001000
I (92) boot:  3 ota_0            OTA app          00 10 00020000 00177000
I (100) boot:  4 ota_1            OTA app          00 11 001a0000 00177000
I (107) boot:  5 storage          WiFi data        01 02 00317000 00010000
I (115) boot: End of partition table
I (119) boot: ota rollback check done
I (124) boot_comm: chip revision: 1, min. application chip revision: 0
I (131) esp_image: segment 0: paddr=0x00020020 vaddr=0x3f400020 size=0x24f78 (151416) map
I (193) esp_image: segment 1: paddr=0x00044fa0 vaddr=0x3ffbdb60 size=0x031c4 ( 12740) load
I (198) esp_image: segment 2: paddr=0x0004816c vaddr=0x40080000 size=0x00400 (  1024) load
0x40080000: _WindowOverflow4 at C:\Users\John\Documents\AWS\my_custom_aws_project\build/../freertos/freertos_kernel/portable/ThirdParty/GCC/Xtensa_ESP32_IDF3/xtensa_vectors.S:1713

I (200) esp_image: segment 3: paddr=0x00048574 vaddr=0x40080400 size=0x07a9c ( 31388) load
I (222) esp_image: segment 4: paddr=0x00050018 vaddr=0x400d0018 size=0x898ac (563372) map
0x400d0018: _flash_cache_start at ??:?

I (420) esp_image: segment 5: paddr=0x000d98cc vaddr=0x40087e9c size=0x0c9dc ( 51676) load
0x40087e9c: prvIsQueueEmpty at C:\Users\John\Documents\AWS\my_custom_aws_project\build/../freertos/freertos_kernel/queue.c:2429

I (454) boot: Loaded app from partition at offset 0x20000
I (454) boot: Disabling RNG early entropy source...
I (454) cpu_start: Pro cpu up.
I (458) cpu_start: Application information:
I (463) cpu_start: Project name:     esp-idf
I (467) cpu_start: App version:      18ce8e7
I (472) cpu_start: Compile time:     Feb  8 2021 20:44:24
I (478) cpu_start: ELF file SHA256:  2a538f5807e30a95...
I (484) cpu_start: ESP-IDF:          v3.3.2-241-g1d2d93acd
I (491) cpu_start: Single core mode
I (495) heap_init: Initializing. RAM available for dynamic allocation:
I (502) heap_init: At 3FFAFF10 len 000000F0 (0 KiB): DRAM
I (508) heap_init: At 3FFB6388 len 00001C78 (7 KiB): DRAM
I (514) heap_init: At 3FFB9A20 len 00004108 (16 KiB): DRAM
I (520) heap_init: At 3FFBDB5C len 00000004 (0 KiB): DRAM
I (526) heap_init: At 3FFC9260 len 00016DA0 (91 KiB): DRAM
I (533) heap_init: At 3FFE0440 len 0001FBC0 (126 KiB): D/IRAM
[... 637 lines cut ...]
I (496) heap_init: Initializing. RAM available for dynamic allocation:
I (502) heap_init: At 3FFAFF10 len 000000F0 (0 KiB): DRAM
I (508) heap_init: At 3FFB6388 len 00001C78 (7 KiB): DRAM
I (515) heap_init: At 3FFB9A20 len 00004108 (16 KiB): DRAM
I (521) heap_init: At 3FFBDB5C len 00000004 (0 KiB): DRAM
I (527) heap_init: At 3FFC9260 len 00016DA0 (91 KiB): DRAM
I (533) heap_init: At 3FFE0440 len 0001FBC0 (126 KiB): D/IRAM
I (539) heap_init: At 40078000 len 00008000 (32 KiB): IRAM
I (546) heap_init: At 40094878 len 0000B788 (45 KiB): IRAM
I (552) cpu_start: Pro cpu start user code
I (234) cpu_start: Starting scheduler on PRO CPU.
I (244) wifi station: ESP_WIFI_MODE_STA
I (264) wifi:wifi driver task: 3ffbd51c, prio:23, stack:3584, core=0
I (264) system_api: Base MAC address is not set, read default base MAC address from BLK0 of EFUSE
I (264) system_api: Base MAC address is not set, read default base MAC address from BLK0 of EFUSE
I (284) wifi:wifi firmware version: 1afdae0
I (284) wifi:config NVS flash: enabled
I (284) wifi:config nano formating: disabled
I (284) wifi:Init dynamic tx buffer num: 32
I (284) wifi:Init data frame dynamic rx buffer num: 32
I (294) wifi:Init management frame dynamic rx buffer num: 32
I (294) wifi:Init management short buffer num: 32
I (304) wifi:Init static rx buffer size: 1600
I (304) wifi:Init static rx buffer num: 10
I (314) wifi:Init dynamic rx buffer num: 32

yourslab, 2021-02-08:
It looks like you are getting a panic from certificate provisioning. Seems as though the root cause is that you are logging a location in memory that is not initialized.

Johnnyjax, 2021-02-09:
It's really difficult to try to create my own main.c file with my custom code. There's just too many things I don't fully understand about the entire amazon freertos structure and how the demo programs work. I'm currently working as an intern in a startup and was given a product to write the firmware for with amazon freertos but no one in the company has ever used amazon freertos for the embedded side of things. Do you think it's right to just edit the demo files for developing a product and not just a hobby Or do you think its important to create my own custom code and figure all these out. Thanks.

elieDaan, 2021-02-09:
To share my experience, yes it's a struggle to use AFR outside the demo project. But until you've succeed to find the correct CMakeList in your project to use AFR as a subdirectory, it's gonna be easier to develop your product. This [issue](https://github.com/aws/amazon-freertos/issues/832) could help you understand how to use AFR as a subdirectory. 

Sometimes, you just have to copy/paste the same code as the demo files  because there is no value in your application to understand exactly what's going on like the initialize functions of wifi and mqtt. 

Also, I recommend to use a maximum of components directory in your project to avoid a big main.c file..

Johnnyjax, 2021-02-09:
Setting it up as a subdirectory and adjusting the CMakeList file isn't an issue and I've been able to do that. [This repository](https://github.com/tgsong/amazon-freertos-examples) have been helpful with that. But I had to basically had to copy the [main.c file](https://github.com/aws/amazon-freertos/blob/master/vendors/espressif/boards/esp32/aws_demos/application_code/main.c) into my main.c file and I've been able to run any demo project i want. But that method isn't really customizable for a project. Even copying a demo project directly into my main file won't work because there are lot of other dependencies. The only way I've been able to get any customization is by directly editing the demo project files for my specific application while leaving my main.c file the same as this [main.c file](https://github.com/aws/amazon-freertos/blob/master/vendors/espressif/boards/esp32/aws_demos/application_code/main.c). For instance, If i want to send my own customized mqtt messages, I'll go to to the [mqtt_demo_mutual_auth.c](https://github.com/aws/amazon-freertos/blob/fd27d7bbd0d370004817c217b7fb95242f136b0d/demos/coreMQTT/mqtt_demo_mutual_auth.c) file and directly edit the topic and message to send or whatever else. And this method works. I just want to find out if its okay to do this even in a professional setting or if i have to figure out how to create my own customized main.c file.

yourslab, 2021-02-09:
Yes, people have built applications by using the `demos` directory in the past, and it's usually the easiest way to create a working application without worrying too much about what dependencies you need. One disadvantage is that your build times may take a while. The way the project is setup, a lot of source files are linked by default even though your specific application may not need them. Ultimately, code that is unneeded will be stripped off in the final image, but you may notice some files being built even though your application does not need them at all.

Johnnyjax, 2021-02-09:
Yeah. That makes sense. The build time is really long. I'll stick to working with the demo files for now and worry about stripping off the unnecessary parts later on. One part of the project that I've been trying to figure out though is the wifi provisioning part so that the wifi ssid and password can be done through an app. I saw [here](https://docs.aws.amazon.com/freertos/latest/userguide/ble-demo.html) that it seems the wifi provisioning support for amazon freertos is only through BLE for now. Is there a way to provision it over WIFI SoftAP instead? My second question is i found out that there's an android sdk for freertos available [here](https://github.com/aws/amazon-freertos-ble-android-sdk) for provisioning via ble. Is there a way to download the app without having to setup android studio and all. all i need is the app.

yourslab, 2021-02-09:
Yes, it is possible to provision over WiFi SoftAP as detailed here:
https://docs.aws.amazon.com/freertos/latest/userguide/freertos-wifi.html
However, as the docs state, you will have to supply the additional functionality yourself, including DHCP and HTTP server capabilities, to achieve full support of AP mode.

Unfortunately, I don't believe there is a way to only download the app.

Johnnyjax, 2021-02-09:
You can close the issue. Thank you so much. I really appreciate the help.

yourslab, 2021-02-09:
Okay, as always, feel free to reach out again

Johnnyjax, 2021-02-23:
Hi @yourslab. I noticed incoming publish don't get handled automatically in the event callback  immediately they arrive. Even PUBACK don't get processed automatically unless one calls a prvWaitForPacket function or an MQTT_ProcessLoop function. Is there a way to get It handled automatically immediately any incoming publish arrives.

yourslab, 2021-02-23:
The event callback is called by MQTT_ProcessLoop. It is necessary that MQTT_ProcessLoop is called for publishes from the broker to be handled. MQTT_ProcessLoop can be called from a separate task in case your task is already busy doing other work. However, you will need to guard the `MQTTContext_t` because it is not thread safe. The MQTT Agent uses queues to achieve this:
https://github.com/FreeRTOS/freertos-mqtt-agent
You can copy code from here and use it in your own program.

Johnnyjax, 2021-02-23:
Oh. Thanks. I added the MQTT_ProcessLoop  to my main loop so that it always call it whenever the loop runs and that solved it.

Johnnyjax, 2021-02-25:
Hi @yourslab. Thanks for all your responses. I really appreciate it. I've basically created a loop where my MQTT_ProcessLoop continuously gets called over and over. And it works great to an extent, when messages come in, it displays the incoming messages and also takes care of the acknowledge responses. The problem I have is when something doesn't go as expected. For instance, If i try to publish to a topic I don't have the permission to publish to or an incoming publish comes in that I'm not subscribed to, it just continuously shows errors nonstop on the serial monitor and I'd have to reset the device. Is there a way to kind of reset it so that if i try to publish to a topic Im not authorized to or network connection breaks, It just shows the error once, clears and just continues with it's normal operation? Here's a sample of what it shows right now

 46 4000 [Tmr Svc]
47 4001 [Tmr Svc] [INFO] [MQTT] [mqtt_demo_helpers.c:892] 48 4001 [Tmr Svc] PUBLISH sent for topic $aws/things/John/shadow/name/+/update to broker with packet ID 2.

49 4001 [Tmr Svc]
50 4297 [iot_thread] [ERROR] [Transport_Secure_Sockets] [transport_secure_sockets.c:189] 51 4298 [iot_thread] Failed to receive data over network. bytesReceived=-3084852 4298 [iot_thread]
53 4298 [iot_thread] [ERROR] [MQTT] [core_mqtt_serializer.c:2396] 54 4298 [iot_thread] A single byte was not read from the transport: transportStatus=-30848.55 4298 [iot_thread]
56 4298 [iot_thread] [ERROR] [MQTT] [core_mqtt.c:1320] 57 4298 [iot_thread] Receiving incoming packet length failed. Status=MQTTRecvFailed58 4298 [iot_thread]
59 4298 [iot_thread] [ERROR] [MQTT] [core_mqtt.c:2192] 60 4298 [iot_thread] Exiting process loop due to failure: ErrorStatus=MQTTRecvFailed61 4298 [iot_thread]
62 4298 [iot_thread] [ERROR] [Transport_Secure_Sockets] [transport_secure_sockets.c:189] 63 4298 [iot_thread] Failed to receive data over network. bytesReceived=-2764864 4298 [iot_thread]
65 4298 [iot_thread] [ERROR] [MQTT] [core_mqtt_serializer.c:2396] 66 4298 [iot_thread] A single byte was not read from the transport: transportStatus=-27648.67 4298 [iot_thread]
68 4298 [iot_thread] [ERROR] [MQTT] [core_mqtt.c:1320] 69 4298 [iot_thread] Receiving incoming packet length failed. Status=MQTTRecvFailed70 4298 [iot_thread]
71 4298 [iot_thread] [ERROR] [MQTT] [core_mqtt.c:2192] 72 4298 [iot_thread] Exiting process loop due to failure: ErrorStatus=MQTTRecvFailed73 4298 [iot_thread]
74 4298 [iot_thread] [ERROR] [Transport_Secure_Sockets] [transport_secure_sockets.c:189] 75 4298 [iot_thread] Failed to receive data over network. bytesReceived=-2764876 4298 [iot_thread]
77 4298 [iot_thread] [ERROR] [MQTT] [core_mqtt_serializer.c:2396] 78 4298 [iot_thread] A single byte was not read from the transport: transportStatus=-27648.79 4298 [iot_thread]
80 4298 [iot_thread] [ERROR] [MQTT] [core_mqtt.c:1320] 81 4298 [iot_thread] Receiving incoming packet length failed. Status=MQTTRecvFailed195 4298 [iot_thread] Failed to receive data over network. bytesReceived=-27648196 4300 [iot_thread]
197 4300 [iot_thread] [ERROR] [MQTT] [core_mqtt_serializer.c:2396] 198 4300 [iot_thread] A single byte was not read from the transport: transportStatus=-27648.199 4300 [iot_thread]
345 4300 [iot_thread] Receiving incoming packet length failed. Status=MQTTRecvFailed346 4302 [iot_thread]
347 4302 [iot_thread] [ERROR] [MQTT] [core_mqtt.c:2192] 478 4302 [iot_thread]
479 4302 [iot_thread] [ERROR] [MQTT] [core_mqtt.c:2192] 480 4303 [iot_thread] Exiting process loop due to failure: ErrorStatus=MQTTRecvFailed543 4305 [iot_thread] Failed to receive data over network. bytesReceived=-27648544 4305 [iot_thread]
545 4305 [iot_thread] [ERROR] [MQTT] [core_mqtt_serializer.c:2396] 697 4305 [iot_thread]
698 4307 [iot_thread] [ERROR] [Transport_Secure_Sockets] [transport_secure_sockets.c:189] 699 4307 [iot_thread] Failed to receive data over network. bytesReceived=-27648700 4307 [iot_thread]
849 4307 [iot_thread] Receiving incoming packet length failed. Status=MQTTRecvFailed850 4309 [iot_thread]
851 4309 [iot_thread] [ERROR] [MQTT] [core_mqtt.c:2192] 852 4309 [iot_thread] Exiting process loop due to failure: ErrorStatus=MQTTRecvFailed853 4309 [iot_thread]
1003 4309 [iot_thread]

It just keeps repeating this and I have to reset the esp32 to do anything again. Thanks

yourslab, 2021-02-25:
I think you need to update your logic so that if `MQTT_ProcessLoop` returns `MQTTRecvFailed`, you will try to re-establish the connection with the MQTT broker by using the same logic as below:
https://github.com/aws/amazon-freertos/blob/master/demos/coreMQTT/mqtt_demo_mutual_auth.c#L514-L533

Johnnyjax, 2021-02-26:
Thank you so much 👍 That fixed it.

Johnnyjax, 2021-02-26:
Hi @yourslab. I have 2 questions.
First question:  I added the amazon-freertos as a submodule into my project like the project [here](https://github.com/tgsong/amazon-freertos-examples) but I've made a lot of project specific changes to the amazon-freertos for my project, for instance, creating my own demo project in the demo folder and editing lots of the configuration files. I want another intern in my company to be able to clone my project and try it out on her esp32 board and also be a contributor. How do I upload this to my own private repository and the changes I made to the amazon-freertos submodule directory reflect when she clones it because whenever I push my project onto my repository, changes in the submodule aren't reflected.

Second question: still related to the first though(maybe the same). I also want to do some work with stripping down the amazon-freertos so that the build time is less and I want all the changes to be reflected too in my github repository so that my version gets cloned. Thanks

yourslab, 2021-02-26:
I think you should not have a demo project in the demo folder but just have all your code in the `src` and `freertos-configs` directories. As we discussed before, that should also decrease your build times significantly.

Johnnyjax, 2021-02-26:
Oh. I guess that's the next step I have to figure out then. It was just way easier to do it this way because the [DEMO_RUNNER_RunDemos](https://github.com/tgsong/amazon-freertos-examples/blob/99ca388a22b9419c04f1f48f90d89b30d38b7677/src/main.c#L164) function automatically does all the network initialization so adjusting the necessary config files to setup my project as a demo project seemed like an easier alternative. I've tried doing the network initialization myself but I haven't been successful.

yourslab, 2021-02-26:
One thing you can try doing is creating a `demos` folder under `src`, then you can move your demo there. From the top-level `CMakeLists.txt`, insert the following line:
```
add_subdirectory(src/demos/<demo-folder>)
```

This may or may not improve the build times. Let me know if this is able to improve build times for you, but I'm a little doubtful. Ideally, the network initialization code can be copied into the `src` directory and you can then remove the call to `DEMO_RUNNER_RunDemos`.

yourslab, 2021-02-26:
Also, make sure to add that line after `add_subdirectory(freertos)`

Johnnyjax, 2021-02-27:
Thanks. I'll let you know how it goes

Johnnyjax, 2021-02-27:
Got the project to work without having to mess with the freertos folder at all. I removed the call to DEMO_RUNNER_RunDemos and called my own network initialization code and moved my project files out of the freertos demo directory and It's working. The only issue I'm having now is that my code still depends on some of the freertos config files. The one that's been giving me issues is the [aws_iot_network_config.h](https://github.com/aws/amazon-freertos/blob/fd27d7bbd0d370004817c217b7fb95242f136b0d/vendors/espressif/boards/esp32/aws_demos/config_files/aws_iot_network_config.h) file. I want to disable the BLE without having to edit the file directly in that directory because I wanted to make sure my freertos folder is even with [amazon-freertos](https://github.com/aws/amazon-freertos)  so I copied the aws_iot_network_config.h to my freertos-configs folder like it is [here](https://github.com/tgsong/amazon-freertos-examples/tree/master/freertos-configs) and disabled the BLE there. I also added "include_directories(BEFORE freertos-configs)" in my project cmakelist. But for some reason, my aws_iot_network_config.h file that I placed in my freertos-config folder is being ignored and the one in the amazon freertos config directory is being used instead. Do you have any idea on how I can solve this issue? Thanks

Johnnyjax, 2021-02-27:
Figured out the problem. I only copied the aws_iot_network_config.h file from the amazon freertos config directory and didn't copy aws_demo_config.h and iot_config.h file like it is [here](https://github.com/tgsong/amazon-freertos-examples/tree/master/freertos-configs). copying the 3 files solved the issue. Thanks a lot for your help so far. I wouldn't have been able to come this far without your input. 👍

yourslab, 2021-02-28:
I just provided the guide, but you figured it out on your own 😉 Glad I was able to help.

By the way, did your build times improve from making that change? My guess is that you still end up compiling all the demos, but I have a plan in place so that a project structure like yours will only compile the relevant sources for your project.

Johnnyjax, 2021-02-28:
So you're right. The build time didn't improve. My file structure is basically like the one [here](https://github.com/tgsong/amazon-freertos-examples). I tried with the suggestion you gave of creating a demos folder under src and adding add_subdirectory(src/demos/<demo-folder>) to the cmakelists file but I'll also need a cmakelists file in the subdirectory itself so it brought up errors . I couldn't really figure out how to get it to work that way but I ended up creating a project folder in my components directory and adding my header files to the include folder and source files to the src folder of the component. I also created a cmakelists file for the component too and edited the general cmakelists file too to see my component. Any time I built and it complained of any file missing, I just copy the file from the demos directory and add it to either the src or include folder and I also had to edit some of the header files and source files so that It doesn't try to run any demos but my application instead and that took a bit of work but it worked and built successfully. But when I went to my cmakelist file like it is [here](https://github.com/tgsong/amazon-freertos-examples/blob/master/CMakeLists.txt) and I removed the AFR::demo_core_mqtt from this-
 target_link_libraries(
    afr_demo
    PRIVATE
    AFR::demo_core_mqtt
    AFR::common_io
)
It didn't build successfully and was complaining about some files missing so I left that there which means its still compiling the demos and my build time hasn't been able to improve. so any guide you can give would be helpful on how i can improve that. Thanks.

yourslab, 2021-03-03:
Thanks for going into detail with this. I'm currently working on this issue and will let you know when it's fixed.

Johnnyjax, 2021-03-03:
Alright. Thanks a lot.

Yoimer, 2021-03-09:
> I've setup amazon freertos and esp-idf and have been able to successfully run the demo program. I've also gone through the "Using FreeRTOS in your own CMake project for ESP32" in the getting started [...]

@Johnnyjax Hello there. Please would you mind telling which sensors were you able to handle and how?

I am trying to do the same using this based on [tgsong](https://github.com/tgsong/amazon-freertos-examples). I already handled the building and flashing for the `demo_core_mqtt`. Now I want to bring my sensors to it.
I was able to read that sensor data on `ESP-IDFV4.2` using [UncleRus](https://github.com/UncleRus/esp-idf-lib). But when it comes to AWS FreeRTOS ESP32, I could not handle it. Thanks in advance.

Johnnyjax, 2021-03-09:
Hello. I guess there will be different ways of handling this. If you just want to quickly get this running, just add the UncleRus as an extra component directory in the cmakelists [here](https://github.com/aws/amazon-freertos/blob/fd27d7bbd0d370004817c217b7fb95242f136b0d/demos/coreMQTT/CMakeLists.txt) using this 

get_filename_component(
    EXTRA_COMPONENT_DIRS
    "unclerus_component_path"
)
If you've done that, you can directly include the components in the [mqtt_mutual_auth ](https://github.com/aws/amazon-freertos/blob/fd27d7bbd0d370004817c217b7fb95242f136b0d/demos/coreMQTT/mqtt_demo_mutual_auth.c)demo and adjust the topic and other parameters for your project. You can adjust [here](https://github.com/aws/amazon-freertos/blob/fd27d7bbd0d370004817c217b7fb95242f136b0d/demos/coreMQTT/mqtt_demo_mutual_auth.c#L822) too so that it sends you parameters to aws iot instead.

Yoimer, 2021-03-10:
@Johnnyjax Thanks for replying. I will check this right now

Yoimer, 2021-03-11:
> Hello. It was a temperature sensor I used. I guess there will be different ways of handling this. If you just want to quickly get this running, just add the UncleRus as an extra component directory in the cmakelists [here](https://github.com/aws/amazon-freertos/blob/fd27d7bbd0d370004817c217b7fb95242f136b0d/demos/coreMQTT/CMakeLists.txt) using this
> 
> get_filename_component(
> EXTRA_COMPONENT_DIRS
> "unclerus_component_path"
> )
> If you've done that, you can directly include the components in the [mqtt_mutual_auth ](https://github.com/aws/amazon-freertos/blob/fd27d7bbd0d370004817c217b7fb95242f136b0d/demos/coreMQTT/mqtt_demo_mutual_auth.c)demo and adjust the topic and other parameters for your project. You can adjust [here](https://github.com/aws/amazon-freertos/blob/fd27d7bbd0d370004817c217b7fb95242f136b0d/demos/coreMQTT/mqtt_demo_mutual_auth.c#L822) too so that it sends you parameters to aws iot instead.

> Hello. It was a temperature sensor I used. I guess there will be different ways of handling this. If you just want to quickly get this running, just add the UncleRus as an extra component directory in the cmakelists [here](https://github.com/aws/amazon-freertos/blob/fd27d7bbd0d370004817c217b7fb95242f136b0d/demos/coreMQTT/CMakeLists.txt) using this
> 
> get_filename_component(
> EXTRA_COMPONENT_DIRS
> "unclerus_component_path"
> )
> If you've done that, you can directly include the components in the [mqtt_mutual_auth ](https://github.com/aws/amazon-freertos/blob/fd27d7bbd0d370004817c217b7fb95242f136b0d/demos/coreMQTT/mqtt_demo_mutual_auth.c)demo and adjust the topic and other parameters for your project. You can adjust [here](https://github.com/aws/amazon-freertos/blob/fd27d7bbd0d370004817c217b7fb95242f136b0d/demos/coreMQTT/mqtt_demo_mutual_auth.c#L822) too so that it sends you parameters to aws iot instead.

@Johnnyjax Hi. I just did what you suggested and that did not work unfortunately. Mind tell me which sensor you used and how you implemented is library? Thanks in advance.

yourslab, 2021-03-11:
Hi @Johnnyjax. For improving the build times, I would recommend using the [FreeRTOS Configuration Wizard](https://docs.aws.amazon.com/freertos/latest/userguide/freertos-ocw.html). It will allow you to create a customized distribution of FreeRTOS for esp32 that only contains the libraries you need. This should improve your build times significantly because only libraries being used will be built. Unfortunately, there is no way to do that by just using amazon-freertos as is from this repository because the project files need to contain every single source file.

Johnnyjax, 2021-03-15:
Thanks a lot @yourslab. I'll try that.

Johnnyjax, 2021-03-15:
Sorry for the late reply @Yoimer. I used an LM35 temperature sensor which uses the esp ADC. I didn't need to import any external components since the esp-idf already has a built in "driver/adc.h" library which I used.

Yoimer, 2021-03-15:
@Johnnyjax No need to apologize. In fact, thanks for responding. I was able to import the dht11 library on AWS FreeRTOS after long hours of reading and try and error.

Johnnyjax, 2021-03-15:
That's great. Congrats on getting that working. 👍🏽

closed 2021-03-15 by Johnnyjax