aws/amazon-freertos #2978: Further to post #832: provide a working template for novice CMake users to create a new ESP32 cmake project outside the FreeRTOS source tree

icrowe, 2021-02-07:
Issue #832 was about troubleshooting how to build an ESP32 project with FreeRTOS as a submodule.  The issue was closed, after documentation was improved.   

I am producing code for my company's new product, it is sensible to keep it separate from the FreeRTOS repository, and certainly not included as hacked demo code.

The various messages in the thread are somewhat instructive, but make it very clear that a CMake newbie is in for a steep learning curve.  That curve would be greatly eased with a working example.  This hypothetical example should show an ESP32 application being built from its own subdirectory structure - with stubs for multiple source files, private includes and preferably a ULP source.  The source files *must* be able to include and link in functions from esp-idf.
 
I've followed the steps in the Getting started guide, and various other threads to fiddle with the build environment enough to get simple things to compile.  But as soon as I start to add my own app code to call things from esp-idf everything falls apart.
  
For example, I added `#include esp_log.h` and a call to `ESP_LOGE()` to a c file.  Then found the cmakefile magic to get it to find the h-file.  AFR::logging needed to be added to target_link_libraries in the top level CMakeLists.txt, although I only found this out by guessing a module name.  Once that line was added At this point, the esp_log.h file was included, but the includes within that to more esp-idf internal files do not resolve.  I've been chasing down the CMakeLists.txt maze for several workdays now, and getting increasingly frustrated with the whole process.

Is there a published fork or specific howto that shows me how to approach this problem.  The examples in it should work as shown!  It should not assume an implicit under-graduate understanding of how cmake files work in the Amazon FreeRTOS implementation.  I can just about manage cmake to get my own files to build.  Other experts can (and do) maintain the Amazon portion, but ideally, it should be possible to include FreeRTOS with straightforward top level cmake commands rather than having to poke through the maze - that is the point of including it as a module.

Thank you!

yourslab, 2021-02-07:
I'm sorry to hear about your frustration. There are improvements underway to make consumption of the Amazon FreeRTOS libraries easier that won't requiring users to know too many unnecessary details about the CMake. For now, we have a working template here that you can use for the ESP32:
https://github.com/tgsong/amazon-freertos-examples.git

icrowe, 2021-02-08:
I cloned that template, and can confirm that it works as advertised.  It doesn't include ULP samples, but if that becomes a problem I will open a new issue.

Thank you very much.

closed 2021-02-08 by icrowe

genie-ltran, 2022-02-17:
The template works but only for one component.  Namely the "foo" addition, but I'd like to add more than one.  Is there a method to do this?  I initially thought it would take more than one, but I keep having issues locating the additional components.