aws/amazon-freertos #2998: Guidance for new projects - should LTS be used in preference to this repository?

thealastair, 2021-02-19:
**Is your feature request related to a problem? Please describe.**  
The recommended path forward for use of AWS Services (Shadow, OTA etc) with FreeRTOS is very unclear. It appears that this repository will soon become deprecated and https://github.com/FreeRTOS/FreeRTOS-LTS should be used going forward. This (LTS) repository does not yet contain the OTA library, and there is no clear example of using the OTA library with Shadow/direct MQTT access.

**Describe the solution you would like.**  
A clear and concise demo/example of using AWS Shadow/OTA and direct MQTT with (specifically) the ESP32 platform.

**Additional context**  
Our major reason for wanting to upgrade AWS libraries/FreeRTOS is to allow a single MQTT connection to be used for Shadow/ OTA and direct MQTT publishing. This is not supported with the version of FreeRTOS we are currently using (1.4.7). Our device is battery powered and only periodically within wifi range, we would like to reduce the time required to synchronise shadow and check for OTA jobs, and our intended approach is to reduce the connection time by sharing a connection for these services.

leegeth, 2021-02-24:
@thealastair,

Thanks for bring these questions and concerns to our attention. Please find answers to your questions below.
1. First of all, this repository is not going to be deprecated and will serve as a channel for the FreeRTOS reference integrations of the FreeRTOS and AWS libraries. Please find the FreeRTOS page about reference integrations [here](https://freertos.org/aws-reference-integrations.html). 
2. OTA library doesn't have a LTS release yet. However, we are working on a LTS release for OTA library currently. 
3. About the connection sharing, we are working on MQTT Agent APIs to enable MQTT connection sharing between multiple tasks. Please refer to the [Lab-Project-coreMQTT-Agent](https://github.com/FreeRTOS/Lab-Project-coreMQTT-Agent) for the MQTT Agent APIs and the demo applications for connection sharing. Note that this is a work in progress, but we are not expecting a drastic change in the MQTT Agent APIs. We are currently working on a plan to release these APIs. 

Please let us know if you have any further questions about this.

abhidixi11, 2021-03-03:
Hello @thealastair ,

Just wanted to share that OTA library is now part LTS release and it available to download from [here](https://freertos.org/lts-libraries.html) or from GitHub [repository ](https://github.com/FreeRTOS/FreeRTOS-LTS)

Thanks!

thealastair, 2021-03-03:
Thanks for the response, we'll check it out.

dan4thewin, 2021-03-08:
I'll close this issue now.  Please feel free to open another if you help.

closed 2021-03-08 by dan4thewin