aws/amazon-freertos #3000: FreeRTOS initialation files

arantza12345, 2021-02-19:
Hi! 

I recently bought a nxp kit that is activated by saying a wake up word, so recognsies voices. This kit has a wifi and bluetooth module that I would like to turn on but though I have been trying I haven´t been susccessful. I am working with freertos and MCUExpresso IDE. 

So nxp provided me a folder that does have a wifi configuration, with the different libraries needed, different folders... but when I open the SDK in MCUExpresso the folders that appears as available in the demo project doesn´t include the FreeRTOS + TCP library nor any file referencing any wifi configuration. 

I tried different things in mcu. Firstly,  deleting the sdk demo files and including the ones nxp gave me (the ones that include the wifi configuration) but didn´t work because some files of the sdk initial demo were needed. Then I copied all of the files with the demo configuration, but a lot of files were duplicated and gave me a lot of problems. And then I tried including the files one by one but it s a really tedious work that would take me for ages. 

The explanations I have seen to power on the wifi with freertos do not say exactly how this file inclusion should be done, but the steps that need to be taken in order to turn on the wifi don´t seem to be difficult but errors don´t stop to appear. 

In addition, the code has more than one main.c file and I don´t if there is any possibility to decide which main to execute. 

Thank you in advanced!

leegeth, 2021-02-23:
@arantza12345 

Thanks for bringing this issue to our attention. I have a few follow up questions to better help you.
1. Which NXP platform are you using? The NXP platform([LPC54018](https://devices.amazonaws.com/detail/a3G0L00000AANtAUAX/LPC54018-IoT-Solution)) integrated in this repo supports only Wifi connectivity. So I believe you are using a different platform which has a Bluetooth connectivity as well.
2. Which FreeRTOS version are using with this device?

arantza12345, 2021-02-23:
Answering to your first question, I might not be right about the bluetooth connectivity because the nxp platform I am using is the LPC54018. Regarding the second question, I am using the 1.4 freertos version. 

Thank you for answering !

leegeth, 2021-02-23:
@arantza12345,

Thanks for getting back to us with all the details. A few more questions/comments to help you better.
1. What is the code base you are using? Is it provided by NXP or is it [amazon-freertos 1.4.0](https://github.com/aws/amazon-freertos/tree/v1.4.0) itself? If it is different from amazon-freertos 1.4.0, would you be able to share the code with us, so that we can take a look at what exactly is happening?
2. NXP doesn't use FreeRTOS+TCP. Instead, it uses a custom hardware for TCP. Wifi end point configurations are done in the header file [aws_clientcredential.h](https://github.com/aws/amazon-freertos/blob/v1.4.0/demos/common/include/aws_clientcredential.h). Is this file present in the code that you are using?
3. Have you tried adding just the new files to the already existing[ project file in amazon-freertos 1.4.0](https://github.com/aws/amazon-freertos/tree/v1.4.0/demos/nxp/lpc54018iotmodule/mcuxpresso)?

arantza12345, 2021-03-10:
@leegeth,

The code I am using is provided by NXP. What is the difference between the two options you have mentioned ?

The aws_clientcredential.h is present in my code. And what  I have tried is introducing the files that I need (or think I need) in this program that I know it already works. I am trying to upload the code but I´m not able to attach files because the .zip files are to big.

arantza12345, 2021-03-10:
I am using mcu expresso, but have seen that there is another IDE called WICED that I really don´t know if it would change something or what is it for. Does anyone know?

leegeth, 2021-04-09:
@arantza12345,

Apologies for the delay in getting back to you.  Please find my responses below.
1. If it is a version of amazon-freertos you have received from NXP, it is possible that there are some modifications done by NXP on it. It would be better if we can take a look at it. You may be able to share it by either pushing to a github repo and sharing the link with us or sharing it through a file sharing platform.
2. iar and mcuxpresso are supported by amazon-freertos for NXP. WICED is not a supported IDE yet in amazon-freertos for NXP.
3. Could you also try referring to the [project file exists in this repo](https://github.com/aws/amazon-freertos/tree/v1.4.0/demos/nxp/lpc54018iotmodule/mcuxpresso) as a base for creating your project?

paulbartell, 2021-04-26:
Closing this issue. Feel free to re-open if you have additional questions.

closed 2021-04-26 by paulbartell