aws/amazon-freertos #3001: AFR & ESP32: Correction in the error message for the esp_wifi_get_mode() failure

dipen-1337lab, 2021-02-19:
**Describe the bug**
Within the `WIFI_ConnectAP()` function of the `iot_wifi.c` source file, the below failure message is incorrect for the `esp_wifi_get_mode()` function.

```
ret = esp_wifi_get_mode( &xCurMode );
        if (ret != ESP_OK) {
            ESP_LOGE(TAG, "%s: Failed to set wifi mode %d", __func__, ret);
            xSemaphoreGive( xWiFiSem );
        	return eWiFiFailure;
        }
```

**System information**
- Hardware board: ESP32-DevKitC
- IDE used: N/A
- Operating System [Windows 10 x64 -- Version: 20H2 -- OS Build: 19042.804]
- Version of FreeRTOS: latest (master branch)
- Project: N/A

**Expected behavior**
The error message should be corrected as below.
`ESP_LOGE(TAG, "%s: Failed to get wifi mode %d", __func__, ret);`

**Screenshots or console output**
N/A

**To reproduce**
The latest code in the master branch has this incorrect message, as can be viewed [here](https://github.com/aws/amazon-freertos/blob/master/vendors/espressif/boards/ports/wifi/iot_wifi.c#L480).

**Code to reproduce the bug**
N/A

**Additional context**
N/A

Thanks | Regards,
Dipen

leegeth, 2021-02-24:
@dipen-1337lab ,

Thanks for pointing out this issue. We have merged a PR with the fix to master branch. Please let us know if you have any further questions.

dipen-1337lab, 2021-02-24:
Namaste @leegeth,

You are welcome. Happy to have helped improve **amazon-freertos** in a small way.

Thanks | Regards,
Dipen

closed 2021-02-24 by dipen-1337lab