aws/amazon-freertos #3002: How to build, compile and flash AWS FreeRTOS ESP32 on VS Code

Yoimer, 2021-02-21:
Hello crew,

Is there any manual of how to build, program, and flash AWS FreeRTOS ESP32 on VS Code? How can we configure CMake on VScode in order to flash AWS FreeRTOS ESP32 projects?

All the info I have is this:
https://www.freertos.org/2021/01/using-visual-studio-code-for-freertos-development.html

But it is not complete yet. I would really appreciate any comments on this. I've been searching (and even consulted on https://forums.freertos.org/t/how-to-compile-an-aws-freertos-esp32-project-using-vs-code/11785) and I have no answers so far. 

Thanks in advance.

lundinc2, 2021-02-25:
Hello @Yoimer,

Espressif provides documentation for setting up an IDF project. You can set up this repository by following these guides.

[Espressif VS Code extension getting started guide](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/get-started/vscode-setup.html).
[VS Code extension GitHub forum](https://github.com/espressif/vscode-esp-idf-extension/#quick-installation-guide).
[A sample config](https://gist.github.com/motla/ab7fdcf14303208996c40ca7fefe6f11). Note for this one you need to change the paths to `$AMAZON_FREERTOS_ROOT/vendors/espressif/esp-idf/export.sh.

In your terminal, you can bootstrap the file paths needed by the ESP-IDF VS Code by using the following commands.
```bash
source $AMAZON_FREERTOS_ROOT/vendors/espressif/esp-idf/export.sh
code $AMAZON_FREERTOS_ROOT
```

I added `"CMAKE_EXPORT_COMPILE_COMMANDS": "1",` to the `settings.json` mentioned in the [FreeRTOS post](https://forums.freertos.org/t/how-to-compile-an-aws-freertos-esp32-project-using-vs-code/11785) you mention on VS Code. This will create a [compile_commands.json](https://cmake.org/cmake/help/latest/variable/CMAKE_EXPORT_COMPILE_COMMANDS.html) that is parsed by VS Code in order to configure intellisense.

Hope this is helpful.

Thanks,

Carl

Yoimer, 2021-02-25:
Hello @lundinc2 Thanks a lot for replying. I'm still a bit confused though. I really hope you could keep with further explanation, please. 
On [A sample config](https://gist.github.com/motla/ab7fdcf14303208996c40ca7fefe6f11) they don't say anything related to  `CMake Extension` [CMake Tools](https://marketplace.visualstudio.com/items?itemName=ms-vscode.cmake-tools) and the `"CMAKE_EXPORT_COMPILE_COMMANDS": "1"` configuration you commented, has to with that. Should I install it anyways?

About the `$AMAZON_FREERTOS_ROOT/vendors/espressif/esp-idf/export.sh`. I use Windows, so I successfully ran: `C:\esp-idf-v4.2\FreeRTOS\vendors\espressif\esp-idf\export.bat` and added these tools to my `path` so I can run `cmake`, `xtensa-esp32-elf-gcc` and the rest of the tools on any path on my PC. VS code even recognizes the compiler. But still I can't build from it.

I can run from `cmd` the demos without any issues. The commands I use are:

from `C:\esp-idf-v4.2\FreeRTOS`

```
cmake -DVENDOR=espressif -DBOARD=esp32_devkitc -DCOMPILER=xtensa-esp32 -GNinja -S . -B demos/coreMQTT/build
python C:/esp-idf-v4.2/FreeRTOS/vendors/espressif/esp-idf/tools/idf.py build -B demos/coreMQTT/build
python C:/esp-idf-v4.2/FreeRTOS/vendors/espressif/esp-idf/tools/idf.py erase_flash flash monitor -p COM4 -B demos/coreMQTT/build
```
And from AWS I can push and subscribe flawlessly.

How and where should I write those commands on VS code to make possible the building, compiling and flashing? Thanks in advance.

lundinc2, 2021-02-25:
Hi @Yoimer,

I used the CMake tools extension alongside the IDF extension, in order to generate the proper cmake when I opened VS Code. 

Can you clarify how you are compiling from VS Code? 

Thanks,

Carl

Yoimer, 2021-02-25:
@lundinc2 Hello,

I will paste here later, how I am trying to compile the AWS FreeRTOS ESP32 on VS Code. Also (as mere reference) how I compile the native (vanilla) demos from Espressif IDF. This last works out of box with no issues at all. I don't even have to configure anything when I test those.

Yoimer, 2021-03-21:
@lundinc2 I hope you're doing great. I just opened this https://github.com/espressif/vscode-esp-idf-extension/issues/355 where I gave specific details of what I do regarding VS Code and AWS FreeRTOS ESP32 now using `ESP-IDFV3.3.2.` Please, could you be kind enough to check it out? I just forgot that I already had this thread here and created that issue there. I don't want to close it though. Thanks in advance for your assistance

Yoimer, 2021-03-30:
Hello @lundinc2. 

Now, I am able to build and compile `FreeRTOS V202002.00` from VSCODE. I am not using `Espressif VS Code extension`. Instead I just installed `https://dl.espressif.com/dl/esp-idf-tools-setup-1.2.exe` on my Windows path along the latest version of `CMake` (3.20.0). 

On VScode I installed this extension https://marketplace.visualstudio.com/items?itemName=ms-vscode.cmake-tools.

On `FreeRTOS\.vscode\settings.json` I have this configuration:
```
{
    "cmake.configureOnOpen": false,
    "cmake.buildDirectory": "${workspaceFolder}/demos/mqtt/build",
    "cmake.sourceDirectory": "${workspaceFolder}",
    "cmake.buildArgs": [
        "-GNinja -S . -B"
    ],
    "cmake.configureSettings": {
        "COMPILER": "xtensa-esp32",
        "CMAKE_C_COMPILER_WORKS": "TRUE",
        "VENDOR": "espressif",
        "BOARD": "esp32_devkitc"
    },
    "C_Cpp.default.configurationProvider": "ms-vscode.cmake-tools"
}
```
Which maps this command as it were used from `cmd`
Go to FreeRTOS folder
`cmake -DVENDOR=espressif -DBOARD=esp32_devkitc -DCOMPILER=xtensa-esp32 -GNinja -S . -B demos/mqtt/build`

On VSCode, I click on `Compilation Version` button and it begins compiling all.

The output is this:

```
[main] Compilando la carpeta: FreeRTOS 
[main] Configurando la carpeta: FreeRTOS 
[proc] Ejecutando el comando: C:\cmake-3.20.0\bin\cmake.EXE --no-warn-unused-cli -DCOMPILER:STRING=xtensa-esp32 -DCMAKE_C_COMPILER_WORKS:STRING=TRUE -DVENDOR:STRING=espressif -DBOARD:STRING=esp32_devkitc -DCMAKE_EXPORT_COMPILE_COMMANDS:BOOL=TRUE -DCMAKE_BUILD_TYPE:STRING=Debug -DCMAKE_C_COMPILER:FILEPATH=C:\Espressif\ESP-IDF-Tools\tools\bin\xtensa-esp32-elf-gcc.exe -DCMAKE_CXX_COMPILER:FILEPATH=C:\Espressif\ESP-IDF-Tools\tools\bin\xtensa-esp32-elf-g++.exe -Hc:/Users/Yoimer/Documents/development/poc/esp32-aws-freertos/FreeRTOS -Bc:/Users/Yoimer/Documents/development/poc/esp32-aws-freertos/FreeRTOS/demos/mqtt/build -G Ninja
[cmake] Not searching for unused variables given on the command line.
[cmake] -- The C compiler identification is GNU 5.2.0
[cmake] -- The CXX compiler identification is GNU 5.2.0
[cmake] -- The ASM compiler identification is GNU
[cmake] -- Found assembler: C:/Espressif/ESP-IDF-Tools/tools/bin/xtensa-esp32-elf-gcc.exe
[cmake] -- Found Git: C:/git/cmd/git.exe (found version "2.31.0.windows.1") 
[cmake] -- Checking Python dependencies...
[cmake] Python requirements from C:/Users/Yoimer/Documents/development/poc/esp32-aws-freertos/FreeRTOS/vendors/espressif/esp-idf\requirements.txt are satisfied.
[cmake] -- IDF_TARGET not set, using default target: esp32
[cmake] -- Building for target esp32
[cmake] -- Component names: soc log heap freertos vfs newlib esp_ringbuf driver tcpip_adapter esp_event partition_table app_update spi_flash mbedtls micro-ecc bootloader_support efuse xtensa-debug-module app_trace ethernet nvs_flash pthread smartconfig_ack wpa_supplicant espcoredump esp32 cxx amazon-freertos-common secure_sockets freertos_plus_tcp bootloader nimble bt console esp_adc_cal esptool_py expat wear_levelling sdmmc fatfs freemodbus lwip nghttp openssl spiffs ulp
[cmake] -- Component paths: C:/Users/Yoimer/Documents/development/poc/esp32-aws-freertos/FreeRTOS/vendors/espressif/esp-idf/components/soc;C:/Users/Yoimer/Documents/development/poc/esp32-aws-freertos/FreeRTOS/vendors/espressif/esp-idf/components/log;C:/Users/Yoimer/Documents/development/poc/esp32-aws-freertos/FreeRTOS/vendors/espressif/esp-idf/components/heap;C:/Users/Yoimer/Documents/development/poc/esp32-aws-freertos/FreeRTOS/vendors/espressif/boards/esp32/components/freertos;C:/Users/Yoimer/Documents/development/poc/esp32-aws-freertos/FreeRTOS/vendors/espressif/esp-idf/components/vfs;C:/Users/Yoimer/Documents/development/poc/esp32-aws-freertos/FreeRTOS/vendors/espressif/esp-idf/components/newlib;C:/Users/Yoimer/Documents/development/poc/esp32-aws-freertos/FreeRTOS/vendors/espressif/esp-idf/components/esp_ringbuf;C:/Users/Yoimer/Documents/development/poc/esp32-aws-freertos/FreeRTOS/vendors/espressif/esp-idf/components/driver;C:/Users/Yoimer/Documents/development/poc/esp32-aws-freertos/FreeRTOS/vendors/espressif/boards/esp32/components/freertos_tcpip/tcpip_adapter;C:/Users/Yoimer/Documents/development/poc/esp32-aws-freertos/FreeRTOS/vendors/espressif/esp-idf/components/esp_event;C:/Users/Yoimer/Documents/development/poc/esp32-aws-freertos/FreeRTOS/vendors/espressif/esp-idf/components/partition_table;C:/Users/Yoimer/Documents/development/poc/esp32-aws-freertos/FreeRTOS/vendors/espressif/esp-idf/components/app_update;C:/Users/Yoimer/Documents/development/poc/esp32-aws-freertos/FreeRTOS/vendors/espressif/esp-idf/components/spi_flash;C:/Users/Yoimer/Documents/development/poc/esp32-aws-freertos/FreeRTOS/vendors/espressif/boards/esp32/components/mbedtls;C:/Users/Yoimer/Documents/development/poc/esp32-aws-freertos/FreeRTOS/vendors/espressif/esp-idf/components/micro-ecc;C:/Users/Yoimer/Documents/development/poc/esp32-aws-freertos/FreeRTOS/vendors/espressif/esp-idf/components/bootloader_support;C:/Users/Yoimer/Documents/development/poc/esp32-aws-freertos/FreeRTOS/vendors/espressif/esp-idf/components/efuse;C:/Users/Yoimer/Documents/development/poc/esp32-aws-freertos/FreeRTOS/vendors/espressif/esp-idf/components/xtensa-debug-module;C:/Users/Yoimer/Documents/development/poc/esp32-aws-freertos/FreeRTOS/vendors/espressif/esp-idf/components/app_trace;C:/Users/Yoimer/Documents/development/poc/esp32-aws-freertos/FreeRTOS/vendors/espressif/boards/esp32/components/free [... cut ...]
[build] esptool.py v2.8-dev
[build] Compilación finalizada con el código de salida 0
```
It generates the essential binaries needed for flashing `aws_demos.bin`, `bootloader.bin`, `partition-table.bin` (and many others)

So my question is:

How could flash the ESP32 directly from VScode?

Based on my `settings.json` file, how could I pass commands like: 

`cmake --build demos/mqtt/build --target flash`

or 

`python vendors/espressif/esp-idf/tools/idf.py build -B demos/mqtt/build`

Thanks in advance for your help.

muneebahmed10, 2021-03-31:
Hi @Yoimer,

I have not tried flashing from VS Code, but have you tried configuring your [tasks.json](https://code.visualstudio.com/docs/editor/tasks) to create a task for flashing the board? You can run just the task (Ctrl-Option-T on Mac, it may be the same on Windows) or configure [launch.json](https://code.visualstudio.com/docs/cpp/launch-json-reference) for debugging, and add flashing as a pre launch task.

Additionally, have you looked into [PlatformIO IDE for VS Code](https://docs.platformio.org/en/latest/platforms/espressif32.html#tutorials)? The extension includes building and flashing, and they seem to have support for [ESP32-DevKitC](https://docs.platformio.org/en/latest/tutorials/espressif32/espidf_debugging_unit_testing_analysis.html#tutorial-espressif32-espidf-debugging-unit-testing-analysis).

Yoimer, 2021-03-31:
@muneebahmed10 Hello. Thanks a lot for your support. Actually I was able to compile, debug and even flash my AWS FreeRTOS ESP32 project from VSCODE. Using `ESP-IDF3.3`

Kindly, check the `.vscode.zip` attached.

Now my goal is to debug AWS FreeRTOS ESP32 projects from VSCODE, I want to make breakpoints, check variable values on the ESP32. Currently, I own this `Debugger`, which is the official one from Espressif [ESP-Prog JTAG Debug & Program Downloader](https://www.aliexpress.com/item/32905400681.html). 

I know AWS doc talks a bit about this here https://docs.aws.amazon.com/freertos/latest/userguide/getting-started-cmake.html#developing-third-party, but they mention something about  `stm32l475_discovery` which is another vendor.

The reference for ESP32, is there too but merely from Command Line Interface:
https://docs.aws.amazon.com/freertos/latest/userguide/getting_started_espressif.html#getting_started_espressif_debugging

So my question is, should I follow the same instructions for AWS FreeRTOS as I were debugging a regular project on ESP-IDF 3.3? I want to keep using VS Code as my IDE.

Regarding `PlatformIO`, I haven't tried that yet. But eventually I will in the future.

Looking forward to hearing from you. Thanks a lot for your kind support.

[vscode.zip](https://github.com/aws/amazon-freertos/files/6239246/vscode.zip)

cobusve, 2021-05-17:
I have used VSCode before with OpenOCD to set up debugging. You can find more information on that over here https://docs.espressif.com/projects/esp-idf/en/v3.3.2/api-guides/jtag-debugging/setup-openocd-windows.html 

What you are looking for is not specific to freertos, but rather specific to VSCode or ESP IDF tools, so this may not be the best place for you to get help. That link lundinc posted to the VSCode extention with debug adaptor for VSCode should help.

Yoimer, 2021-05-17:
> I have used VSCode before with OpenOCD to set up debugging. You can find more information on that over here [...]

@cobusve Hi, thanks for commenting. Fortunately, I was able to figure out how to make breakpoints, step through, in and out on AWS FreeRTOS ESP32, using that `OpenOCD` software. I did it using VSCODE as my UI.

Thanks again for commenting.

aditi230, 2021-06-13:
> @cobusve Hi, thanks for commenting. Fortunately, I was able to figure out how to make breakpoints, step through, in and out on AWS FreeRTOS ESP32, using that `OpenOCD` software. I did it using [...]

Hey, Can you guide for same. I am trying AWS Free RTOS Esp32 on VS Code and I am stuck at a step: "Build, flash, and run the FreeRTOS demo project" . I followed [https://docs.aws.amazon.com/freertos/latest/userguide/getting_started_espressif.html#setup-espressif-idf33] this documentation.
I will be very helpful, Thank you in advance.

muneebahmed10, 2021-07-21:
Hi @aditi230,

If using VS Code, I'd suggest configuring [tasks.json](https://code.visualstudio.com/docs/editor/tasks) and defining tasks for running any shell commands that you need.

As @cobusve pointed out, this probably isn't the best place to get help for VS Code issues, so I would suggest trying another forum that's focused towards VS Code if you need further help.

closed 2021-07-24 by yourslab