aws/amazon-freertos #3004: Adding ESP-IDF components inside AFR

anubhav1, 2021-02-21:
Sorry for opening an issue. It is a question. 

How to add esp-idf components like [esp32-camera-screen](https://github.com/Xinyuan-LilyGo/esp32-camera-screen) or [esp-who](https://github.com/espressif/esp-who) inside AFR. I want to use an esp32 camera based module like ESP32-EYE or [ESP32-T-Camera-Plus](https://www.amazon.de/-/en/gp/product/B07S38R8PR/ref=ox_sc_act_title_1?smid=A2V746ST1SO6EN&psc=1).

Johnnyjax, 2021-02-21:
I think you can follow the steps [here](https://docs.aws.amazon.com/freertos/latest/userguide/getting_started_espressif.html#getting_started_espressif_cmake_project) to figure out how to add your own custom components. The folder structure should look like [this](https://github.com/tgsong/amazon-freertos-examples) when you're done. You can just add the components for your project into the components folder of your freertos project and adjust the cmakelist file. You will then be able to call any function you need from your main.c file.

anubhav1, 2021-02-23:
I am able to make it work for a simple hello_world project. Now i am not able to figure out, how to add header files which are inside `src/include` folder in CMakeLists.txt.
My project structure:

```
- freertos
- src
  -include
    -headerFile.h (How to add this file in CMakeLists.txt?)
  - main.c 
- CMakeLists.txt
```

Johnnyjax, 2021-02-23:
If your project looks like this [here](https://github.com/tgsong/amazon-freertos-examples), you should be able to edit the cmakelists file to set your component path. Since yours is in the src/include folder, just replace
get_filename_component(
    EXTRA_COMPONENT_DIRS
    "components/foo" ABSOLUTE
)
with 
get_filename_component(
    EXTRA_COMPONENT_DIRS
    "src/include" ABSOLUTE
)
in the cmakelists file

anubhav1, 2021-02-23:
Then what will be the solution if my project  has a component too:

```
- freertos
- components
  - foo
    - include
      - foo.h
    - src
      - foo.c
    - CMakeLists.txt
- src
  -include
    -app_priv.h (How to add this file in CMakeLists.txt?)
  - main.c
- CMakeLists.txt
```

I tried this in CMakeLists.txt file which is in the root folder of the project:

```
get_filename_component(
    EXTRA_COMPONENT_DIRS
    "components/foo" ABSOLUTE
)

list(APPEND IDF_EXTRA_COMPONENT_DIRS ${EXTRA_COMPONENT_DIRS})

get_filename_component(
    EXTRA_COMPONENT_DIRS
    "src/include" ABSOLUTE
)

list(APPEND IDF_EXTRA_COMPONENT_DIRS ${EXTRA_COMPONENT_DIRS})
```

But getting error :  
`fatal error: app_priv.h: No such file or directory`

`app_priv.h` is the header file in `src/include folder`.

Johnnyjax, 2021-02-23:
The easiest way to fix this will be to have a separate components directory like is shown [here](https://github.com/tgsong/amazon-freertos-examples)
- freertos
- components
  - foo
    - include
      - foo.h
    - src
      - foo.c
    - CMakeLists.txt
- src
  -include
    -app_priv.h (How to add this file in CMakeLists.txt?)
  - main.c
- CMakeLists.txt
 instead of having the include directory in your src folder. You can rename the foo directory to whatever you want and if you have other components, you can add them into the components folder. delete the foo.h and foo.c files too and add your own components to the folder. You can then edit the cmakelists files in the components/YOUR_COMPONENT/cmakelists.txt for your project. Which means you replace the foo.h and foo.c with your include file names instead

Johnnyjax, 2021-02-23:
If you delete the foo.h and foo.c files, you'll need to remove any reference to them in the main.c file. i.e remove the include line for the foo.h or wherever else it is used in the main.c file

leegeth, 2021-02-23:
Thanks @Johnnyjax for helping out with this question. Just thought of adding one more point related to adding include path following the approach tried out. 
@anubhav1, Have you tried out adding the include path as shown in [CMakeLists.txt here](https://github.com/tgsong/amazon-freertos-examples/blob/master/CMakeLists.txt#L21)?
In your case it will be
```include_directories(BEFORE freertos-configs src/include)```

anubhav1, 2021-02-23:
Yes I just found out myself.  Thanks anyway!

> Thanks @Johnnyjax for helping out with this question. Just thought of adding one more point related to adding include path following the approach tried out.
> @anubhav1, Have you tried out adding the [...]

closed 2021-02-23 by anubhav1