aws/amazon-freertos #3005: Even thought OTA jobs are scheduled from the AWS cloud, Some times OTA jobs does not start on the device.

kumarfirmware, 2021-02-22:
Hi Team,

I am using AWS FreeRTOS version 202007 and ESP32.

I am seeing an issue, Some times OTA jobs does not start on the device.

Even thought OTA jobs are scheduled from the AWS cloud, OTA jobs does not start on the Device(ESP).

On debugging I could see the problem, device is subscribing to the topics  jobs/notify-next,  jobs/$next/get/accepted, but it is not publishing the topic jobs/$next/get and jobs/$next/get/accepted some times. I could see clearly this is the issue.

This does not happen always, this happens occasionally. But making over all system unstable.

This issue is clearly happening from OTA library.

Can you please check this as soon as possible. This is an urgent issue. This is stopping the release.

Need your support as soon as possible.

Thanks,
Kumar.

pvyawaha, 2021-02-23:
Hello Kumar

Which network are you using BLE or WiFi?

Also if you can provide the complete logs it will be helpful.

kumarfirmware, 2021-02-23:
Hi pvyawaha,

Thanks for your immediate response.

I saw this issue multiple times randomly while using BLE network.

You mean to say device will get the jobs in two possible ways.

1. At the start Device will publish to the topic jobs/$next/get and jobs/$next/get/accepted and get the jobs.
2. This publish is not periodic and it is done only once at the start of OTA.
3. How does the device come to know that jobs are scheduled or pending? Through the notification from the cloud through the subscription. Can you explain what notification cloud will send?
4. What is the chance of notification getting missed and this problem occurring? Is this possible?

Since it is very random difficult to capture OTA logs.

You can refer to the cloud watch log attached here, which will give the clear picture. I doubt it will give any clue, but this is the exact issue occurrence.

[logs.pdf](https://github.com/aws/amazon-freertos/files/6031365/logs.pdf)

Thanks,
Kumar.

pvyawaha, 2021-02-23:
Hello,

The BLE solution uses a middleware between device and the cloud and my colleague @ravibhagavandas can offer more help.

ravibhagavandas, 2021-02-23:
Hi Kumar, 
The mobile device in case of BLE network acts as a proxy for MQTT messages, so if the device has subscribed with a Qos1 subscription then the subscription is relayed to broker. Also if broker sends a Qos1 message to device the PUBACK for this message is relayed to broker after device has received the PUBLISH message. So its unlikely that message is lost if there is a valid connection between ESP32 and mobile device.

Could you forward the device side logs, and the mobile phone logs (if possible) when the issue occurs ? Do you see any BLE disconnects or reconnects, while the issue is happening?

kumarfirmware, 2021-02-23:
Hi Ravi,

Thanks for your immediate response.

Since it is very random difficult to capture OTA logs.

You can refer to the cloud watch log attached here, which will give the clear picture. I doubt it will give any clue, but this is the exact issue occurrence.

So the only possibility of jobs not starting on the device is when this notify is failed to be delivered on the device. Please confirm. What are other possible causes you can think of?

Thanks,
Kumar
[logs.pdf](https://github.com/aws/amazon-freertos/files/6031898/logs.pdf)

ravibhagavandas, 2021-02-23:
Thanks for the logs.
It seems from logs on cloud side that the MQTT connection gets dropped after it subscribes to a topic `jobs/$next/get/accepted` and then reconnects and resubscribes to the topic. This can happen whenever the proxy mobile device losses connectivity with the ESP32 device over BLE, it drops the connection with cloud as well. Could you check anyways if the mobile device to ESP32 connection is stable ?  Is there a way to get the proxy device logs to know that there was a BLE disconnect?

kumarfirmware, 2021-02-24:
Hi Ravi,

Since it is a random issue there are no logs, only logs available were cloud watch logs which I already shared with you.

But you are not concluding anything? what are the possible causes and what are the possible failures?

When is the possibility for this issue to occur? why is it not able to recover?

If we discuss this only we can solve this issue. If we keep waiting for logs we cannot progress, because it is a random issue, very difficult to get logs.

"This can happen whenever the proxy mobile device losses connectivity with the ESP32 device over BLE, it drops the connection with cloud as well."

What do you mean by this? when there is BLE disconnect between ESP32 and mobile why MQTT connection between the cloud and Mobile will drop? I am slightly confused here. Can you explain in detail how this proxy works?

Also why will the BLE connection drop or become unstable between ESP32 and mobile when they are at a close distance of 1meter?

So this is a big bottle neck, if notify does not reach device by mistake, jobs will never reach device? Is it not a poor design, should there not be a periodic check from device at least from the OTA library for any existence of jobs?

Thanks,
Kumar.

ravibhagavandas, 2021-02-24:
> What do you mean by this? when there is BLE disconnect between ESP32 and mobile why MQTT connection between the cloud and Mobile will drop? I am slightly confused here. Can you explain in detail how this proxy works?

The mobile phone keeps MQTT connection with cloud only for exchanging messages with devices over BLE. So when device disconnects over BLE, it need not keep the connection open. The connection  is closed to free up the resources.

As you can see from the logs, device reconnects, and resubscribes to topics so it recovers from the disconnect. 

But the real issue is it keeps disconnecting again. 

> So this is a big bottle neck, if notify does not reach device by mistake, jobs will never reach device? Is it not a poor design, should there not be a periodic check from device at least from the OTA library for any existence of jobs?
As said before Notifications are send by Qos1 publishes and if it does not reach device it could retransmit the notification as per the MQTT protocol. 

When is the possibility for this issue to occur? why is it not able to recover?
We have seen a similar issue with ESP32 device recently that when it receives a job document through OTA and prepares the flash partition for writing, it sometimes fails to send a BLE heart beat within the connection interval time negotiated for BLE. This causes a BLE timeout and  a disconnect. 

Here is the traces for the issue after enabling ESP debug log on device:
```
host tx hci data; handle=0 length=5
ble_hs_hci_acl_tx(): 0x00 0x00 0x05 0x00 0x01 0x00 0x04 0x00 0x13
I (51746) ota_pal: prvPAL_GetPlatformImageState
I (51746) esp_ota_ops: aws_esp_ota_get_boot_flags: 1
I (51746) esp_ota_ops: [0] aflags/seq:0x2/0x1, pflags/seq:0xffffffff/0x0
I (51756) ota_pal: Writing to partition subtype 17 at offset 0x1a0000
Number of Completed Packets: num_handles=1
handle:0 pkts:1
80 5078 [OTA Agent Task] [prvParseJSONbyModel] Extracted parameter [ timestamp: 1611800466 ]
81 5078 [OTA Agent Task] [prvParseJSONbyModel] Extracted parameter [ jobId: AFR_OTA-xc ]
82 5078 [OTA Agent Task] [prvParseJSONbyModel] Extracted parameter [ protocols: ["MQTT"] ]
83 5078 [OTA Agent Task] [prvParseJSONbyModel] Extracted parameter [ streamname: AFR_OTA-4923b5c3-6dbf-477a-89c0-58995b7f1660 ]
84 5078 [OTA Agent Task] [prvParseJSONbyModel] Extracted parameter [ filepath: dsaasdsa ]
85 5078 [OTA Agent Task] [prvParseJSONbyModel] Extracted parameter [ filesize: 815936 ]
86 5078 [OTA Agent Task] [prvParseJSONbyModel] Extracted parameter [ fileid: 0 ]
87 5078 [OTA Agent Task] [prvParseJSONbyModel] Extracted parameter [ certfile: /NA ]
88 5078 [OTA Agent Task] [prvParseJSONbyModel] Extracted parameter [ sig-sha256-ecdsa: MEYCIQC33dkrDdWe8LFsVZBiFHLzspUW... ]
89 5078 [OTA Agent Task] [prvParseJobDoc] Job was accepted. Attempting to start transfer.
90 5095 [iot_thread] [INFO ][DEMO][50950] State: WaitingForJob  Received: 2   Queued: 0   Processed: 0   Dropped: 0
 
Number of Completed Packets: num_handles=1
handle:0 pkts:1

Disconnection Complete: status=0 handle=0 reason=8

Revised our id addr:
0x8a 0x84 0x5c 0xd3 0x15 0x58
Revised our id addr:
0x8a 0x84 0x5c 0xd3 0x15 0x58
 ```` 

The logs  you posted from cloud watch  also indicates similar issue (because the disconnect happens right after a job is requested, and a job document is sent to the device.
```
14:52:10 Publish-in jobs/$next/get.   ==> Job Request
14:52:10 Publish-Out jobs/$next/get/accepted. == > Job document sent to device
Mqtt drop, disconnected and connect, subscribe to all topic from device
````

This issue occurs occasionally  only when the connection interval value negotiated (depending on the mobile) is small and the ESP flash partition erase time exceeds that.

We are working with ESP team to fix this issue, but to confirm if this issue at your side, we need logs from device side.

ravibhagavandas, 2021-02-24:
If this is the issue, as a short term workaround you could update the supervision timeout value to a larger value so that the timeout does not occurs during the partition erase. This can be done using the connection parameter update request API , inside the BLE connected callback as follows:
```
void _connectionCb( uint16_t connId,
                    uint8_t serverIf,
                    bool connected,
                    BTBdaddr_t * pBda )
{
    BTStatus_t status = eBTStatusSuccess;

    BTInterface_t * pBTInterface =  BTGetBluetoothInterface();
    BTBleAdapter_t * pBleAdapter = pBTInterface->pxGetLeAdapter();

    if( connected == true )
    {

        status = pBleAdapter->pxConnParameterUpdateRequest( pBda, 0x10 /* min interval */, 0x20 /* maxInterval */, 0x00 /* latency */, 600 /* Timeout */);
        configASSERT(status == eBTStatusSuccess);
   }

 ......
 ......
 }
```

This sends a connection parameter update request to central with the preferred values as above. However the final timeout value will be decided by central.

kumarfirmware, 2021-02-24:
Hi Ravi,

Thanks for the detailed explanation. 

I am not sure the above is the issue or not. But this might be very close.

What should I enable to see the logs as the above when the issue occurs.

I will not use the work around, but try to reproduce with the logs. But tell me what should I enable for logs?

Also I was trying to move to 202012 release to see if the issue exists there, I get this compilation error. Can you please check this?

fatal error: aws_iot_shadow.h: No such file or directory
compilation terminated.

The above compilation error is seen only in  202012 but not in 202007.

Can you please let me know what is the issue here? I am using shadow functions and above header file. But it does not compile.

Thanks,
Kumar.

ravibhagavandas, 2021-02-25:
> What should I enable to see the logs as the above when the issue occurs.

You can turn on ESP debug logs by updating following configs within `sdkconfig` file in build folder:
```
CONFIG_LOG_DEFAULT_LEVEL=4
CONFIG_LOG_DEFAULT_LEVEL_DEBUG=y
```
> Also I was trying to move to 202012 release to see if the issue exists there, I get this compilation error. Can you please check this?
 Could you tell me which demo are you compiling and which build system are you using ? The header file belongs to previous shadow library. From 202012 onwards we recommend to use new shadow demo which uses our updated shadow library:
 https://github.com/aws/amazon-freertos/tree/202012.00/demos/device_shadow_for_aws

kumarfirmware, 2021-03-03:
Hi Ravi,

In 202012  new shadow demo looks changed drastically. I guess it is more generalized now.

1. I don't think there is any need for AwsIotShadow_Init() function. Please confirm.
2. Also when there is a disconnect in MQTT function and reconnect again,  I do not need to call  AwsIotShadow_Cleanup(); please confirm.
3. why do I need to start always with delete any Shadow document in the cloud? what is the reason to do it? Is there any logic behind this? Is this must and should? Please confirm.
4. How do I replace AwsIotShadow_SetUpdatedCallback  function?
5. How do I replace AwsIotShadow_TimedUpdate function?

Thanks,
Kumar.

kumarfirmware, 2021-03-05:
Hi Ravi,

Can you please help me with the above questions as soon as possible?

Thanks,
Kumar.

ravibhagavandas, 2021-03-08:
Hi kumar,

The release 202012  onwards uses our new coreMQTT library which is more modular, tested and of low memory footprint. To read more about the library take a look at  the [README](https://github.com/FreeRTOS/coreMQTT) for the repo. Shadow library is built on the top of this library and contains necessary functions to filter and decode  shadow messages over MQTT.

Please find my answers below:

> I don't think there is any need for AwsIotShadow_Init() function. Please confirm.

Yes with new library, what you need is the MQTT init and shadows are wrappers to just  filter shadow topics and parse shadow messages. 

> Also when there is a disconnect in MQTT function and reconnect again, I do not need to call AwsIotShadow_Cleanup(); please confirm.

Thats true, you only need to maintain the states for MQTT connection.

> why do I need to start always with delete any Shadow document in the cloud? what is the reason to do it? Is there any logic behind this? Is this must and should? Please confirm.
This is done purely for demo purpose as the shadow document here is maintained entirely by demo and self contained. In real applications you don't need to delete the shadow document at start.

> How do I replace AwsIotShadow_SetUpdatedCallback function?
With the new library you need to set the MQTT event callback and parse the shadow messages from the MQTT messages received through callback. For how to do this take a look at `prvUpdateAcceptedHandler` function in the demo.

> How do I replace AwsIotShadow_TimedUpdate function?

Timed update did a blocking MQTT publish operation waiting for the acknowledgment of publish. `prvPublishToTopic` function shows how to do a blocking publish by waiting for the puback. The message whether shadow document was accepted will be received in `prvUpdateAcceptedHandler` function. 

> Can you please confirm why are you publishing twice to SHADOW_TOPIC_STRING_UPDATE in Demo code? what is the reason or logic behind this?

Usually shadow update involves a different device to update the desired state of the shadow. But since the demo is written as as self contained, it does the job of updating desired state as well as reported state once the device receives the delta. Hence  
the  two publishes.

> what is this xMqttContext? is it same as MQTT connection?

It keeps track of the internal buffer and book keeping used by the MQTT library for an MQTT connection.

kumarfirmware, 2021-03-10:
Hi Ravi,

Thanks for the message. I have the following questions.

1. As you told, I updated the supervision time out to 600, I could still see once OTA job is not started on the device for ever. So the above fix is not solving the problem. So how to proceed in this case?

2.The release 202012 uses ESP version 3.3, if I want to move to ESP 4.4 how should I proceed? can you please explain the steps in detail? Is release 202012 compatible with ESP 4.4 version?

3. Currently in the release 202012 ESP is having BLE 4.2 version, if I want to use the ESP chip with BLE 5.0, is the release 202012 compatible? what changes should I make to use 202012 with BLE5.0 ESP module? can you please explain the steps in detail.

4. What is the safe time gap between MQTT disconnect and reconnect, because when I do a MQTT disconnect and leave a gap of 10 seconds and re connect again I still see AWS disconnecting the connection with the error DUPLICATE_CLIENT_ID in the cloud watch? can you please explain the right process of MQTT disconnect and reconnect and keep alive time out so that we can safely avoid error DUPLICATE_CLIENT_ID.

5. In the release 202012 is it must to use new coreMQTT library or we can still use the old MQTT library? what are the key differences between both?

Please let me know your thoughts on the same as soon as possible.

Thanks
Kumar.

ravibhagavandas, 2021-03-12:
Hi Kumar,
> 1. As you told, I updated the supervision time out to 600, I could still see once OTA job is not started on the device for ever. So the above fix is not solving the problem. So how to proceed in this case?

Did you verify that connection parameter (supervision timeout) was negotiated successfully to 600. You can check the return value of the API  and also log the final negotiated value using the  [BTConnParameterUpdateCallback_t](https://github.com/aws/amazon-freertos/blob/master/libraries/abstractions/ble_hal/include/bt_hal_manager_adapter_ble.h#L269)

Also could you provide logs for the one case where the issue occured, or steps to reproduce it ?

> 2.The release 202012 uses ESP version 3.3, if I want to move to ESP 4.4 how should I proceed? can you please explain the steps in detail? Is release 202012 compatible with ESP 4.4 version?
ESP 4.2 uses an upgraded version of toolchain and is **not directly compatible with 202012** release. You can check the latest master branch for ESP 4.2 . Also getting started guide [here]
(https://docs.aws.amazon.com/freertos/latest/userguide/getting_started_espressif.html)  explains how to install toolchain and compile latest ESP4.2

> 4.What is the safe time gap between MQTT disconnect and reconnect, because when I do a MQTT disconnect and leave a gap of 10 seconds and re connect again I still see AWS disconnecting the connection [...]

Could you elaborate more on how disconnects and reconnects are performed, is the underlying TCP connection torn down ? As per the docs and MQTT spec, connecting with a duplicate client id should only disconnect previous connection and the new connection should always be established. https://docs.aws.amazon.com/iot/latest/developerguide/audit-chk-conflicting-client-ids.html

> 5. In the release 202012 is it must to use new coreMQTT library or we can still use the old MQTT library? what are the key differences between both?
There is a shim layer provided for applications using old MQTT library to use the same APIs but with the coreMQTT library under the hood. But for best performance and future versions its recommended to use coreMQTT library directly if possible.

kumarfirmware, 2021-03-31:
Hi Ravi,

Thanks for the message. Can you help me with the following as soon as possible. This is troubling me a lot.

4.What is the safe time gap between MQTT disconnect and reconnect, because when I do a MQTT disconnect and leave a gap of 10 seconds and re connect again I still see AWS disconnecting the connection with the error DUPLICATE_CLIENT_ID in the cloud watch? can you please explain the right process of MQTT disconnect and reconnect and keep alive time out so that we can safely avoid error DUPLICATE_CLIENT_ID.

Could you elaborate more on how disconnects and reconnects are performed, is the underlying TCP connection torn down ? As per the docs and MQTT spec, connecting with a duplicate client id should only disconnect previous connection and the new connection should always be established. https://docs.aws.amazon.com/iot/latest/developerguide/audit-chk-conflicting-client-ids.html

for example when sending the shadow is failed, I do a MQTT disconnect in the following way

IotMqtt_Disconnect( xConnection.xMqttConnection, false );
 vMqttDemoDeleteNetworkConnection( &xConnection );
 xNetworkConnected = pdFALSE;

I wait for 5 to 10 seconds and do a reconnect again in the following way.

IotMqtt_Connect( &( xConnection.xNetworkInfo ),
                                            &xConnectInfo,
                                            otaCONN_TIMEOUT_MS, &( xConnection.xMqttConnection ) ) == IOT_MQTT_SUCCESS )                                                     
                    
But looks like AWS rejects the connection with DUPLICATE_CLIENT_ID several times and the device keeps retrying in a loop.

it takes long time to establish successful connection with out DUPLICATE_CLIENT_ID.

what could be the problem here? 

May be because I kept the keep alive time as 30  seconds and connection retry as 10 seconds, is this causing the problem?

Also how much time AWS cloud takes to understand that device disconnected?

Because this is not good, clearly I can see DUPLICATE_CLIENT_ID this in the AWS cloud watch when this problem happens.

But I am not doing anything wrong. 

This is an urgent issue. can you please help me in this as soon as possible?

Also can you tell me the shadow reported happens automatically from the AWS shadow library or is it the application responsibility to trigger shadow reported when ever there is any change in the data? can you please let me know how this works? Because I read some where shadow report will happen automatically every 5 seconds or 3 seconds if any change in data. I am confused here. I think it is application responsibility to generate shadow if there is nay data change.

Thanks,
Kumar.

ravibhagavandas, 2021-03-31:
Hi,
Could you tell  if the MQTT connection is made through WiFi or BLE, so that I can try reproducing the same on my setup ? if BLE, are you using IOS or Android as the mobile device ?

Additionally I see that the connection is torn down using `vMqttDemoDeleteNetworkConnection()` but not re-created using ` prxCreateNetworkConnection()` before calling `IotMqtt_Connect()`. Is the underlying connection recreated before calling MQTT connect ?

The disconnect should be instantaneous at server, since an MQTT disconnect packet is sent to server by the API`IotMqtt_Disconnect`.

> I think it is application responsibility to generate shadow if there is any data change.
It's applications responsibility to generate the shadow reported state change document. Shadow library provides API to update a shadow document change, which application can choose to use at convenience. It does not auto generate the reported state change document.

kumarfirmware, 2021-04-01:
Hi Ravi,

Thanks for the message.

It is my bad, I have written it wrongly, I am doing the reconnect using prxCreateNetworkConnection(); This is the standard, I am not changing anything here.

This issue can be easily reproduced in Wi-Fi. This issue will start easily when you are trying to report a shadow with 2000 bytes and also when you are running OTA also this issue can be reproduced. 

Set keep alive time as 20 seconds and you do a MQTT connect and disconnect and reconnect in 5 seconds.

"The disconnect should be instantaneous at server, since an MQTT disconnect packet is sent to server by the APIIotMqtt_Disconnect."

I seriously doubt this, if this is the case why AWS cloud is disconnecting the connection with  DUPLICATE_CLIENT_ID.

So some where there is synchronization problem between connect/dsiconnect/reconnect and keep alive timeout and when device is disconnected cloud is not notified. I guess this.

Because I dont think from my side I am doing nay thing wrong, because I am just using the APIs. There is no way I should see DUPLICATE_CLIENT_ID in AWS logs and end less loop of disconnect and reconnect.

Is there any relation between connection retry time and keep alive time? this might be critical please let me know. 

Please test this at your end and try to reproduce this as soon as possible. Please consider this as an urgent issue.

Thanks,
Kumar.

kumarfirmware, 2021-04-07:
Hi Ravi,

Hope you had an excellent refreshing weekend.
Hope you had an excellent Easter celebration.
Any luck with the above issue?

Thanks.
Kumar.

ravibhagavandas, 2021-04-07:
Hi Kumar,
It would be great if you could share a sample code snippet that reproduce the issue and the version of FreeRTOS you are seeing the issue. I can't reproduce this in a simple connect-disconnect-reconnect flow. Here is the code snippet:
```
for( ; ; )
    {
        xStatus = xMqttDemoCreateNetworkConnection( &connectionContext, AWSIOT_NETWORK_TYPE_WIFI );
        if( xStatus == pdTRUE )
        {
            IotLogInfo("Successfully created a network connection. ");

            connectionContext.xDisconnectCallback = prvNetworkDisconnectCallback;

            /* Set the members of the connection info not set by the initializer. */
            connectInfo.awsIotMqttMode = awsIotMqttMode;
            connectInfo.cleanSession = true;
            connectInfo.awsIotMqttMode = true;
            connectInfo.keepAliveSeconds = 20;
            connectInfo.clientIdentifierLength = ( uint16_t ) strlen( clientcredentialIOT_THING_NAME );
            connectInfo.pClientIdentifier = clientcredentialIOT_THING_NAME;

            /* Establish the MQTT connection. */
            if( status == EXIT_SUCCESS )
            {
                IotLogInfo( "MQTT demo client identifier is %.*s (length %hu).",
                            connectInfo.clientIdentifierLength,
                            connectInfo.pClientIdentifier,
                            connectInfo.clientIdentifierLength );

                connectStatus = IotMqtt_Connect( &connectionContext.xNetworkInfo,
                                                &connectInfo,
                                                OTA_DEMO_CONNECTION_TIMEOUT_MS,
                                                &connectionContext.xMqttConnection );

                if( connectStatus != IOT_MQTT_SUCCESS )
                {
                    IotLogError( "MQTT CONNECT returned error %s.",
                                IotMqtt_strerror( connectStatus ) );

                    status = EXIT_FAILURE;
                    break;
                }

                IotMqtt_Disconnect( connectionContext.xMqttConnection, false ); //Disconnect by sending a DISCONNECT Message.
                vMqttDemoDeleteNetworkConnection( &connectionContext );
                vTaskDelay(pdMS_TO_TICKS(5000)); // 5 seconds between reconnect.
                
            }
        }
    }
```
Keep alive timeout is used by server to disconnect the session only when the client does not initiate the disconnect. For client initiated disconnects, the session should be terminated gracefully and keep alive interval should not be a factor.

kumarfirmware, 2021-04-08:
Hi Ravi,

Thanks for the message.

You are not able to reproduce the issue because in the above code, you are not sending any shadow and you have not started any OTA. This is the reason you are not able to see the issue. No need to download the OTA job, just start OTA functions so that they can subscribe for notifications etc.

As I mentioned above this issue occur only when shadow sending is failed or when OTA subscription is failed.

You should allow either shadow update fail or OTA subscription fail then only this issue occurs.

Please find my code snippet attached here. Also please let me know if any other sequence is wrong in the attached code snippet.
[Test.txt](https://github.com/aws/amazon-freertos/files/6276817/Test.txt)

I am using FreeRTOS version 202007.

Please let me know if you need any additional information. Kindly consider this as an urgent issue.

Thanks.
Kumar.

ravibhagavandas, 2021-04-09:
Hi Kumar,

I don't see any obvious difference between the code snippet above and the previous code snippet used to reproduce the issue. In the failure path of shadow `xNetworkConnected` is set to `pdFALSE` and both code disconnects MQTT connection and reconnects.

Few things would make debug the case easier:

Could you elaborate what causes the shadow send failure and OTA subscription failures, how can I reproduce that failures ?

`xNetworkConnected` seems to be a global variable not defined in the scope. Is it modified out of the scope of this loop ? if so can you paste the code snippet for that as well ?

kumarfirmware, 2021-04-09:
Hi Ravi,

Thanks for the message.

"Could you elaborate what causes the shadow send failure and OTA subscription failures, how can I reproduce that failures ?"

From 8 months both of us working on this together and both of us discussing on this together, we never found a way or at least we could not find a way to debug, why shadow reporting fails sometimes? why subscription of OTA fails sometimes? Why some times messages are not received? why sometimes messages are not published to AWS IOT? we could not crack this problem together or we could not find a way to debug also.

Also they are internal functions and we do not know why they fail some times? Even Wireshark does not give any clue for this. Even enabling MQTT debug also does not give any clue for both of us. you already know this is old issue running from past 8 months.

Now the only clue is in AWS cloudwatch is we see DUPLICATE_CLIENT_ID  and at the device side continuous disconnect and reconnect.

xNetworkConnected is a gloable variable and it is same as example code, nothing is changed from that.

Please let me know if you need any additional information.

Thanks,
Kumar.

ravibhagavandas, 2021-04-09:
@kumarfirmware 

Sorry to hear that this takes longer to resolve. There seems to be two different issues mentioned in this thread and the issue with DUPLICATE_CLIENT_ID seems to be different from the original issue posted. The original issue was periodic and occurs over BLE, whereas the issue with DUPLICATE_CLIENT_ID is random and occurs on any network.

> xNetworkConnected is a gloable variable and it is same as example code, nothing is changed from that.

[OTA sample code](https://github.com/aws/amazon-freertos/blob/202007.00/demos/ota/aws_iot_ota_update_demo.c)  in version 202007 does not contain the  `xNetworkConnected` flag, so could you point me to the code you are referring to.

Regarding DUPLICATE_CLIENT_ID error:

As mentioned  before and also from this [doc](https://docs.aws.amazon.com/iot/latest/developerguide/life-cycle-events.html)  the error occurs when there are more than one connections made with same client ID. When this occurs, the later connection succeeds and old connection drops. So in case of a single device creating a single connection using the client ID, disconnects and then reconnects - either the connection is torn down gracefully and recreated or in the event that the connection is not closed at server side, still the new connection succeeds and old connection drops. So this scenario shouldn't cause a DUPLICATE_CLIENT_ID error.

Do you have multiple devices in your setup which connects using the same client ID to same AWS account ?
```
xConnectInfo.clientIdentifierLength = ( uint16_t ) strlen( eos_deviceID );
xConnectInfo.pClientIdentifier = eos_deviceID;
```
If so if a different device connects  using same client ID, the connection on this device (old connection) can get dropped and this can lead to a loop of disconnect reconnects as both devices tries to retry creating a new connection.

kumarfirmware, 2021-04-09:
Hi Ravi,

Thanks for the message.

I do not have multiple devices in my setup with same client ID.

Also from one device, I am not making multiple MQTT connections.

Please find the updated code attached here with the details of xNetworkConnected .

Please let me know if you need any additional information.

Thanks,
Kumar.
[Test1.txt](https://github.com/aws/amazon-freertos/files/6286758/Test1.txt)

ravibhagavandas, 2021-04-09:
@kumarfirmware 

Thanks for providing more details. Looking through the sample code I can see a potential problem where  MQTT is disconnected and the connection  could be deleted concurrently in two threads of execution, specifically in `prvNetworkDisconnectCallback()` and the main loop.  Depending on how tasks gets executed, they could both delete the same network connection and could cause issues as its not thread safe. Is your intention of using `prvNetworkDisconnectCallback` to disconnect if there is an MQTT error? if so could you just set `xNetworkConnected = false` in the callback so that main loop take care of disconnect and reconnect.

kumarfirmware, 2021-04-12:
Hi Ravi,

Thanks for the message.

"I can see a potential problem where MQTT is disconnected and the connection could be deleted concurrently in two threads of execution, specifically in prvNetworkDisconnectCallback() and the main loop."

I did not understand this, can you please explain it in detail.

If you see the test code I shared with you,  I already commented all the code in the prvNetworkDisconnectCallback() function. Only code active is  xNetworkConnected = pdFALSE;

So I guess this is taken care, please confirm.

Thanks, 
Kumar.

ravibhagavandas, 2021-04-12:
@kumarfirmware 

I tried to reproduce the issue with your code in my setup and only way I could reproduce DUPLICATE_CLIENT_ID error is when a new connection is made without tearing down old connection. Even then the new connection always succeeds and there is no loop of disconnect-reconnects.
In my setup it has only one device and uses thing name (registered in console) as the Client ID.  Once connected and OTA agent starts, the demo always stays in AWS_IDLE_STATE and never disconnects. So I am not sure what needs to be changed to replicate your scenario.

Additionally I was able to debug using connected disconnected events by subscribing to topic `$aws/events/presence/+/<client id>`  from AWS IoT MQTT Test Client as mentioned in:
https://docs.aws.amazon.com/iot/latest/developerguide/life-cycle-events.html

```
$aws/events/presence/connected/<client id>
April 12, 2021, 12:43:22 (UTC-0700)
{
  "clientId": "TEST_DEVICE",
  "timestamp": 1618256602624,
  "eventType": "connected",
  "sessionIdentifier": "15f30367-98bb-438d-86f1-05b53e2bab94",
  "principalIdentifier": "XXXX",
  "ipAddress": "67.168.38.188",
  "versionNumber": 7
}
$aws/events/presence/disconnected/<client id>
April 12, 2021, 12:43:22 (UTC-0700)
{
  "clientId": "TEST_DEVICE",
  "timestamp": 1618256602605,
  "eventType": "disconnected",
  "clientInitiatedDisconnect": false,
  "sessionIdentifier": "37a94b6f-7f59-4076-9866-ff84e8bb743d",
  "principalIdentifier": "XXXX",
  "disconnectReason": "DUPLICATE_CLIENTID",
  "versionNumber": 6
}
```

Since both events originated from same device the `principalIdentifier` field should be same for both disconnected and connected events. Could you subscribe to the topics and check for the principal identifier  when the issue occurs.

kumarfirmware, 2021-04-14:
Hi Ravi,

Thanks for the message.

This is excellent. This is the way to debug. I will check this and come back as soon as possible.

But can you please confirm, are you sure of the following. This is very important.

"only way I could reproduce DUPLICATE_CLIENT_ID error is when a new connection is made without tearing down old connection. Even then the new connection always succeeds and there is no loop of disconnect-reconnects."

Say for example Device lost connectivity for few seconds and MQTT connection got disconnected and device could not notify the same to cloud because of lack of connectivity. cloud still thinks device is connected to it. Devices restores connectivity in few seconds and will do a reconnect. All this happens very quickly. At this point what does cloud think? cloud thinks already a connection exists and will disconnect this connection? and this will continue as loop of connect/disconnect/reconnect on device.

Please look in to this. This is very important and this is the exact situation happening now.

Thanks,
Kumar.

ravibhagavandas, 2021-04-15:
Hi Kumar,

> Say for example Device lost connectivity for few seconds and MQTT connection got disconnected and device could not notify the same to cloud because of lack of connectivity. cloud still thinks device [...]

As per the  tests and and as per the doc [here](https://docs.aws.amazon.com/iot/latest/developerguide/audit-chk-conflicting-client-ids.html), it drops old connection and new connection succeeds. So it should not enter the loop of connect-disconnect.

Here is the first paragraph in the doc:
```
Multiple connections were made using the same client ID, causing an already connected device to be disconnected. The MQTT specification allows only one active connection per client ID, so when another device connects using the same client ID, it knocks the previous one off the connection.
```
 You could confirm this at your side by  replicating this behavior ( for example turning off and on your access point) at subscribing to connect, disconnect events while the device reconnects to server.

kumarfirmware, 2021-04-15:
Hi Ravi,

Thanks for the message. I will test this and let you know soon.

**Meanwhile, I am struck at this. Can you please help me in the following as soon as possible?**

**how do I decode the following shadow document in JSON format using the JSON APIs? I am using V202007. This is an array object with 2 elements. How do I know the number of elements present and their values?**

"room": 
{
    "values":
   [
      {
        "temp": "25",
        "humidity": "80",
        "occupancy": "2"
      },
      {
    "temp": "26",
    "humidity": "90",
    "occupancy": "3"
      }
    ]
  }

The APIs and examples are not clear for this, how to decode JSON format array of objects using JSON APIS.

Sample example also is not clear and not complete.

This does not show how to decode and this example is not complete. 

TEST( Serializer_Unit_JSON_deserialize, find_nested_key_array_of_objects_value )
{
    IotSerializerDecoderObject_t nestedObject;

    IotSerializerError_t json_error = _decoder.find( &rootObject, "related", &nestedObject );

    TEST_ASSERT_EQUAL( json_error, IOT_SERIALIZER_SUCCESS );
    TEST_ASSERT_EQUAL( nestedObject.type, IOT_SERIALIZER_CONTAINER_MAP );

    json_error = _decoder.find( &nestedObject, "types", &childObject );
    TEST_ASSERT_EQUAL( json_error, IOT_SERIALIZER_SUCCESS );
    TEST_ASSERT_EQUAL( childObject.type, IOT_SERIALIZER_CONTAINER_ARRAY );

    _decoder.destroy( &nestedObject );
}

Can you please provide me the code which can decode above JSON document as soon as possible?

Thanks in advance.

Thanks,
Kumar.

kumarfirmware, 2021-04-16:
Hi Ravi,

Good Morning. Can you please help me with the above as soon as possible?
Thanks in advance.

Thanks,
Kumar.

ravibhagavandas, 2021-04-16:
Hi,

You can get size of the array or  step in through the array elements as follows:
```
 /* Get Size of the array */
size_t arraySize;
TEST_ASSERT_EQUAL( IOT_SERIALIZER_SUCCESS, _decoder.getSizeOf( &childObject, &arraySize ) );
TEST_ASSERT_EQUAL( 2, arraySize );

/* Test iterating through the array and extracting the first and last elements. */

IotSerializerDecoderIterator_t arrayIterator = IOT_SERIALIZER_DECODER_ITERATOR_INITIALIZER;
IotSerializerDecoderObject_t arrayElement = IOT_SERIALIZER_DECODER_OBJECT_INITIALIZER;

TEST_ASSERT_EQUAL( IOT_SERIALIZER_SUCCESS, _decoder.stepIn( &childObject, &arrayIterator ) );

for(size_t  i = 0; i < arraySize; i++)
{
    TEST_ASSERT_EQUAL( IOT_SERIALIZER_SUCCESS, _decoder.get( arrayIterator, &arrayElement ) );
    TEST_ASSERT_EQUAL( arrayElement.type, IOT_SERIALIZER_CONTAINER_MAP );
}

TEST_ASSERT_EQUAL( IOT_SERIALIZER_SUCCESS, _decoder.stepOut( arrayIterator, &childObject ) );

```

For more details you can refer this example of CBOR which uses same APIs. The examples shows how to step through an array and a nested object within array, the API syntax applies to JSON as well.
https://github.com/aws/amazon-freertos/blob/master/libraries/c_sdk/standard/serializer/test/iot_tests_serializer_cbor.c#L929

kumarfirmware, 2021-04-16:
Hi Ravi,

There is no function  _decoder.getSizeOf() in iot_serializer_json_decoder.c

can you please give me a clear code how to decode json array using the API.

I am fine even if you use corejson of Version 202012 also.

Can you please give me the clear code how do you extract above json array.

I spent 2 days and could not crack this. Kindly help me as soon as possible.

Thanks,
Kumar.

kumarfirmware, 2021-04-19:
Hi Ravi,

Good Morning. Hope you had an excellent weekend.

Since the above solution which you mentioned did not work, I moved to core Json.

I attached my code here, but I am afraid still it is not efficient code.

I just used  jsonStatus = JSON_Search(test_data, strlen(test_data), "state.reported.wifi", strlen("state.reported.wifi"), &value, &valueLength);

I used the above JSON_Search function repeatedly and got the correct result. I am afraid it is not efficient.

Also when I am using JSON_Search function repeatedly, I need to take backup of my original source buffer and I need to copy it back before I use it for the next time otherwise its contents are changed. I do not know why JSON_Search  function modifies source buffer.

[Test_Json.txt](https://github.com/aws/amazon-freertos/files/6338899/Test_Json.txt)
Also I do not know how to use JSON_Iterate(value, valueLength, &start, &next, &pair); I am not able to use this and extract array values.

How do I find out the number of elements in array easily?

Please help me with this as soon as possible.

Please find my code attached here.

Thanks,
Kumar.

ravibhagavandas, 2021-04-19:
Hi,

> I do not know why JSON_Search function modifies source buffer.

`JSON_Search` should not modify the source buffer as it accepts a pointer to a constant source buffer as first argument, which cannot be modified internally.
https://github.com/FreeRTOS/coreJSON/blob/f329562694feeb0fd1b12adaef58d1072943bfe3/source/core_json.c#L1681
Its not clear from the code snippet you posted, how the `pCallbackParam->u.callback.pDocument` is received. Is it possible that `pCallbackParam->u.callback.pDocument` gets modified from other tasks?

> Also I do not know how to use JSON_Iterate(value, valueLength, &start, &next, &pair); I am not able to use this and extract array values.How do I find out the number of elements in array easily?

The unit test [here](https://github.com/FreeRTOS/coreJSON/blob/26fe707548fc4e975c453aa4475ae056bb70247b/test/unit-test/core_json_utest.c#L967) shows an example of how to iterate elements inside a JSON array. You need to first validate the document using `JSON_Validate()` and then iterate through elements using `JSON_Iterate()` until the function returns `JSONNotFound`. 

Hope that helps.

dan4thewin, 2021-04-19:
Here's a working example based on your code:

    #include <stdio.h>
    #include <string.h>
    #include <core_json.h>
    
    const char json[] = "{ \"room\": { \"values\": [ { \"room number\": \"25\", \"temp\": \"25\", \"humidity\": \"80\" }, { \"room number\": \"26\", \"temp\": \"26\", \"humidity\": \"90\" } ] } }";
    
    #define key "room.values"
    #define keyLength (sizeof(key) - 1)
    #define key2 "temp"
    #define key2Length (sizeof(key2) - 1)
    
    void decode_json()
    {
        size_t jsonLength;
        JSONStatus_t jsonStatus;
        const char *array;
        size_t start = 0, next = 0, arrayLength = 0, count = 0;
        JSONPair_t pair = { 0 };
    
        jsonLength = strlen(json);
    
        jsonStatus = JSON_Validate(json, jsonLength);
    
        if (jsonStatus == JSONSuccess)
            jsonStatus = JSON_SearchConst(json, jsonLength, key, keyLength, &array, &arrayLength, NULL);
    
        if (jsonStatus == JSONSuccess)
            jsonStatus = JSON_Iterate(array, arrayLength, &start, &next, &pair);
    
        while (jsonStatus == JSONSuccess)
        {
            const char *value;
            size_t valueLength;
            count++;
    
            jsonStatus = JSON_SearchConst(pair.value, pair.valueLength, key2, key2Length, &value, &valueLength, NULL);
            if (jsonStatus == JSONSuccess)
                printf( "value: %.*s\n", (int) valueLength, value );
    
            jsonStatus = JSON_Iterate(array, arrayLength, &start, &next, &pair);
        }
    
        printf("Final array count=%ld\n", count);
    }
    
    int main()
    {
        decode_json();
        return 0;
    }

kumarfirmware, 2021-04-20:
Hi Dan,

Thanks for your message. The above sample code you provided perfectly works fine. This is  better version than mine as you are using JSON_Iterate function.

can you please explain what %.*s is doing in this? printf( "value: %.*s\n", (int) valueLength, value );

If I understand correctly, there is no direct function or API to get array count. Also, we need to use JSON_SearchConst() to get the value of each element in the array.

This is solved now. 

But the problem now is how to keep a dynamic array in shadow reported.

Because for shadow reported I am using the following way. This is static and this does not work any more.

Because in the following JSON,  room.values sometimes have two elements, sometimes three elements, sometimes 4 elements. I do not know it before as static and it has to be dynamic and in run time.

Half part of the SHADOW_REPORTED_JSON  is static and the room part is dynamic. so how do I handle half part as static and half part as dynamic?

How do I formulate a dynamic array in SHADOW_REPORTED_JSON  ?

Can you please let me know the same as soon as following?

#define SHADOW_REPORTED_JSON    \
    "{"                         \
    "\"state\":{"               \
    "\"reported\":{"            \
    "\"serialnumber\":\"12345678\"" \
    ","                        \
    "\"room\":{"                \
    "\"activeroom\":\"25\""     \
    ","                         \
    "\"values\":["                \
    "{"                         \
    "\"roomnumber\":\"25\""           \
    ","                         \
    "\"temp\":\"23\""            \
    ","                         \
    "\"humidity\":\"50\""            \
    "},"                        \
    "{"                         \
    "\"roomnumber\":\26\""           \
    ","                         \
    "\"temp\":\"27\""            \
    ","                         \
    "\"humidity\":\"28\""            \
    "}"                        \
    "]"                        \
    "},"                         \
    "\"clientToken\":\"%06lu\"" \
    "}"                         \
    "}"                         \
    "}"                     

The following code does not work anymore if I have to use dynamic shadow. I am confused here how to proceed?

Because sometimes it will be 2 rooms,  sometimes it will be three rooms and so on.

status = snprintf(pUpdateDocument,
                        2000,//EXPECTED_REPORTED_JSON_SIZE + 1,
                        SHADOW_REPORTED_JSON,
                         serialnumber, activeroom, values[0].roomnumber, values[0].temp, values[0].humidity,
                        values[1].roomnumber, values[1].temp, values[1].humidity,
                        (long unsigned)(IotClock_GetTimeMs() % 1000000));

updateStatus = AwsIotShadow_TimedUpdate( mqttConnection,
                                             &updateDocument,
                                             AWS_IOT_SHADOW_FLAG_KEEP_SUBSCRIPTIONS,
                                             5000 );

Thanks,
Kumar.

dan4thewin, 2021-04-21:
Hi, @kumarfirmware .
The %.*s tells printf to use two arguments together.  The first is a length, and the second is a string.  Using this length argument, the string doesn't need to have a sentinel NUL character.

There is no coreJSON API for directly counting array elements.  The coreJSON design doesn't maintain state between calls, so counting requires the same parsing work as operating on each element.

Do you have to use JSON_Search() to extract elements from objects?  It's one option.  JSON_Iterate() works with objects as well as arrays.

Today, coreJSON doesn't have functions to create a JSON document.  While I have a plan to add such, it's pretty simple to create them with snprintf (or more basic functions when MISRA compliance is desired).  Consider:

    #include <stdio.h>
    #include <core_json.h>

    #define HEAD_FORMAT \
    "{ \"state\":{ \"reported\":{ \"serialnumber\":\"%u\", \"room\":{ \"activeroom\":\"%u\", \"values\":["
    #define ROOM_FORMAT \
    "%s{ \"roomnumber\":%d, \"temp\":%d, \"humidity\":%d }"
    #define TAIL_FORMAT \
    "]}}}}"

    char buffer[BUFSIZ];

    struct room_t {
        int number, temp, humidity;
    };

    #define MAX_ROOMS 10

    struct room_t room[MAX_ROOMS];

    int make_report(unsigned serial, unsigned active)
    {
        int i, length = 0, count = 0;

        length += snprintf(buffer, BUFSIZ, HEAD_FORMAT, serial, active);

        for (i = 0; (i < MAX_ROOMS) && (length < BUFSIZ); i++)
        {
            if (room[i].number != 0)
            {
                length += snprintf(buffer + length, BUFSIZ - length, ROOM_FORMAT,
                                   count > 0 ? "," : "",
                                   room[i].number, room[i].temp, room[i].humidity);
                count++;
            }
        }

        if (length < BUFSIZ)
            length += snprintf(buffer + length, BUFSIZ - length, TAIL_FORMAT);

        if ((length < BUFSIZ) && (JSON_Validate(buffer, length) == JSONSuccess))
            return length;
        else
            return -1;
    }

    int main()
    {
        int i, length;

        for (i = 0; i < MAX_ROOMS; i++) {
            room[i].number = i + 1;
            room[i].temp = i * 10;
            room[i].humidity = i * 5;
        }

        length = make_report(123, 4);
        printf("%s\n", length > 0 ? buffer : "ERROR");

        for (i = 0; i < MAX_ROOMS; i++) {
            room[i].number = i + 1;
            room[i].temp = i * 7;
            room[i].humidity = i * 2;
        }

        length = make_report(124, 4);
        printf("%s\n", length > 0 ? buffer : "ERROR");

        /* remove 3 rooms from report */
        room[0].number = 0;
        room[1].number = 0;
        room[2].number = 0;

        length = make_report(125, 4);
        printf("%s\n", length > 0 ? buffer : "ERROR");

        return 0;
    }

kumarfirmware, 2021-04-22:
Hi Dan,

Thanks for your message. This is very detailed and very clear.

Thanks a lot.

Thanks,
Kumar.

dan4thewin, 2021-04-23:
I'm happy to help.  I'll close this issue now.  If you have any more trouble, please feel free to open a new issue and reference this one if needed.

closed 2021-04-23 by dan4thewin