aws/amazon-freertos #3010: Attempting unit testing in private project

MrMelon1232, 2021-02-23:
Hi,

I would like to set up unit testing in my project. I had an issue with including unity.h as it was not detected and found that unit test folders for AWS components always have a unity folder in them. Would it be possible to make unity.h as a component that can be included in projects or is the only way to also include a unity folder in my project ?

Thank You

lundinc2, 2021-02-25:
Hi @MrMelon1232,

Would adding unity as a [dependency](https://github.com/aws/amazon-freertos/blob/079e6a20e26b9be9ac000a167de858cfdd7401d0/tests/CMakeLists.txt#L42) resolve your issue?

Alternatively, you could mirror this in your project, and pull Unity manually by mimicking how it is added to [Amazon FreeRTOS](https://github.com/aws/amazon-freertos/blob/079e6a20e26b9be9ac000a167de858cfdd7401d0/libraries/3rdparty/CMakeLists.txt#L250). 

Please let me know the details of your project so I can help. For example what version you are working against, platform and how your cmake is configured. From what I recall you have a CMake that is pulling Amazon FreeRTOS as a component, is that correct?

Thanks,

Carl

MrMelon1232, 2021-03-01:
HI @lundinc2,

I tried adding unity as a dependency but got no luck with that. I am currently working on commit: dc53c9cbe68977e0892dcbfb023256d4055d1dd7 of freertos. 
Here is my project hierarchy:
-freertos
-fsonet_lib
 --> include
 --> src
 --> test
      --> include
      --> src
      --> CMakeLists.txt
 --> CMakeLists.txt
-test
 --> src
      --> main.c
 --> CMakeLists.txt
-src
 --> main.c
-CMakeLists.txt

Here are my CMakeLists.txt
file under fsonet_lib/test/CMakeLists.txt
```
set(COMPONENT_ADD_INCLUDEDIRS include)
set(COMPONENT_SRCDIRS src)

register_component()
```
file under test/CMakeLists.txt
```
cmake_minimum_required(VERSION 3.13)

project(unit_tests)

add_executable(unit_test src/main.c)

# Tell IDF build to link against this target.
set(IDF_PROJECT_EXECUTABLE unit_test)
set(CMAKE_EXECUTABLE_SUFFIX ".elf")

# Add some extra components. IDF_EXTRA_COMPONENT_DIRS is an variable used by ESP-IDF
# to collect extra components.
get_filename_component(
    EXTRA_COMPONENT_DIRS
    "../fsonet_lib" ABSOLUTE
)
list(APPEND IDF_EXTRA_COMPONENT_DIRS ${EXTRA_COMPONENT_DIRS})

#Configs folders
include_directories(BEFORE "${CMAKE_CURRENT_LIST_DIR}/../fsonet-configs")

#Set sdkconfig
set(IDF_SDKCONFIG_DEFAULTS "${CMAKE_CURRENT_LIST_DIR}/../fsonet-configs/sdkconfig.defaults")

# Add freertos as an subdirectory. AFR_BOARD tells which board to target.
set(AFR_BOARD espressif.esp32_devkitc CACHE INTERNAL "")
add_subdirectory("${CMAKE_CURRENT_LIST_DIR}/../freertos" [binary_dir])

target_link_libraries(unit_test PRIVATE AFR::demo_core_mqtt)
target_link_libraries(unit_test PRIVATE AFR::demo_device_shadow)
target_link_libraries(unit_test PRIVATE AFR::demo_ota)
target_link_libraries(unit_test PRIVATE AFR::shadow)
#target_link_libraries(unit_test PRIVATE AFR::common_io)
include_directories(unit_test PRIVATE fsonet-configs)
#include_directories(unit_test PRIVATE "amazon-freertos/demos/network_manager")
include_directories(unit_test PRIVATE "amazon-freertos/demos/dev_mode_key_provisioning/include")

#Linking esp-idf libraries
set(esp_idf_dir "${CMAKE_CURRENT_LIST_DIR}/../freertos/vendors/espressif/esp-idf")
set(IDF_TARGET esp32)
set(ENV{IDF_PATH} ${esp_idf_dir})
include(${esp_idf_dir}/tools/cmake/idf_functions.cmake)
set(IDF_TOOLCHAIN_FILE ${CMAKE_TOOLCHAIN_FILE})
```

lundinc2, 2021-03-01:
Hi @MrMelon1232,

Thank you for these details. I will replicate your structure, and post an example here once I have completed the integration.

Thanks,

Carl

yanjos-dev, 2021-03-11:
Hi @MrMelon1232, I'm assuming that this is for the esp32 platform. This repository uses CMake to build and link to the Unity library for testing. This is done by creating a CMake target that defines what sources and include paths are part of the Unity library. As Carl shared earlier, you can see where the CMake target is defined in this repository [here](https://github.com/aws/amazon-freertos/blob/079e6a20e26b9be9ac000a167de858cfdd7401d0/libraries/3rdparty/CMakeLists.txt#L250-L268).

There is a CMake file in the tests directory that is used to build and run the unit tests. You can see [here](https://github.com/aws/amazon-freertos/blob/858a26e9043c2baea1b95ffc38d33e5bb8ea1314/tests/CMakeLists.txt#L42) where it is adding this dependency on the Unity target that we defined. 

It looks like you attempted to take this approach without success. Can you share the problems that you ran into?

Otherwise, I did a quick search to see what I could find out about setting up unit tests using the ESP-IDF framework. I found [this resource](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-guides/unit-tests.html) that you may find helpful.

Beyond this, I would recommend asking this question in the [Espressif section of the FreeRTOS forums](https://forums.freertos.org/c/partners-sponsors/espressif/14). This forum is a good resource for general questions related to using this code on an Espressif platform.

ravibhagavandas, 2021-03-22:
Hi @MrMelon1232 
Did the suggestions help resolve your issue ? Please let us know if you have any questions.

archigup, 2021-04-05:
Closing this issue. Please feel free to open a new issue if you have further questions on this topic.

closed 2021-04-05 by archigup