aws/amazon-freertos #3014: AFR & ESP32: Failed to create new socket. tcpSocket=0xffffffff

dipen-1337lab, 2021-02-26:
Namaste Developers,

The Amazon FreeRTOS specifics for our project are 

- FreeRTOS Version is V202002.00.
- coreHTTP library has been ported for use with V202002.00. More details [here](https://github.com/aws/amazon-freertos/issues/2911).
- cmake is run with `-DAFR_ESP_LWIP=1` followed by `ninja` command.

### Issue Description

For our application, RFID authentication is carried out leveraging the `RunCoreHttpMutualAuthDemo` within the `http_demo_mutual_auth.c` source file.

As part of testing, when swiping the RFID card multiple times, the below is encountered. The error is easily reproducible within 10 attempts i.e., 10 swipes of RFID card.
`[iot_thread] [ERROR] [Transport_Secure_Sockets] [transport_secure_sockets.c:420] 2450 62106 [iot_thread] Failed to create new socket. tcpSocket=0xffffffff`

A sample log is enclosed for reference, wherein the first swipe (in the log file) runs the `RunCoreHttpMutualAuthDemo` demo successfully.
`[iot_thread] [INFO] [HTTPDemo] [http_demo_mutual_auth.c:369] 2279 56697 [iot_thread] Demo completed successfully.`

However, the following swipe results in the above error, as can be seen in the log file.
[Failed to create new socket. tcpSocket=0xffffffff.txt](https://github.com/aws/amazon-freertos/files/6051331/Failed.to.create.new.socket.tcpSocket.0xffffffff.txt)

**The error message is printed from [here](https://github.com/aws/amazon-freertos/blob/master/libraries/abstractions/transport/secure_sockets/transport_secure_sockets.c#L442) and, seems to be because of insufficient memory.**

### Query

1. For further debugging, I'm unable to understand which of the two `SOCKETS_Socket` function gets called, which is generating the error. Is it [this one](https://github.com/aws/amazon-freertos/blob/master/libraries/abstractions/transport/secure_sockets/transport_secure_sockets.c#L442) in `libraries/abstractions/secure_sockets/freertos_plus_tcp` or [this one](https://github.com/aws/amazon-freertos/blob/master/libraries/abstractions/secure_sockets/lwip/iot_secure_sockets.c#L350) in `libraries/abstractions/secure_sockets/lwip`.

2. How to enable the debug logs to capture some additional information on the serial console so as to understand the root cause of the insufficient memory error and, resolve the same.

Thanks | Regards,
Dipen

lundinc2, 2021-03-01:
Hello @dipen-1337lab,

Thank you for this report. We are investigating and will provide guidance shortly.

Thank you,

Carl

lundinc2, 2021-03-02:
Hi @dipen-1337lab,

From the logs it seems that sockets are not being closed, and the array of available sockets is running out. As you have pointed out, it is the second link that contains the secure_sockets implementation your application is calling into. 

The secure sockets implementation defines this https://github.com/aws/amazon-freertos/blob/066e66a1bee61daebace8482fdcc1bd1b19196f8/libraries/abstractions/secure_sockets/lwip/iot_secure_sockets.c#L125 integer, that contains the max amount of sockets you can open at once. 

Everytime you open a socket, the available sockets is decremented here. https://github.com/aws/amazon-freertos/blob/066e66a1bee61daebace8482fdcc1bd1b19196f8/libraries/abstractions/secure_sockets/lwip/iot_secure_sockets.c#L380 Once it is closed, the socket is returned here. https://github.com/aws/amazon-freertos/blob/066e66a1bee61daebace8482fdcc1bd1b19196f8/libraries/abstractions/secure_sockets/lwip/iot_secure_sockets.c#L149

The likely issue you are hitting is here https://github.com/aws/amazon-freertos/blob/066e66a1bee61daebace8482fdcc1bd1b19196f8/libraries/abstractions/secure_sockets/lwip/iot_secure_sockets.c#L363 where there are no more sockets to grab.

For the ESP32 the default value of `socketsconfigDEFAULT_MAX_NUM_SECURE_SOCKETS` is 10, which makes given that you can do 10 swipes, and then you hit the error.

Can you verify that you are closing your sockets after opening a connect? You can set a breakpoint for `prvSocketsClose` to see if you are closing your sockets. 

Here is what my call stack looked like when I hit the break point. The API you likely want to use is `SecureSocketsTransport_Disconnect`, though I did not look too closely at the porting work you did in the previous issue. 

```
(gdb) where
#0  prvSocketsClose (ctx=0x3ffdd30c) at ../libraries/abstractions/secure_sockets/lwip/iot_secure_sockets.c:146
#1  0x4015b7f9 in prvDecrementRefCount (ctx=0x3ffdd30c) at ../libraries/abstractions/secure_sockets/lwip/iot_secure_sockets.c:193
#2  0x4015bbee in SOCKETS_Close (xSocket=0x3ffdd30c) at ../libraries/abstractions/secure_sockets/lwip/iot_secure_sockets.c:668
#3  0x400dccea in SecureSocketsTransport_Disconnect (pNetworkContext=<optimized out>) at ../libraries/abstractions/transport/secure_sockets/transport_secure_sockets.c:571
#4  0x400d5501 in RunCoreMqttMutualAuthDemo (awsIotMqttMode=<optimized out>, pIdentifier=<optimized out>, pNetworkServerInfo=<optimized out>, pNetworkCredentialInfo=<optimized out>, pNetworkInterface=<optimized out>) at ../demos/coreMQTT/mqtt_demo_mutual_auth.c:611
#5  0x400d4b72 in runDemoTask (pArgument=<optimized out>) at ../demos/demo_runner/iot_demo_freertos.c:314
#6  0x4015d376 in _threadRoutineWrapper (pArgument=0x3ffb274c) at ../libraries/abstractions/platform/freertos/iot_threads_freertos.c:85
#7  0x4008fc24 in vPortTaskWrapper (pxCode=0x4015d36c <_threadRoutineWrapper>, pvParameters=0x3ffb274c) at ../freertos_kernel/portable/ThirdParty/GCC/Xtensa_ESP32/port.c:154
```

Please let me know how I can further help you.

Thanks,

Carl

dipen-1337lab, 2021-03-02:
Namaste @lundinc2,

> it is the second link that contains the secure_sockets implementation your application is calling into.

Noted. Thank you for the inputs and, for sharing your call stack.

> Here is what my call stack looked like when I hit the break point.

A quick navigation to the source code for understanding my "Socket Closing Call Stack" yields the below. As can be understood, there is a difference in the two call stacks.
`RunCoreMqttMutualAuthDemo` --> `SecureSocketsTransport_Disconnect` --> `SOCKETS_Shutdown` --> `lwip_shutdown` --> `SOCKETS_Close` --> `lwip_close`.

> Can you verify that you are closing your sockets after opening a connect? You can set a breakpoint for `prvSocketsClose` to see if you are closing your sockets.

Interestingly, the `prvSocketsClose` is not found in my `iot_secure_sockets.c` source file. Should I update my `../libraries/abstractions/secure_sockets/lwip` folder to get the latest code and, verify if the issue is reproduced?

On a side note, within the log file I have shared in the first post, there aren't any logs corresponding to failure due to socket closure.

Thanks | Regards,
Dipen

mahavirj, 2021-03-02:
@dipen-1337lab 

> iot_secure_sockets.c.txt is the source file I've integrated in the AFR 202002.00. This file and other dependencies, were downloaded from the master branch near about a year ago, to run the cmake command with -DAFR_ESP_LWIP=1 option. The file header reads FreeRTOS Secure Sockets V1.1.9 while that on the master branch reads FreeRTOS Secure Sockets V1.3.0.

I am not sure about reason why you integrated custom secure socket port layer but ideally for `202002.00` version you should be using the one from `ports` directory at https://github.com/aws/amazon-freertos/blob/987d9697ad4dc924bd71333c57b2f68bbe81b172/vendors/espressif/boards/esp32/ports/secure_sockets/lwip/iot_secure_sockets.c

This was enabled through build system:
https://github.com/aws/amazon-freertos/blob/987d9697ad4dc924bd71333c57b2f68bbe81b172/vendors/espressif/boards/esp32/components/secure_sockets/component.mk#L6

Can you please try with this port layer once?

Mahavir

jsstrandberg, 2021-03-02:
Hello,

Posting here because it seems like this might be the same issue. If you think I should file a different one or want me to provide additional context, I'll happily do so.

I experienced a similar issue where after a number of network disconnects, I started receiving the `Failed to create new socket. tcpSocket=0xffffffff` error. In my case, it looks like [`SOCKETS_Close` doesn't get called if `SOCKETS_Shutdown` fails](https://github.com/aws/amazon-freertos/blob/master/libraries/abstractions/transport/secure_sockets/transport_secure_sockets.c#L550-L590).

I discovered that `SOCKETS_Shutdown` fails when the socket is disconnected (explanation below the horizontal rule). This matches the [shutdown behavior documented here](https://pubs.opengroup.org/onlinepubs/007908799/xns/shutdown.html), and is therefore presumably correct. [Close actually cleans up the resource](https://pubs.opengroup.org/onlinepubs/007908799/xns/close.html), and should always be called even if shutdown isn't. [Here's a Stack Overflow post from 2010](https://stackoverflow.com/questions/4160347/close-vs-shutdown-socket/4160356#4160356) backing that claim up, since I'm posting this from a newly created account.

The current implementation of `SecureSocketsTransport_Disconnect` won't call `SOCKETS_Close` if it's called after a network change which causes the socket to disconnect, leaking the socket resource. This seems like an oversight to me, but as someone only moderately familiar with the sockets interface, I could be missing something. Is this correct?

---

This is the implementation I have for `SOCKETS_Shutdown`:
```c
int32_t SOCKETS_Shutdown( Socket_t xSocket,
                          uint32_t ulHow )
{
    ss_ctx_t * ctx;
    int ret;

    if( SOCKETS_INVALID_SOCKET == xSocket )
    {
        return SOCKETS_EINVAL;
    }

    ctx = ( ss_ctx_t * ) xSocket;

    if( 0 > ctx->ip_socket )
    {
        return SOCKETS_SOCKET_ERROR;
    }

    ret = lwip_shutdown( ctx->ip_socket, ( int ) ulHow );

    if( 0 > ret )
    {
        return SOCKETS_SOCKET_ERROR;
    }

    return SOCKETS_ERROR_NONE;
}
```

The call fails due to `lwip_shutdown` returning -1 after `netconn_shutdown` returns -11: Not Connected.

lundinc2, 2021-03-02:
Hi @jsstrandberg I believe you are correct. I will discuss with my colleagues and open a PR to fix this bug.

jsstrandberg, 2021-03-03:
@lundinc2 good to know. Thank you.

dipen-1337lab, 2021-03-03:
Namaste @jsstrandberg and @lundinc2, 

> The current implementation of `SecureSocketsTransport_Disconnect` won't call `SOCKETS_Close` if it's called after a network change which causes the socket to disconnect, leaking the socket resource.

That's a good catch! Thank you for bringing this to AFR's notice.

Thank you @lundinc2 for the fix. I'll need to test my application behaviour with the changes, considering the fact that there isn't any network change encountered.

Thanks | Regards,
Dipen

dipen-1337lab, 2021-03-03:
Namaste @mahavirj,

Appreciate your reply to provide assistance on this issue.

> I am not sure about reason why you integrated custom secure socket port layer but ideally for `202002.00` version you should be using the one from `ports` directory at [...]

You are correct! Neither of the two files highlighted in my original post ( File path - `FreeRTOS/libraries/abstractions/secure_sockets/..` ) seem to get included in the project directory structure.
I've verified this by introducing a syntax error within both the files, which doesn't generate the expected compilation/ build error.

The `iot_secure_sockets.c` file in use, is indeed the one residing at the below path, which is within the `ports` directory.
`FreeRTOS\vendors\espressif\boards\esp32\ports\secure_sockets\lwip`.

> This was enabled through build system:
> https://github.com/aws/amazon-freertos/blob/987d9697ad4dc924bd71333c57b2f68bbe81b172/vendors/espressif/boards/esp32/components/secure_sockets/component.mk#L6
> 
> Can you please try with this port layer once?

The `component.mk` file includes the above Line Number 6 pointed out; meaning the issue is encountered with this change already present within my FreeRTOS folder structure.

Thanks | Regards,
Dipen

dipen-1337lab, 2021-03-03:
Namaste @lundinc2, 

> Once it is closed, the socket is returned here.
> https://github.com/aws/amazon-freertos/blob/066e66a1bee61daebace8482fdcc1bd1b19196f8/libraries/abstractions/secure_sockets/lwip/iot_secure_sockets.c#L149

I did a test to print the value of the `sockets_allocated` variable, every time the `SOCKETS_Close()` function gets executed.

Multiple `configPRINTF()` statements added in the `SOCKETS_Close()` function can be viewed in the below attachment.
[SOCKETS_Close_with_configPRINTF_debug_prints.c.txt](https://github.com/aws/amazon-freertos/files/6074285/SOCKETS_Close_with_configPRINTF_debug_prints.c.txt)

And below is the complete console log for the error encountered.
[sockets_allocated_count_verification_log.txt](https://github.com/aws/amazon-freertos/files/6074291/sockets_allocated_count_verification_log.txt)

The before and after values for the `sockets_allocated++;` variable are **always viewed as 8 and 9**, as can be confirmed in the above console log file.
```
964 23541 [iot_thread] INFO : Before sockets_allocated = 8
965 23541 [iot_thread] INFO : After sockets_allocated = 9
```

After multiple successful Socket Open and Socket Close operations, a new attempt results in the socket creation error, as below.
```
1002 24403 [iot_thread] [ERROR] [Transport_Secure_Sockets] [transport_secure_sockets.c:420] 1003 24403 [iot_thread] Failed to create new socket. tcpSocket=0xffffffff
```

Let me know if any additional testing is to be carried out.

Thanks | Regards,
Dipen

dipen-1337lab, 2021-03-03:
Namaste @all,

> I'll need to test my application behaviour with the changes, considering the fact that there isn't any network change encountered.

Sharing an update that the original error was reproduced even with the suggested change of always calling the `SOCKETS_Close()` after the `SOCKETS_Shutdown()`.

Thanks | Regards,
Dipen

dipen-1337lab, 2021-03-03:
Namaste Again,

One question.

I've set [xSocketsConfig.enableTls = false;](https://github.com/aws/amazon-freertos/blob/master/demos/coreHTTP/http_demo_mutual_auth.c#L370) in my application because our server we are connecting to, currently does not support the same. Could this be reason for the error encountered?

Thanks | Regards,
Dipen

lundinc2, 2021-03-03:
HI @dipen-1337lab,

Thanks a lot for your review! Apologies for that mistake, I made the change in a rush :) I have not yet had time to test and discuss the change with my teammates, but whatever the fix is, should land in said PR.

Will update you with more info here, noting your new issues as well, and will take a look

Thanks,

Carl

dipen-1337lab, 2021-03-04:
Namaste All,

Here's an interesting observation I encountered.

So, I added the below simple log statement [here](https://github.com/aws/amazon-freertos/blob/master/libraries/abstractions/transport/secure_sockets/transport_secure_sockets.c#L504).
```
    LogInfo( ( "********** INFO : Entered SecureSocketsTransport_Connect.\n" ) );
```

When running my application, I could not see this message on the Serial console.

**I then once again, introduced a syntax error within this `FreeRTOS/libraries/abstractions/transport/transport_secure_sockets.c` source file and to my surprise, the `project.bin` file was successfully generated.**

I could not find any other **function definition** for the `SecureSocketsTransport_Connect` function within my FreeRTOS project directory structure. Neither did I find anything on the master branch.

Can someone please help me understand this behaviour? Which and where is the file, from where the `SecureSocketsTransport_Connect` and `SecureSocketsTransport_Disconnect` functions are getting called?

Below are the commands I am sequentially executing for a clean and fresh generation of the binary image. The commands are executed from the FreeRTOS folder.
- rmdir /q /s build
- cmake -DVENDOR=espressif -DBOARD=esp32_devkitc -DCOMPILER=xtensa-esp32 -DAFR_ESP_LWIP=1 -GNinja -S . -B build
- cd build
- ninja
- cd ..
- .\vendors\espressif\esp-idf\tools\idf.py erase_flash flash monitor -B build

Let me know if any additional information is required.

Thanks | Regards,
Dipen

dan4thewin, 2021-03-04:
Hi, @dipen-1337lab .  Is the syntax error within a LogInfo() statement or something similar?  I'm wondering if that is a macro defaulting to an empty expansion.  What happens if you use configPRINTF() instead?

dipen-1337lab, 2021-03-05:
Namaste @dan4thewin,

Thank you for replying to provide assistance.

The forcefully introduced syntax error was by removing the semi-colon at the end of the LogInfo statement, as below. The binary image (.bin file) gets successfully generated when a compilation/ build error is expected.
```
    LogInfo( ( "********** INFO : Entered SecureSocketsTransport_Connect.\n" ) )
```

**Below are some related observations.**

1. LogError prints the debug message as desired. Console log is as below.
```
1336 26972 [iot_thread] [ERROR] [Transport_Secure_Sockets] [transport_secure_sockets.c:487] 1337 26972 [iot_thread] ********** INFO : Entered SecureSocketsTransport_Connect.
```

2. Removing the semi-colon at the end of the LogError debug message generates a compilation/ build error, which is as desired.
```
    LogError( ( "********** INFO : Entered SecureSocketsTransport_Connect.\n" ) )
```

3. configPRINTF() prints the debug message on the console. Thank you for suggesting this alternative 👍 
```
1339 26972 [iot_thread] ********** INFO : Entered SecureSocketsTransport_Connect.
```

Let me know if any additional testing is required and/ or, a new bug for the LogInfo observation needs to be logged.

Meanwhile, still struggling to resolve the original issue.

Thanks | Regards,
Dipen

dipen-1337lab, 2021-03-11:
Namaste All,

Sharing an update after identifying the reason for this issue and, kind-off resolving the same.

So, in the ESP-IDF Release Version 3.3, which is part of the AFR 202002.00 we are using for development, there's this `ESP_THREAD_SAFE` macro defined.

Within the `lwip_close` function in `sockets.c` file, the [freeing of the `free_socket` is conditional](https://github.com/espressif/esp-lwip/blob/release/v3.3/src/api/sockets.c#L877), based on `ESP_THREAD_SAFE`, which is set/ enabled by default. With CONFIG_LWIP_MAX_SOCKETS set to 10 in sdkconfig.h file for ESP-IDF, new sockets were getting created up to 10, after which the original error was resulting.

I've modified the logic as below, which in a way makes the original #if !ESP_THREAD_SAFE check void.
```
#if !ESP_THREAD_SAFE
    free_socket(sock, is_tcp);
#else
    free_socket(sock, is_tcp);
#endif
```

Browsing through the `sockets.c` file in the master branch brings out the fact that the [`ESP_THREAD_SAFE` is no more used](https://github.com/espressif/esp-lwip/blob/master/src/api/sockets.c#L807).

I'm closing this issue but would appreciate your inputs if there's a better way to implement `free_socket` than what I have explained above. Our application firmware is working fine with this change.

Once again, appreciate each one's assistance on this issue. Thank you!

Thanks | Regards,
Dipen

closed 2021-03-11 by dipen-1337lab

mahavirj, 2021-03-11:
@dipen-1337lab 

As I had mentioned in my earlier comment at https://github.com/aws/amazon-freertos/issues/3014#issuecomment-788996856, if you are using `secure_socket` layer from ESP32 `ports` directory then you should not require this fix. 

Please see below code snippet: 

https://github.com/aws/amazon-freertos/blob/987d9697ad4dc924bd71333c57b2f68bbe81b172/vendors/espressif/boards/esp32/ports/secure_sockets/lwip/iot_secure_sockets.c#L118-L129

For v3.3 release, we recommended use of posix API (routed through our VFS layer), rather than using direct lwIP calls. This allowed applications to have thread safe assumptions while using same socket from multiple task context. This was already handled in `secure_socket` layer as shown above (use of `close` rather `lwip_close`).

Can you please create simple test application on v202002 release which can reproduce this issue, I can try out on my side?

Test code could be as simple as:
```
while (iterations--) {
   sock = SOCKETS_Socket();
   SOCKETS_Close(sock);
}
```

dipen-1337lab, 2021-03-31:
Namaste @mahavirj, 

> Can you please create simple test application on v202002 release which can reproduce this issue, I can try out on my side?

At the outset, I apologise for this delayed response to your above question. As mentioned in my original post, we have ported the coreHTTP library for using with Amazon FreeRTOS version V202002.00. Could that be the cause for my original issue, further resulting in an unexpected behaviour that requires a fix, which should not be required?
> * coreHTTP library has been ported for use with V202002.00. More details [here](https://github.com/aws/amazon-freertos/issues/2911).

Further, regret my inability to assist you in reproducing the issue on AFR V202002.00, at this juncture. If possible, I request your assistance to understand if the issue is reproducible on AFR V202002.00.

Thanks | Regards,
Dipen