aws/amazon-freertos #3018: @pvyawaha, @nickfaughey Hello. Is there a way to implement this in the latest AWS FreeRTOS ESP32 version?

Yoimer, 2021-02-28:
@pvyawaha, @nickfaughey Hello. Is there a way to implement this in the latest AWS FreeRTOS ESP32 version?

_Originally posted by @Yoimer in https://github.com/aws/amazon-freertos/issues/2569#issuecomment-786667185_

Yoimer, 2021-03-01:
@lundinc2 Hello, thanks for taking your time to respond. Basically, I want to achieve the same @pvyawaha did here [V202002.00](https://github.com/aws/amazon-freertos/issues/2569#issuecomment-708716699) but now using the `V202012.00`. There are a lot of differences between from `V202002.00` to `V202012.00`. I am using AWS FreeRTOS ESP32 `V202012.00`.
I appreciate your help on this. Thanks in advance.

dan4thewin, 2021-03-04:
Hi, @Yoimer .  I think the MQTT agent project addresses this need.  Have a look: https://github.com/FreeRTOS/Lab-Project-coreMQTT-Agent/

Yoimer, 2021-03-04:
Hello @dan4thewin thanks a lot for replying. I am checking this [ota_over_mqtt_demo.c](https://github.com/FreeRTOS/Lab-Project-coreMQTT-Agent/blob/main/source/ota_over_mqtt_demo.c) and it says it is `FreeRTOS V202011.00`. Currently I am using the `FreeRTOS V202012.00`. Would the code you recommended work for the `FreeRTOS V202011.00`? Thanks in advance for your help.

dan4thewin, 2021-03-06:
I believe it will work.  It has recent commits.  If you encounter problems, please let us know.  We're here to help.

Yoimer, 2021-03-06:
@dan4thewin Thanks a lot for replying and your time. I will test and let you know

dan4thewin, 2021-03-08:
De nada.  I'll close this now.  Please feel free to open another issue if you encounter problems.  Thanks.

closed 2021-03-08 by dan4thewin

Yoimer, 2021-03-19:
> De nada. I'll close this now. Please feel free to open another issue if you encounter problems. Thanks.

@dan4thewin Thanks. Sure. I will open another one in a few minutes