aws/amazon-freertos #3019: Running the Greengrass Discovery demo with ECDSA

vvelciu, 2021-03-01:
Hello,

I managed to run the Greengrass Discovery demo with ECDSA (I generated the keys for the device and the Greengrass core as described [here,](https://aws.amazon.com/it/blogs/iot/elliptic-curve-cryptography-and-forward-secrecy-support-in-aws-iot-3/) but I needed to make some changes in the _iot_tls.c_ file. The default root certificates did not include the ATS3 one, so I manually added it:
[image]
Is there any proper way of doing this? Greengrass is hardcoded to using default certificates:
[image]

Thanks,
Veronica

_Originally posted by @vvelciu in https://github.com/aws/amazon-freertos/issues/2406#issuecomment-783405838_

lundinc2, 2021-03-01:
Hi @vvelciu,

This approach is correct. I will consider adding ATS3 to the default chain, and will post back here either with further comments or a PR you can track.

Thanks,

Carl

lundinc2, 2021-03-04:
Hi @vvelciu I have merged said change. Please re-open this issue if you have further questions.

closed 2021-03-04 by lundinc2