aws/amazon-freertos #3020: How to use DHT22, MQ2,BMP180 and ublox-NEO-6M on AWS FreeRTOS ESP32

Yoimer, 2021-03-01:
Hello there. How could I possibly use those sensors (DHT22, MQ2,BMP180) and ublox-NEO-6M GPS on your latest AWS FreeRTOS  ESP32 version(202012.00 LTS). I successfully ran those on Arduino IDE using their respective ESP32 libraries. Is there any way I could import those ESP32 Arduino Libraries on the AWS FreeRTOS ESP32 code?
Would those libraries work exactly the same on AWS FreeRTOS the same way they do on Arduino ESP32?
Is there any sort of tutorial or documentation I could get that information from?. Looking forward to hearing from you soon. Thanks in advance.

lundinc2, 2021-03-01:
Hi @Yoimer,

I am investigating this and will provide some more details for you shortly.

Thanks,

Carl

Yoimer, 2021-03-01:
@lundinc2 Thanks a lot for your cooperation.

Yoimer, 2021-03-02:
Hello @lundinc2. Double checking here with my crew, instead of DTH22, we are using DHT11. Thanks in advance

elieDaan, 2021-03-02:
I think you just need to find the equivalent esp idf (v3.3) libraries of your component. AWS Freertos for ESP32 is using esp-idf v3.3. 

For example,  [BMP180 esp idf ](https://github.com/krzychb/bmp180-tests/tree/master/components)

Yoimer, 2021-03-02:
@elieDaan Thanks a lot for commenting. Yes, you're right AWS FreeRTOS for ESP32 `esp-idf v3.3`. However, I was able to adapt the `ESP-IDF 4.2` to compile, and even flash code to AWS FreeRTOS ESP32 `V202012.00`(latest one). I am attaching snapshots of the configuration.

Do you think that using the `ESP-IDF 4.2` would cause any issues with the components section on FreeRTOS?

Thanks in advance.

[image]
[image]

elieDaan, 2021-03-02:
Sorry, I'm wrong. I am outdated about the last updates on AWS AFR.. It seems that AFR is now supporting [esp-idf v4.2](https://github.com/espressif/esp-afr-sdk)

paulbartell, 2021-03-02:
@Yoimer, the libraries that @elieDaan linked to look like a good place to start.

When porting new drivers to a system using FreeRTOS, you will need to place calls to drivers that are not thread safe or are timing critical in a [critical section](https://www.freertos.org/taskENTER_CRITICAL_taskEXIT_CRITICAL.html) to prevent the scheduler from interrupting them.

The DHT11 is particularly affected by this because the interface bit-banged via a gpio pin. [This driver](https://github.com/UncleRus/esp-idf-lib/blob/master/components/dht/dht.c) uses critical sections when necessary.

Regarding GPS / GNSS, Here's a link to an [esp-idf example](https://github.com/igrr/libnmea-esp32) for parsing NMEA sentences produced by modules like the NEO-6M.

Yoimer, 2021-03-02:
@paulbartell Thanks a lot for replying. At this stage I was able to run DHT11 and BMP180 sensor on `IDF 4.2` from [UncleRus](https://github.com/UncleRus/esp-idf-lib). I haven't tried that on AWS FreeRTOS ESP32 (but that's my goal). Is this the right path then?
I will be checking the NEO-6M link you passed me. Thanks a lot for your help and guidance.

paulbartell, 2021-03-03:
@Yoimer  If you are only targeting espressif platforms, the [esp-aws-iot repository](https://github.com/espressif/esp-aws-iot) from espressif or the [Core2-for-AWS-IoT-EduKit repository](https://github.com/m5stack/Core2-for-AWS-IoT-EduKit) from m5stack could be better places to start.

This repository is a set of [reference integrations / demos ](https://www.freertos.org/aws-reference-integrations.html) for connecting a variety of microcontroller evaluation boards to AWS and other cloud services. It's meant to give you examples of how to integrate parts of the [aws iot device sdk](https://github.com/aws/aws-iot-device-sdk-embedded-C ) into a cmake style project that targets platforms from multiple vendors.

I'm going to close this Issue since it's more of a general inquiry. Feel free to make a thread on the [FreeRTOS forums](https://forums.freertos.org/) and we can address any followup questions over there.

closed 2021-03-03 by paulbartell