aws/amazon-freertos #3025: MQTT has issues reconnecting after wifi gets beacon timeout disconnect

MrMelon1232, 2021-03-03:
As the title suggests, wifi gets a beacontimeout and reconnects but mqtt has trouble reconnecting afterwards.
Here are the outputs
```
321772 6540417 [old mqtt app] [INFO ][OLD_MQTT_APP][65404170] Wifi is connected!
321773 6540418 [old mqtt app] [INFO ][OLD_MQTT_APP][65404180] Attempting mqtt connection...
321802 6541176 [old mqtt app] [ERROR] [PKCS11] [core_pkcs11_mbedtls.c:3922] 321803 6541177 [old mqtt app] Failed sign operation. mbedtls_pk_sign failed: mbed TLS error = RSA - The private key operation failed : BIGNUM - Memory allocation failed.
321805 6541178 [old mqtt app] ERROR: Failure in signing callback: 6 
321806 6541179 [old mqtt app] ERROR: Handshake failed with error code <No-High-Level-Code> : <No-Low-Level-Code> 
321807 6541181 [old mqtt app] TLS_Connect fail (0x7d3, aau8njf5d2xbz-ats.iot.us-east-1.amazonaws.com)
321808 6541182 [old mqtt app] [ERROR][NET][65411820] Failed to establish new connection. Socket status: -1.
321809 6541185 [old mqtt app] [ERROR][MQTT][65411830] Failed to establish new MQTT connection, error NETWORK ERROR.
321810 6541187 [old mqtt app] [ERROR][OLD_MQTT_APP][65411860] Error in establishing a mqtt connection...
321811 6541188 [old mqtt app] [ERROR][OLD_MQTT_APP][65411880] Mqtt failed...
```

Any type of response is appreciated!

yanjos-dev, 2021-03-11:
Hi @MrMelon1232, what platform and version of the repository or you using in your code?

MrMelon1232, 2021-03-18:
Hi @yanjos-dev,

I am currently using esp32 with this commit of the repository ```dc53c9cbe68977e0892dcbfb023256d4055d1dd7```. I have solved the memory leak issue which turned out to be on my end but the beacon timeout still occurs.

MrMelon1232, 2021-04-01:
This issue has been solved! I will be closing it.

closed 2021-04-01 by MrMelon1232