aws/amazon-freertos #3026: ESP32 - Cannot use sdkconfig.defaults from custom location

amiotk, 2021-03-03:
**Describe the bug**
I'm using amazon-freertos in existing CMake project and would like to use sdkconfig.defaults from custom directory. I was hopping that specifying `IDF_SDKCONFIG_DEFAULTS` should make it work (like is does for native ESP-IDF projects) but it does not due to hard-coded path to sdkconfig.default in one of [CMakeLists.txt](https://github.com/aws/amazon-freertos/blob/066e66a1bee61daebace8482fdcc1bd1b19196f8/vendors/espressif/boards/esp32/CMakeLists.txt#L581-L584):
```cmake
idf_build_process(esp32
                    SDKCONFIG ${CMAKE_BINARY_DIR}/sdkconfig
                    SDKCONFIG_DEFAULTS ${CMAKE_BINARY_DIR}/sdkconfig.defaults
                    BUILD_DIR ${CMAKE_BINARY_DIR})
```

I can make it work by changing the path to `IDF_SDKCONFIG_DEFAULTS`:
```cmake
idf_build_process(esp32
                    SDKCONFIG ${CMAKE_BINARY_DIR}/sdkconfig
                    SDKCONFIG_DEFAULTS ${IDF_SDKCONFIG_DEFAULTS}
                    BUILD_DIR ${CMAKE_BINARY_DIR})
```

I wonder if this solution is correct and would be accepted as PR?

**System information**
- esp32-devkitc
- VsCode
- Linux
- 202012.00-227-g9205e3c94
- Custom Application

**To reproduce**
Steps to reproduce the behavior:

1. Set `IDF_SDKCONFIG_DEFAULTS`
2. Delete build directory
3. CMake configure

shubhamkulkarni97, 2021-03-04:
Hi @amiotk, thanks for reporting this issue.

Solution proposed by you looks good. Would you like to create a PR with the change? Or else I'll create one.

Thanks,
Shubham

amiotk, 2021-03-04:
Hi @shubhamkulkarni97

Sure, PR #3027 is now ready for review.

lundinc2, 2021-03-09:
Hi @shubhamkulkarni97 and @amiotk,

This solution works for me, but I have on question regarding this snippet from the [IDF documentation](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-guides/build-system.html#custom-sdkconfig-defaults). 

"""
To override the name of this file or to specify multiple files, set the SDKCONFIG_DEFAULTS environment variable or set SDKCONFIG_DEFAULTS in top-level CMakeLists.txt. If specifying multiple files, use semicolon as the list separator. File names not specified as full paths are resolved relative to current project.
"""

When I try and use a relative path, I have the following issue.

```
Loading defaults file sdkconfig.minimal...
Traceback (most recent call last):
  File "/Users/lundinc/Documents/amazon-freertos/vendors/espressif/esp-idf/tools/kconfig_new/confgen.py", line 641, in <module>
    main()
  File "/Users/lundinc/Documents/amazon-freertos/vendors/espressif/esp-idf/tools/kconfig_new/confgen.py", line 285, in main
    raise RuntimeError("Defaults file not found: %s" % name)
RuntimeError: Defaults file not found: sdkconfig.minimal
CMake Error at vendors/espressif/esp-idf/tools/cmake/kconfig.cmake:225 (message):
  Failed to run confgen.py
  (python;/Users/lundinc/Documents/amazon-freertos/vendors/espressif/esp-idf/tools/kconfig_new/confgen.py;--kconfig;/Users/lundinc/Documents/amazon-freertos/vendors/espressif/esp-idf/Kconfig;--sdkconfig-rename;/Users/lundinc/Documents/amazon-freertos/vendors/espressif/esp-idf/sdkconfig.rename;--config;/Users/lundinc/Documents/amazon-freertos/build/sdkconfig;--defaults;sdkconfig.minimal;--env-file;/Users/lundinc/Documents/amazon-freertos/build/config.env).
  Error 1
Call Stack (most recent call first):
  vendors/espressif/esp-idf/tools/cmake/build.cmake:442 (__kconfig_generate_config)
  vendors/espressif/boards/esp32/CMakeLists.txt:581 (idf_build_process)
  CMakeLists.txt:70 (include)
```

See also:
https://forums.freertos.org/t/using-a-custom-sdkconfig-defaults-file/11908/5

Thanks,

Carl

shubhamkulkarni97, 2021-03-10:
Hi @lundinc2,

The documentation snippet you mentioned is true when building IDF project. In AFR, IDF is used as a library. IDF build system uses value of `SDKCONFIG_DEFAULTS` defaults argument passed in idf_build_process [here](https://github.com/aws/amazon-freertos/blob/858a26e9043c2baea1b95ffc38d33e5bb8ea1314/vendors/espressif/boards/esp32/CMakeLists.txt#L583). 

Please check documentation for idf_build_process in [CMake Build System API](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-guides/build-system.html#esp-idf-cmake-build-system-api) which specifies the following about `SDKCONFIG_DEFAULTS`:

> SDKCONFIG_DEFAULTS - list of files containing default config to use in the build (list must contain full paths); defaults to empty. For each value filename in the list, the config from file filename.target, if it exists, is also loaded.

Please find the patch attached below and check if it works.

[0001-esp32-Fix-issue-with-custom-sdkconfig.defaults.patch.zip](https://github.com/aws/amazon-freertos/files/6114183/0001-esp32-Fix-issue-with-custom-sdkconfig.defaults.patch.zip)

Thanks,
Shubham

shubhamkulkarni97, 2021-04-06:
This issue can be closed as the fix is already merged

closed 2021-04-06 by amiotk

archigup, 2021-04-06:
Thanks! Closing the issue.