aws/amazon-freertos #3031: Contradictory licenses

xypron, 2021-03-07:
The LICENSE file on the top level is an MIT license.
The license conditions in the source files of the vendors/ directory are very restrictive.
It is unclear which license conditions are binding.
If the MIT license is not applicable to the whole repository, the top-level LICENSE file should be removed and replaced by SPDX license identifiers in each individual file.

yanjos-dev, 2021-03-11:
Thank you for bringing this to our attention, we are looking into it.

aggarw13, 2021-05-25:
Thanks again for your feedback. To make the vendor licensing easier to understand and follow, we've added a new root level [VENDOR_LICENSES](https://github.com/aws/amazon-freertos/blob/main/vendors/VENDOR_LICENSES) file as well as added [information in the README](https://github.com/aws/amazon-freertos#amazon-freertosvendors-license)  to clarify that the vendor directory contains content that may be subject to different license terms that are described in a LICENSE file or source header documentation in each vendor directory.

yanjos-dev, 2021-06-17:
I'm closing this ticket due to inactivity. Please reopen this issue if you have any further questions.

closed 2021-06-17 by yanjos-dev