aws/amazon-freertos #3033: Add support to use a device certificate without a registered CA

cmbpereira, 2021-03-08:
I using the module ESP32-WROOM-32SE and I have registered a client certificate without a registered CA using only the device certificate of the secure element.
The first problem is the template id of my certificate in the secure element is id 3 and the FreeRTOS only found support for template id 1 and id 2.
The second problem is in the io_tls.c I don't have a way to disable the _"Add a Just-in-Time Registration (JITR) device issuer certificate"_ and this routine makes the certificate read from the secure element invalid. 

Thank you!

lundinc2, 2021-03-15:
Hi @cmbpereira,

I have added modifying the JITP/JITR logic to our backlog. For now can you comment out the loading of this certificate.

Can you outline the difficulties for the certificate template id? I am not familiar with this parameter, and would like more details so we can consider this task.

Thanks,

Carl

cmbpereira, 2021-03-16:
Hi @lundinc2 

Thanks to take care of this situation.

The template id is related to the secure element from Microchip (ATECC608A) and in this file, 
_amazon-freertos/libraries/abstractions/pkcs11/ecc608a/core_pkcs11_secure_element.c,_ when we select to use the Device certificate the certificate template load is with id 2.

```
if( !strncmp( pkcs11configLABEL_DEVICE_CERTIFICATE_FOR_TLS, ( char * ) pLabel->pValue, pLabel->ulValueLen ) )
    {
        /* Slot 10 - Device Cert for Slot 0*/
        pkcs11_config_init_cert( pObject, pLabel->pValue, pLabel->ulValueLen );
        pObject->slot = 10;
        pObject->class_type = CK_CERTIFICATE_CATEGORY_TOKEN_USER;
        pObject->size = g_cert_def_2_device.cert_template_size;
        pObject->data = &g_cert_def_2_device;
        #ifdef FREERTOS_ENABLE_UNIT_TESTS
            pObject->flags = PKCS11_OBJECT_FLAG_DESTROYABLE;
        #endif
    }
```

And the templates are defined in this file _amazon-freertos/libraries/abstractions/pkcs11/ecc608a/atca_cert_chain.c_, but the certificate template with id 3 isn't defined, so have to use the template defined here.
[https://github.com/espressif/esp-cryptoauthlib/blob/c3d3a69021cfec3236ca2c0b63be4048ec6643a4/cryptoauthlib/app/tng/tngtls_cert_def_3_device.c](https://github.com/espressif/esp-cryptoauthlib/blob/c3d3a69021cfec3236ca2c0b63be4048ec6643a4/cryptoauthlib/app/tng/tngtls_cert_def_3_device.c)

Thanks,
Carlos