aws/amazon-freertos #3035: Failed to establish new connection. Socket status: -116

aogrcs, 2021-03-09:
I followed the quick connect guide using B-L475, but errors as following:

15 11620 [iot_thread] [ERROR][NET][11620] Failed to establish new connection. Socket status: -116.
16 11620 [iot_thread] [ERROR][MQTT][11620] Failed to establish new MQTT connection, error NETWORK ERROR.
17 11620 [iot_thread] [ERROR][DEMO][11620] MQTT CONNECT returned error NETWORK ERROR.
18 11620 [iot_thread] [INFO ][MQTT][11620] MQTT library cleanup done.
19 11620 [iot_thread] [INFO ][DEMO][11620] memory_metrics::freertos_heap::before::bytes::2085856
20 11620 [iot_thread] [INFO ][DEMO][11620] memory_metrics::freertos_heap::after::bytes::2057664
21 11620 [iot_thread] [INFO ][DEMO][11620] memory_metrics::demo_task_stack::before::bytes::1908
22 11620 [iot_thread] [INFO ][DEMO][11620] memory_metrics::demo_task_stack::after::bytes::1908
23 11620 [iot_thread] [ERROR][DEMO][11620] Error running demo.
24 11623 [iot_thread] [INFO ][INIT][11623] SDK cleanup done.
25 11623 [iot_thread] [INFO ][DEMO][11623] -------DEMO FINISHED-------`

htibosch, 2021-03-09:
I'm assuming that you are using FreeRTOS+TCP?

You may have found out that FreeRTOS+TCP function `FreeRTOS_connect()` will return `-pdFREERTOS_ERRNO_ETIMEDOUT`.

There is a function `FreeRTOS_strerror_r()`, which returns a human readable string for each errno value used in the sources.

There may be many reasons for a timeout on connect: your timeout is set too short, the IP-address is incorrect, the gateways is not configured properly, or not responding.

archigup, 2021-04-05:
Hi, I'm resolving this issue. Feel free to open a new issue if you have further questions on this topic.

closed 2021-04-05 by archigup

aogrcs, 2021-04-08:
@archigup, how to resolve it? thanks