aws/amazon-freertos #3036: How to add esp32-wifi-manager in aws_greengrass_discovery_demo.c program?

monhack, 2021-03-09:
Hi @all,

I want to ask, how do I add a wifi manager function to the program? i got no documentation about it. The problem is that I have to manually connect freertos with the configuration in the aws_clientcredential.h program, this method will not be effective.

Yoimer, 2021-03-11:
@monhack Hi. I am trying to the same with OTA and coreMQTT demos. Have you found anything? By the way. Have you been able to bring sensors (like DHT, MQ2, BMP180 just to name a few) to your greengrass demo? Feel free to reply. Thanks a lot

yanjos-dev, 2021-03-13:
Hello @monhack, what version of the repository are you using?

monhack, 2021-03-15:
@Yoimer I just focused in greengrass and freertos development for my IoT Project. I able to bring sensors in pro mini, not esp32 (freertos). So I send data sensors to freertos using UART, freertos publish data to greengrass using shadow

monhack, 2021-03-15:
@yanjos-dev i'm using freertos version 202007.00

Yoimer, 2021-03-15:
@monhack Hello. Thanks for replying and sharing your experience. By the way did you get any answer about integrating the greengrass demo with the wifi_provisionin?. Thanks again!.

yanjos-dev, 2021-03-15:
@monhack, can you clarify what you are referring to when you say Wi-Fi manager? If this is a module that is specific to Espressif, then I would recommend raising this question on the Espressif section of the FreeRTOS forums ([link](https://forums.freertos.org/c/partners-sponsors/espressif/14)).

closed 2021-03-17 by monhack