aws/amazon-freertos #3040: ESP32 doesn't reset after OTA update completed

ahmedwahdan, 2021-03-10:
Describe the bug
After OTA agent receives all firmware blocks with valid signature a problem happens with the some event keeps ESP32 waiting for an event for ever.

```
11401 33203 [OTA Agent Task] [prvIngestDataBlock] Remaining: 1

11402 33203 [OTA Agent Task] [prvExecuteHandler] Called handler. Current State [WaitingForFileBlock] Event [ReceivedFileBlock] New state [WaitingForFileBlock]

11403 33203 [OTA Agent Task] [DEBUG][THREAD][332030] Locking mutex 0x3ffec1e8.

11404 33203 [OTA Agent Task] [DEBUG][THREAD][332030] Unlocking mutex 0x3ffec1e8.

11405 33203 [OTA Agent Task] [DEBUG][THREAD][332030] Locking mutex 0x3ffec1e8.

11406 33203 [OTA Agent Task] [DEBUG][THREAD][332030] Unlocking mutex 0x3ffec1e8.

11407 33203 [OTA Agent Task] [DEBUG][THREAD][332030] Locking mutex 0x3ffec1e8.

11408 33205 [OTA Agent Task] [DEBUG][THREAD][332050] Locking mutex 0x3ffec1e8.

11409 33205 [OTA Agent Task] [DEBUG][THREAD][332050] Unlocking mutex 0x3ffec1e8.

11410 33205 [OTA Agent Task] [DEBUG][THREAD][332050] Locking mutex 0x3ffec1e8.

11411 33205 [OTA Agent Task] [DEBUG][THREAD][332050] Unlocking mutex 0x3ffec1e8.

11412 33205 [OTA Agent Task] [DEBUG][THREAD][332050] Locking mutex 0x3ffec1e8.

11413 33205 [OTA Agent Task] [DEBUG][THREAD][332050] Unlocking mutex 0x3ffec1e8.

11414 33205 [OTA Agent Task] [DEBUG][THREAD][332050] Unlocking mutex 0x3ffec1e8.

11415 33206 [OTA Agent Task] [INFO ][MQTT][332050] (MQTT connection 0x3ffec1d0) MQTT PUBLISH operation queued.

11416 33207 [OTA Agent Task] [prvRequestFileBlock_Mqtt] OK: $aws/things/CLI_THING_2/streams/AFR_OTA-7d78f0ff-2aa7-455d-b126-c5970e97c8bb/get/cbor

11417 33208 [OTA Agent Task] [prvExecuteHandler] Called handler. Current State [WaitingForFileBlock] Event [RequestFileBlock] New state [WaitingForFileBlock]

11418 33285 [iot_thread] [INFO ][DEMO][332850] State: WaitingForFileBlock  Received: 288   Queued: 0   Processed: 0   Dropped: 0

11419 33317 [NetRecv] [DEBUG][NET][333170] Successfully received 1 bytes.

11420 33317 [NetRecv] [DEBUG][NET][333170] Successfully received 1 bytes.

11421 33317 [NetRecv] [DEBUG][NET][333170] Successfully received 1 bytes.

11422 33318 [NetRecv] [DEBUG][NET][333180] Successfully received 3586 bytes.

11423 33319 [NetRecv] [DEBUG][THREAD][333190] Locking mutex 0x3ffec1e8.

11424 33319 [NetRecv] [DEBUG][THREAD][333190] Unlocking mutex 0x3ffec1e8.

11425 33320 [NetRecv] [DEBUG][THREAD][333200] Locking mutex 0x3ffec1e8.

11426 33321 [NetRecv] [DEBUG][THREAD][333210] Unlocking mutex 0x3ffec1e8.

11427 33321 [NetRecv] [DEBUG][THREAD][333210] Locking mutex 0x3ffce418.

11428 33322 [NetRecv] [DEBUG][THREAD][333220] Posting to semaphore 0x3ffce334.

11429 33323 [iot_thread] [DEBUG][THREAD][333230] Locking mutex 0x3ffce418.

11430 33323 [NetRecv] [DEBUG][THREAD][333230] Unlocking mutex 0x3ffce418.

11431 33323 [iot_thread] [DEBUG][THREAD][333230] Unlocking mutex 0x3ffce418.

11432 33323 [iot_thread] [DEBUG][THREAD][333230] Locking mutex 0x3ffec1e8.

11433 33323 [iot_thread] [DEBUG][THREAD][333230] Unlocking mutex 0x3ffec1e8.

[... 198 lines cut ...]
11534 51323 [iot_thread] [DEBUG][THREAD][513230] Locking mutex 0x3ffce418.

11535 51323 [iot_thread] [DEBUG][THREAD][513230] Unlocking mutex 0x3ffce418.

11536 57201 [iot_thread] [WARN ][THREAD][572010] Timeout waiting on semaphore 0x3ffce334.

11537 57201 [iot_thread] [DEBUG][THREAD][572010] Locking mutex 0x3ffce418.

11538 57201 [iot_thread] [DEBUG][THREAD][572010] Unlocking mutex 0x3ffce418.

11539 57323 [iot_thread] [WARN ][THREAD][573230] Timeout waiting on semaphore 0x3ffce334.

11540 57323 [iot_thread] [DEBUG][THREAD][573230] Locking mutex 0x3ffce418.

11541 57323 [iot_thread] [DEBUG][THREAD][573230] Unlocking mutex 0x3ffce418.
```

System information

ESP32 Wroom kit
IDE used : NA
Operating System : Windows
Version of FreeRTOS : master latest
Project OTA Demo
Thank you!

Yoimer, 2021-03-11:
@ahmedwahdan Hi. Please, try the `FreeRTOS V202012.00` from the zip file generated from `AWS FreeRTOS Console` choosing the OTA demo. Make sure you follow the requisites as shown here [ota-prereqs](https://docs.aws.amazon.com/freertos/latest/userguide/ota-prereqs.html). I tested here and all works OK

ahmedwahdan, 2021-03-11:
This release uses ESP IDF v3. 3, I want to use v4.2.

Yoimer, 2021-03-11:
Actually I am using `ESP-IDF V4.2` with `FreeRTOS V202012.00`. You just need to adapt the `.espressiff\idf_cmd_init.bat` to test the demos

Yoimer, 2021-03-11:
@ahmedwahdan Have you been able to bring sensors (DHT, BMP180 to name a few) to the AWS FreeRTOS ESP32 demos?

yanjos-dev, 2021-03-13:
Hi @ahmedwahdan, it looks like this recent fix ( https://github.com/aws/amazon-freertos/commit/07fd8834962b936323eddfb81a40961cc0d2f928) is likely related to your issue. Can you try running your code after this commit?

ahmedwahdan, 2021-03-13:
Hello @yanjos-dev  The logs are from latest master which includes these changes.

pvyawaha, 2021-03-31:
Hello,

This was due to an issue in the IotNetworkAfr_Close , which is being addressed in the PR https://github.com/aws/amazon-freertos/pull/3082

closed 2021-04-01 by pvyawaha