aws/amazon-freertos #3044: How to configure AP in amazon freertos with a static IP Address

Johnnyjax, 2021-03-15:
in the esp-idf, I configure my AP with a static IP Address with the following lines of code:

            esp_netif_t* wifiAP = esp_netif_create_default_wifi_ap();
            esp_netif_ip_info_t ipInfo;
            IP4_ADDR(&ipInfo.ip, 192,168,2,1);
            IP4_ADDR(&ipInfo.gw, 192,168,2,1);
            IP4_ADDR(&ipInfo.netmask, 255,255,255,0);
            esp_netif_dhcps_stop(wifiAP);
            esp_netif_set_ip_info(wifiAP, &ipInfo);
            esp_netif_dhcps_start(wifiAP);

I can't seem to do the same for Amazon freertos and it seems there's no built in method for configuring a static Ip Address. The WIFI_ConfigureAP() function in amazon freertos doesn't seem to have this functionality and I've tried including esp_netif.h but it shows "no such file or directory". Is there a way I can do this? Thanks.

yanjos-dev, 2021-03-15:
Hello @Johnnyjax. What version of the repository are you using? How are you building your code? This looks like an issue where you are not able to include esp_netif.h. Do you have the path to this file in your include path when you compile the code you're referring to?

Johnnyjax, 2021-03-16:
Thanks @yanjos-dev. I was using the esp idf 3.3 with my amazon freertos. Upgrading to 4.2 made the esp_netif.h library available.

closed 2021-03-16 by Johnnyjax