aws/amazon-freertos #3048: How to integrate coreMQTT(mqtt_demo_mutual_auth.c) with wifi_provisioning(aws_wifi_connect_task.c) on AWS ESP 32??

Yoimer, 2021-03-15:
Hello there guys. Could you please be kind enough to help me out on integrating both demos from the `FreeRTOS V202012.00`. Basically I want to send data to AWS using the coreMQTT example using the wifi_provisioning mobile app over BLE.
At this stage, I have (all the time) to hardcode the ssid and password on the `aws_clientcredential.h` file. Thanks in advance.

ravibhagavandas, 2021-03-18:
Hi @Yoimer,

You can enable WiFi provisioning over BLE by setting the flag `IOT_BLE_ENABLE_WIFI_PROVISIONING` to `1` as mentioned in this doc: https://docs.aws.amazon.com/freertos/latest/userguide/ble-demo.html#ble-demo-wifi
Please note that you don't need to do step 2: `Configure the Network Manager to enable both Bluetooth Low Energy and Wi-Fi.` as the coreMQTT  demo runs only on WiFi.
Please follow further instructions in document under `To run the demo` to provision a wifi credential onto the device. Using network manager APIs, the demo waits for a network to be configured before running the demo entry function. The task `aws_wifi_connect_task.c` will take care of reconnecting to a provisioned network on device reboot or when the current connection is lost.

muneebahmed10, 2021-03-25:
Hello @Yoimer,

Did the [above](https://github.com/aws/amazon-freertos/issues/3048#issuecomment-802282528) response answer your question? We will close this issue if we do not hear back from you.

Thank you

Yoimer, 2021-03-25:
@muneebahmed10 Hello and thanks for your concern. Actually no. I know how to run the wifi_provisioning demo isolated. I'd like to know how to integrate that demo with the rest. I just do not want to keep hardcoding wifi credentials on the tests.
Thanks in advance.

Yoimer, 2021-03-25:
@muneebahmed10 Please if you got time check this out https://github.com/aws/amazon-freertos/issues/3056#issuecomment-806214722. I was able to handle a MQTT and OTA integration but on `V202002.00` version. The goal on this version is the same. I want to use the BLE Wifi_Provisioning capabilities.
Thanks again.

ravibhagavandas, 2021-05-24:
@Yoimer 
Since the question posted in this forum is already answered in #3056, could we close this issue and keep the other one?

Yoimer, 2021-05-24:
@ravibhagavandas hello. Thanks for commenting. Replying to your question, yes. Please close this issue and let's carry on on #3053. Have a great day.

ravibhagavandas, 2021-05-24:
Thanks, Closing this issue.

closed 2021-05-24 by ravibhagavandas