aws/amazon-freertos #3051: Partition offset 0x10000 overlaps end of partition table 0x11000

Raghav3107, 2021-03-16:
I was trying to use an OTA with  Secure Boot and Flash Encryption. 
When  I trying to build the application it's giving an error.

Log:
CSV Error: First partition offset 0x10000 overlaps end of partition table 0x11000
make: *** [/home/horsemann/Desktop/WorkSpace/WIFIlist_fix/ota_test_starboard/vendors/espressif/esp-idf/components/partition_table/Makefile.projbuild:61: /home/horsemann/Desktop/WorkSpace/WIFIlist_fix/ota_test_starboard/vendors/espressif/boards/esp32/aws_demos/build/partition-table-unsigned.bin] Error 2

Partition Table:

[image]

Can anyone help me in this 
Thanks In Advance

ravibhagavandas, 2021-03-19:
Hi @Raghav3107

By default the secure boot and flash encryption is not enabled on the esp32 port in this repo so I am not able to reproduce this behavior. I tried the above partition table without secure boot and flash encryption option and it builds.

However after going through ESP docs, enabling secure boot and flash encryption can increase the size of the bootloader. So it could be possible that the partition table offset (0x8000) configured is not enough. Did you try adjusting the partition table offset as mentioned in this doc:
https://docs.espressif.com/projects/esp-idf/en/latest/esp32/security/secure-boot-bootloader-size.html#secure-boot-bootloader-size

Raghav3107, 2021-03-20:
Hello @ravibhagavandas
Thanks for the reply.

I already tried adjusting the partition table offset from (0x8000) to (0x10000).
But It giving the same error.

ravibhagavandas, 2021-03-22:
Hi @Raghav3107 ,
I think from the error log it looks like the  first partition (nvs) address (0x10000) starts within the partition table boundary , partition table spans up to 0x11000. So I think adjusting the partition offset will not help, instead did you try change the partition table such that `nvs` partition starts from offset 0x11000 and progressively update start offsets of other partitions.
Example:
```
# Name, Type, SubType, Offset, Size, Flags
nvs,      data, nvs,     0x11000,  0x6000
otadata,  data, ota,     0x17000,  0x2000
phy_init, data, phy,     0x19000,  0x1000
factory,  0,    factory, 0x20000,  1400K
ota_0,    0,    ota_0,   ,  1400K
ota_1,    0,    ota_1,   ,         200K
file,  data, spiffs, , 0x6000
storage,  data, nvs,     ,         0x10000
```

Raghav3107, 2021-03-26:
Hello @ravibhagavandas 
Sorry for the late reply.

I tried to do as per your suggestion. I set the partition offset to 0x8000 and modified the partition table CSV file as you have shown in your example. Now I did not the partition overloading error. It builds perfectly but after flash, it showing a flash read error.
Also, I tried after changing partition offset to 0x10000, but the same error is showing.

Error Log:
`rst:0x10 (RTCWDT_RTC_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
flash read err, 1000
ets_main.c 371 
ets Jun  8 2016 00:22:57

rst:0x10 (RTCWDT_RTC_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
flash read err, 1000
ets_main.c 371 
ets Jun  8 2016 00:22:57

rst:0x10 (RTCWDT_RTC_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
flash read err, 1000
ets_main.c 371 
ets Jun  8 2016 00:22:57
`

shubhamkulkarni97, 2021-04-01:
Hi @Raghav3107,

From the logs, it seems you have enabled flash encryption and bootloader is not encrypted.

Please try the steps as below:

* Run `./vendors/espressif/esp-idf/tools/idf.py menuconfig`
* Go-to Partition table and set `Offset of partition table` to 0xa000
* Use partition table mentioned in above [comment](https://github.com/aws/amazon-freertos/issues/3051#issuecomment-804458494)
* Enable flash encryption (in development mode) and secure boot, set Secure bootloader mode as `Reflashable`. Enable all potentially insecure options which helps while debugging.
* Save the configuration and exit menuconfig.
* Build the application by running `./vendors/espressif/esp-idf/tools/idf.py build`
* Burn secure boot key using the command printed during build
* Flash encrypted boatloader by running `./vendors/espressif/esp-idf/tools/idf.py encrpted-bootloader-flash`
* Flash encrypted application by running `./vendors/espressif/esp-idf/tools/idf.py encrypted-flash`

Raghav3107, 2021-04-02:
Hello @shubhamkulkarni97 

I m using esp-idf 3.3, I tried to follow your steps but when I m trying to build them as per your command it shows some cmake error.

Error:
```

Note: You are using Python 3.8.5. Python 3 support is new, please report any problems you encounter. Search for 'Setting the Python Interpreter' in the ESP-IDF docs if you want to use Python 2.7.
WARNING: IDF_PATH environment variable is set to /home/horsemann/Desktop/WorkSpace/esp/esp-idf but idf.py path indicates IDF directory /home/horsemann/Desktop/WorkSpace/SecureBootTest/vendors/espressif/esp-idf. Using the environment variable directory, but results may be unexpected...
Checking Python dependencies...
Python requirements from /home/horsemann/Desktop/WorkSpace/esp/esp-idf/requirements.txt are satisfied.
Running cmake in directory /home/horsemann/Desktop/WorkSpace/SecureBootTest/build
Executing "cmake -G Ninja -DPYTHON_DEPS_CHECKED=1 -DESP_PLATFORM=1 --warn-uninitialized /home/horsemann/Desktop/WorkSpace/SecureBootTest"...
Warn about uninitialized values.
-- The C compiler identification is GNU 9.3.0
-- The CXX compiler identification is GNU 9.3.0
-- Check for working C compiler: /usr/bin/cc
-- Check for working C compiler: /usr/bin/cc -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Check for working CXX compiler: /usr/bin/c++
-- Check for working CXX compiler: /usr/bin/c++ -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Found Git: /usr/bin/git (found version "2.25.1") 
CMake Warning (dev) at libraries/3rdparty/CMakeLists.txt:33 (if):
  uninitialized variable 'AFR_BOARD'
This warning is for project developers.  Use -Wno-dev to suppress it.

CMake Warning (dev) at libraries/3rdparty/CMakeLists.txt:100 (if):
  uninitialized variable 'AFR_BOARD'
This warning is for project developers.  Use -Wno-dev to suppress it.

CMake Error at CMakeLists.txt:34 (message):
  Board is not supported: vendor.board

-- Configuring incomplete, errors occurred!
See also "/home/horsemann/Desktop/WorkSpace/SecureBootTest/build/CMakeFiles/CMakeOutput.log".
cmake failed with exit code 1

```

So, I tried to `make` command to build and menuconfig `espsecure.py` to encrypt the data.
**Steps which I did:**
-  I build the using `make` 

- Log after Build:

```
App "aws_demos" version: 1
Python requirements from /home/horsemann/Desktop/WorkSpace/SecureBootTest/vendors/espressif/esp-idf/requirements.txt are satisfied.

(Secure boot enabled, so bootloader not flashed automatically. See 'make bootloader' output)
To flash app & partition table, run 'make flash' or:
python /home/horsemann/Desktop/WorkSpace/SecureBootTest/vendors/espressif/esp-idf/components/esptool_py/esptool/esptool.py --chip esp32 --port /dev/ttyUSB0 --baud 921600 --before default_reset --after hard_reset write_flash -z --flash_mode dio --flash_freq 80m --flash_size detect 0x17000 /home/horsemann/Desktop/WorkSpace/SecureBootTest/vendors/espressif/boards/esp32/aws_demos/build/ota_data_initial.bin 0x20000 /home/horsemann/Desktop/WorkSpace/SecureBootTest/vendors/espressif/boards/esp32/aws_demos/build/aws_demos.bin 0xa000 /home/horsemann/Desktop/WorkSpace/SecureBootTest/vendors/espressif/boards/esp32/aws_demos/build/partition-table.bin
```

- Build bootloader using `make bootloader` command

- Log after `make bootloader `command:

```
To reflash the bootloader after initial flash:
python /home/horsemann/Desktop/WorkSpace/SecureBootTest/vendors/espressif/esp-idf/components/esptool_py/esptool/esptool.py --chip esp32 --port /dev/ttyUSB0 --baud 921600 --before default_reset --after hard_reset write_flash -z --flash_mode dio --flash_freq 80m --flash_size detect 0x0 /home/horsemann/Desktop/WorkSpace/SecureBootTest/vendors/espressif/boards/esp32/aws_demos/build/bootloader/bootloader-reflash-digest.bin
```

- Encrypted the bootloader by command.
`espsecure.py encrypt_flash_data --keyfile ./my_flash_encryption_key.bin --address 0x0 --output ./build/bootloader/bootloader-reflash-digest_encrypted.bin ./build/bootloader/bootloader-reflash-digest.bin
`
- Encrypted the partition:
`espsecure.py encrypt_flash_data --keyfile ./my_flash_encryption_key.bin --address 0xa000 --output ./build/partition-table_encrypted.bin ./build/partition-table.bin`

- Encrypted the project:
`espsecure.py encrypt_flash_data --keyfile ./my_flash_encryption_key.bin --address 0x20000 --output ./build/aws_demos_encrypted.bin ./build/aws_demos.bin`

- At last, I flash but still, the same error is comming.
- Is there anything I m doing wrong.

shubhamkulkarni97, 2021-04-05:
@Raghav3107,

Steps mentioned in [above comment](https://github.com/aws/amazon-freertos/issues/3051#issuecomment-811658899) are for IDF v4.2

Please check following points:

* You have not flashed otadata partition, please encrypt `ota_data_initial.bin` and flash it at 0x17000
* Bootloader should be flashed at **0x1000** and not at 0x0
* Please check if `CONFIG_SECURE_BOOT_BUILD_SIGNED_BINARIES` is enabled in menuconfig
* Please verify that `CONFIG_PARTITION_TABLE_OFFSET` in menuconfig is set to 0xa000
* Also check if partition table has a overlap with app partitions (`0xa000 + partition table size` should be less than offset of first partition)

Raghav3107, 2021-04-06:
Hello @shubhamkulkarni97 
Thanks for the reply.

Can you explain to me the 5th point? What is the partition table size and how will I know partition table size?

And I tried flash encryption as per the remaining point. 
I encrypted bootloader.bin, ota_data_initial.bin, partition-table.bin, aws_demos_encrypted.bin.
```

- espsecure.py encrypt_flash_data --keyfile ./my_flash_encryption_key.bin --address 0x1000 --output ./build/bootloader/bootloader_encrypted.bin ./build/bootloader/bootloader.bin

- espsecure.py encrypt_flash_data --keyfile ./my_flash_encryption_key.bin --address 0x17000 --output ./build/ota_data_initial_encrypted.bin ./build/ota_data_initial.bin
- espsecure.py encrypt_flash_data --keyfile ./my_flash_encryption_key.bin --address 0x10000 --output ./build/partition-table_encrypted.bin ./build/partition-table.bin
- espsecure.py encrypt_flash_data --keyfile ./my_flash_encryption_key.bin --address 0x20000 --output ./build/aws_demos_encrypted.bin ./build/aws_demos.bin
```
 And flashed all this

```
- esptool.py --port /dev/ttyUSB0 --baud 115200 write_flash 0x1000 ./build/bootloader/bootloader_encrypted.bin
- esptool.py --port /dev/ttyUSB0 --baud 115200 write_flash 0x17000 ./build/ota_data_initial_encrypted.bin
- esptool.py --port /dev/ttyUSB0 --baud 115200 write_flash 0xa000 ./build/partition-table_encrypted.bin
- espsecure.py encrypt_flash_data --keyfile ./my_flash_encryption_key.bin --address 0x20000 --output ./build/aws_demos_encrypted.bin ./build/aws_demos.bin
```
After this, I got an error like this.

```
rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0018,len:4
load:0x3fff001c,len:10520
load:0x40078000,len:20220
load:0x40080400,len:6592
csum err:0xce!=0xff
ets_main.c 371 
ets Jun  8 2016 00:22:57
```
So, I search for a solution to the issue. The solution I found is to increase the partition offset.
So I set partition offset to 10000 and I encrypted and flash the bin the same as earlier except partition_table.bin.

I encrypted and flash the partition bin to `0x10000`

After this I got a secure boot check fail error.

ets Jun  8 2016 00:22:57

```
rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0018,len:4
load:0x3fff001c,len:10520
load:0x40078000,len:20216
load:0x40080400,len:6592
secure boot check fail
ets_main.c 371 
ets Jun  8 2016 00:22:57

rst:0x10 (RTCWDT_RTC_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0018,len:4
load:0x3fff001c,len:10520
load:0x40078000,len:20216
load:0x40080400,len:6592
secure boot check fail
ets_main.c 371 
ets Jun  8 2016 00:22:57
```
Can you tell me what wrong I m doing or if possible can you tell me the overall step of secure boot and flash encryption for esp-idf 3.3?

shubhamkulkarni97, 2021-04-12:
@Raghav3107, Sorry for late response.

> Can you explain to me the 5th point? What is the partition table size and how will I know partition table size?

After build is completed, you can check file size for `partition-table.bin` in build folder. `Partition table offset + size of partition-table.bin` should be less than offset for first app partition.

Steps you have followed look fine. Please note that you should use same flash encryption and secure boot keys that are written in the device.

Please check below pointers for secure boot and flash encryption documentation:
https://docs.espressif.com/projects/esp-idf/en/release-v3.3/security/secure-boot.html
https://docs.espressif.com/projects/esp-idf/en/release-v3.3/security/flash-encryption.html

Raghav3107, 2021-04-12:
Hello @shubhamkulkarni97 
Thanks for the replay,

I tried as per suggestion still I m getting the same "secure boot check fail" error.

Let me tell you the overall steps.
1. make menuconfig, 
[image]
[image]

2. make -j4
`python /home/horsemann/Desktop/WorkSpace/SecureBootTest/vendors/espressif/esp-idf/components/esptool_py/esptool/esptool.py --chip esp32 --port /dev/ttyUSB0 --baud 921600 --before default_reset --after hard_reset write_flash -z --flash_mode dio --flash_freq 80m --flash_size detect 0x1a000 /home/horsemann/Desktop/WorkSpace/SecureBootTest/vendors/espressif/boards/esp32/aws_demos/build/ota_data_initial.bin 0x20000 /home/horsemann/Desktop/WorkSpace/SecureBootTest/vendors/espressif/boards/esp32/aws_demos/build/aws_demos.bin 0x10000 /home/horsemann/Desktop/WorkSpace/SecureBootTest/vendors/espressif/boards/esp32/aws_demos/build/partition-table.bin`

3. make bootloader
```
*******************************************************************************
Bootloader built and secure digest generated. First time flash command is:
python /home/horsemann/Desktop/WorkSpace/SecureBootTest/vendors/espressif/esp-idf/components/esptool_py/esptool/espefuse.py burn_key secure_boot /home/horsemann/Desktop/WorkSpace/SecureBootTest/vendors/espressif/boards/esp32/aws_demos/build/bootloader/secure-bootloader-key-256.bin
python /home/horsemann/Desktop/WorkSpace/SecureBootTest/vendors/espressif/esp-idf/components/esptool_py/esptool/esptool.py --chip esp32 --port /dev/ttyUSB0 --baud 921600 --before default_reset --after hard_reset write_flash -z --flash_mode dio --flash_freq 80m --flash_size detect 0x1000 /home/horsemann/Desktop/WorkSpace/SecureBootTest/vendors/espressif/boards/esp32/aws_demos/build/bootloader/bootloader.bin
*******************************************************************************
To reflash the bootloader after initial flash:
python /home/horsemann/Desktop/WorkSpace/SecureBootTest/vendors/espressif/esp-idf/components/esptool_py/esptool/esptool.py --chip esp32 --port /dev/ttyUSB0 --baud 921600 --before default_reset --after hard_reset write_flash -z --flash_mode dio --flash_freq 80m --flash_size detect 0x0 /home/horsemann/Desktop/WorkSpace/SecureBootTest/vendors/espressif/boards/esp32/aws_demos/build/bootloader/bootloader-reflash-digest.bin
```
4. After this I reflash the two message.
5.  I signed the partition table and aws_demos bin file
```
espsecure.py sign_data --version 1 --keyfile ./my_secure_boot_signing_key.pem --output ./build/partition-table_signed.bin ./build/partition-table-unsigned.bin

spsecure.py sign_data --version 1 --keyfile ./my_secure_boot_signing_key.pem --output ./build/aws_demos_signed.bin ./build/aws_demos-unsigned.bin

```
6. I encrypted the signed partition and aws-demos, ota and bootloader files.
```
espsecure.py encrypt_flash_data --keyfile ./my_flash_encryption_key.bin --address 0x1a000 --output ./build/ota_data_initial_encrypted.bin ./build/ota_data_initial.bin

espsecure.py encrypt_flash_data --keyfile ./my_flash_encryption_key.bin --address 0x10000 --output ./build/partition-table_encrypted.bin ./build/partition-table_signed.bin

espsecure.py encrypt_flash_data --keyfile ./my_flash_encryption_key.bin --address 0x20000 --output ./build/aws_demos_encrypted.bin ./build/aws_demos_signed.bin

espsecure.py encrypt_flash_data --keyfile ./my_flash_encryption_key.bin --address 0x1000 --output ./build/bootloader/bootloader_encrypted.bin ./build/bootloader/bootloader.bin

```

7. flash these encrypted files
```
esptool.py --port /dev/ttyUSB0 --baud 115200 write_flash 0x1000 ./build/bootloader/bootloader_encrypted.bin

esptool.py --port /dev/ttyUSB0 --baud 115200 write_flash 0x10000 ./build/partition-table_encrypted.bin

esptool.py --port /dev/ttyUSB0 --baud 115200 write_flash 0x1a000 ./build/ota_data_initial_encrypted.bin

esptool.py --port /dev/ttyUSB0 --baud 115200 write_flash 0x20000 ./build/aws_demos_encrypted.bin

```
8. make monitor
Error Log:
```
rst:0x10 (RTCWDT_RTC_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0018,len:4
load:0x3fff001c,len:10520
load:0x40078000,len:20216
load:0x40080400,len:6592
secure boot check fail
ets_main.c 371 
ets Jun  8 2016 00:22:57

rst:0x10 (RTCWDT_RTC_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0018,len:4
load:0x3fff001c,len:10520
load:0x40078000,len:20216
load:0x40080400,len:6592
secure boot check fail
ets_main.c 371 
ets Jun  8 2016 00:22:57

```
I check the partition-table.bin files it is 3104 bytes and I set the offset to `10000. `
Previously my first app partition table was `11000`. So, I  increase this to 14000 so the `partition_table.bin file + partition table offset` is less than the first app partition offset.

**Here is the partition file**
[partition-table.csv](https://github.com/aws/amazon-freertos/files/6294987/partition-table.csv)

Also, I used the same key which I used in a secure boot.

Still, I m getting the secure boot check fail error.

**Some doubt which I want clear.**

1. Does every time we need to make a bootloader?
2.  Does every time we need to reflash the bootloader?

Raghav3107, 2021-05-14:
thanks, @shubhamkulkarni97 @ravibhagavandas for your help.

Got some suggestion on esp-forum from ESP_Angus, to flash only `bootloader-reflash-digest.bin at 0x0` [[esp-forum-issues-link](https://esp32.com/viewtopic.php?p=76375&sid=b2b56e9a7d5799c55a5fd69db062788a#p76375)] and test it. I tested it and it worked.

**Steps:**

- `Partition table offset + size of partition-table.bin` should be less than offset for the first app partition.
- Encrypt and flash `bootloader-reflash-digest.bin at 0x0.`
- Encrypt and flash the `partition table`, `OTA`, `aws-demos` as their respective address.

closed 2021-05-14 by Raghav3107

ChrisEAlfred, 2022-08-27:
Has anyone written a complete step by step - it is almost done above, but at the last moment the solution steps are vague:

Steps:

Partition table offset + size of partition-table.bin should be less than offset for the first app partition.
Encrypt and flash bootloader-reflash-digest.bin at 0x0.
Encrypt and flash the partition table, OTA, aws-demos as their respective address.

horsemann07, 2022-08-27:
One thing you need to add in you step:

1. Partition table offset + size of partition-table.bin should be less than offset for the first app partition.

2. Encrypt and flash bootloader-reflash-digest.bin at 0x0.

3. Sign the partition and app bin files.
`espsecure.py sign_data --version 1 --keyfile ./my_secure_boot_signing_key.pem --output ./build/partition-table_signed.bin ./build/partition-table-unsigned.bin`

4. Encrypt and flash the partition table, OTA, aws-demos as their respective address. ( Partition and app file will be signed bin file ).

ABHITHLALC, 2022-09-01:
> **Some doubt which I want clear.**
> 
> 1. Does every time we need to make a bootloader?
> 2. Does every time we need to reflash the bootloader?

@horsemann07 
Are your doubts clear now?
If you have the answers for above questions please share that too. Because I have the same doubts.

horsemann07, 2022-09-01:
Hello, @ABHITHLALC 

Actually, they did not given the clear answer on the above questions.

1. But as per my experience, we no need to flash the bootloader everytime untill we change the partition-table which may affect the offset of bootloader and other partition.
2. And first app partition means the first partition of the partition table. (Eg. "nvs" partition in partition table which is first partition in partition-table).

Hope your doubts will be clear now.
Thanks!

ABHITHLALC, 2022-09-01:
Thanks for replay, my partition table offset is set to 0x10000, and partition-table.bin size is 3140 bytes.
after patition build
```
# ESP-IDF Partition Table
# Name, Type, SubType, Offset, Size, Flags
nvs,data,nvs,0x11000,24K,
phy_init,data,phy,0x17000,4K,
factory,app,factory,0x20000,1M,
```
so in this case I should chance the 1 st app partition ?

mahavirj, 2022-09-12:
@ABHITHLALC 

If you are using [re-flashable software bootloader](https://docs.espressif.com/projects/esp-idf/en/release-v4.2/esp32/security/secure-boot-v1.html#re-flashable-software-bootloader) then it is possible to update bootloader. It is not mandatory to flash bootloader everytime, unless you have some changes in it.

> so in this case I should change the 1 st app partition ? If so how can I

Your partition table looks fine. There is no need for any change.

mahavirj, 2022-09-12:
@ABHITHLALC 

>  If you don't mind, can you check if there are any changes I need to make before proceeding with this.

This looks good to me.

ABHITHLALC, 2022-09-13:
Thank you, @mahavirj