aws/amazon-freertos #3052: /freertos/vendors/espressif/esp-idf/components/driver/include/driver/spi_common.h:21:10: fatal error: soc/spi_periph.h: No such file or directory  #include "soc/spi_periph.h"

hackaprende, 2021-03-17:
**Describe the bug**
Not finding soc/spi_periph.h

**System information**
- Which hardware board or part numbers? MacOS Big Sur
- IDE used: Visual Studio Code
- Operating System [MacOS]
- Version of FreeRTOS (run ``git describe --tags`` to find it)
- Project Custom Application
- If your project is a Custom Application, please add the relevant code snippet in the section `Code to reproduce the bug`.
I updated freertos submodule and this started happending when flashing my evice 

**Expected behavior**
It should flash my device with idf.py flash

**To reproduce**
Steps to reproduce the behavior:

1.Update Freertos to latest version
2. Build and flash the code as before

**Code to reproduce the bug**
The code should be wrapped in the cpp tag in order to be displayed clearly. For example:

```cpp
printf("Hello World")
```

If the code is longer than 30 lines, [GIST](https://gist.github.com) is preferred.

**Additional context**
I fixed the problem by adding     "${esp_idf_dir}/components/soc/soc/include" to my ESP32 CMakeLists.txt, like this, check line 94, it seems you forgot to add that include:

[image]

shubhamkulkarni97, 2021-03-18:
Hi @hackaprende,

From the error log it seems you are using AFR in external CMake project.
When using latest master ESP-IDF is upgraded to v4.2. There are quiet a few build system changes in IDF v4.2 as compared to v3.3.

I have pushed a [branch](https://github.com/shubhamkulkarni97/amazon-freertos-examples/tree/feature/idf_v4.2) in amazon-freertos-examples repo to demonstrate use of AFR as external CMake project with ESP-IDF v4.2. Please adapt CMake changes according to specified branch.

Hope this helps!

mahavirj, 2021-03-18:
@tgsong @lundinc2 @qiutongs 

Can we consider integrating https://github.com/tgsong/amazon-freertos-examples into mainstream AFR repository? I think it would help lot of users, rather than using custom forks, branches etc.

Thanks.

hackaprende, 2021-03-18:
Thanks @shubhamkulkarni97 I edited my CMakeLists.txt and that error is gone, but I keep having another one:

CMake Error at freertos/vendors/espressif/boards/esp32/CMakeLists.txt:454 (add_executable):
  add_executable cannot create target "my_target" because another target with the
  same name already exists.  The existing target is an executable created in
  source directory "path/to/my_target".  See
  documentation for policy CMP0002 for more details.

To solve this I had to comment this:

[image]

And this line:
[image]

In order to ignore the Default esp32 target

ravibhagavandas, 2021-03-19:
@hackaprende 
I pulled the repository [branch](https://github.com/shubhamkulkarni97/amazon-freertos-examples/tree/feature/idf_v4.2) and it build successfully with IDF 4.2. Did you make sure the executable name defined in your project CMakeFile is not `aws_demos.bin` ?

hackaprende, 2021-03-19:
@ravibhagavandas
I copied it exactly as in that branch and it worked without having to comment the esp32 CMakeLists.txt as above, but I still have to add the line:
    "${esp_idf_dir}/components/soc/soc/include"

In order to make it work.

Also it does not detect my custom component, it was working well before:
```
get_filename_component(
    EXTRA_COMPONENT_DIRS
    "components/adc_hx711" ABSOLUTE
)
idf_build_component(${EXTRA_COMPONENT_DIRS})
```

But it seems the other problems I had in this other issue are solved with this, thanks!
https://github.com/aws/amazon-freertos/issues/3054#issuecomment-802549172

shubhamkulkarni97, 2021-03-19:
@hackaprende From error it looks like you have called `add_executable` from top level CMakeLists.txt. Instead you need to set IDF_PROJECT_EXECUTABLE to project name and IDF_EXECUTABLE_SRCS to required source files.

Please check implementation here: https://github.com/shubhamkulkarni97/amazon-freertos-examples/blob/75c7e2aaba9ffb29afb386c7e273836c896cd3a2/CMakeLists.txt#L16

ravibhagavandas, 2021-03-22:
Hi,

I think your issue with adding custom component is fixed by adding `idf::adc_hx711` to the `target_link_libraries()`.

As with   directory path is already included into the `IDF_PROJECT_EXECUTABLE` in the [CMakeFile](https://github.com/aws/amazon-freertos/blob/master/vendors/espressif/boards/esp32/CMakeLists.txt#L460):
```
target_include_directories(
    ${IDF_PROJECT_EXECUTABLE}
    PUBLIC
        "${esp_idf_dir}/components/esp_ringbuf/include"
        $<TARGET_PROPERTY:AFR::kernel,INTERFACE_INCLUDE_DIRECTORIES>
        $<TARGET_PROPERTY:AFR::ble_hal::mcu_port,INTERFACE_INCLUDE_DIRECTORIES>
        $<TARGET_PROPERTY:AFR::wifi::mcu_port,INTERFACE_INCLUDE_DIRECTORIES>
)
```

Did you make sure those lines were not commented out.

hackaprende, 2021-03-23:
I feel so dumb right now, the problem was I had not added the 
    idf::adc_hx711

part at the bottom of the CMakeLists.txt

closed 2021-03-23 by hackaprende