aws/amazon-freertos #3053: /freertos/libraries/freertos_plus/standard/freertos_plus_tcp/portable/NetworkInterface/esp32/NetworkInterface.c:36:10: fatal error: esp_wifi_internal.h: No such file or directory  #include "esp_wifi_internal.h"

hackaprende, 2021-03-17:
**Describe the bug**
When I run try to flash the code with idf.py flash I am getting this error:
/freertos/libraries/freertos_plus/standard/freertos_plus_tcp/portable/NetworkInterface/esp32/NetworkInterface.c:36:10: fatal error: esp_wifi_internal.h: No such file or directory
 #include "esp_wifi_internal.h"

This started happening after today I updated the freertos submoule, it was not happening before, I cannot find any file called esp_wifi_internal.h.

**System information**
- Which hardware board or part numbers ESP32-DevkitC V4
- IDE used: Visual Studio code
- Operating System MacOS
- Version of FreeRTOS (run ``git describe --tags`` to find it)
- Project Custom Application
- If your project is a Custom Application, please add the relevant code snippet in the section `Code to reproduce the bug`. 
This happens inside NetworkInterface.c file

**Expected behavior**
It should flash the device

**Screenshots or console output**
```
FAILED: freertos/CMakeFiles/afr_wifi.dir/libraries/freertos_plus/standard/freertos_plus_tcp/portable/NetworkInterface/esp32/NetworkInterface.c.obj 
/Users/Almaral/esp/xtensa-esp32-elf/bin/xtensa-esp32-elf-gcc -DESP_PLATFORM -DIDF_VER=\"v4.2-12-g3388b409c-dirty\" -D_GNU_SOURCE -I../freertos-configs -I../freertos/libraries/abstractions/wifi/include -I../freertos/libraries/c_sdk/standard/common/include -I../freertos/libraries/freertos_plus/standard/freertos_plus_tcp/include -I../freertos/freertos_kernel/include -I../freertos/vendors/espressif/esp-idf/components/app_update/include -I../freertos/vendors/espressif/esp-idf/components/bootloader_support/include -I../freertos/vendors/espressif/esp-idf/components/esp32/include -I../freertos/vendors/espressif/esp-idf/components/esp_event/include -I../freertos/vendors/espressif/esp-idf/components/efuse/include -I../freertos/vendors/espressif/esp-idf/components/driver/include -I../freertos/vendors/espressif/esp-idf/components/heap/include -I../freertos/vendors/espressif/esp-idf/components/log/include -I../freertos/vendors/espressif/esp-idf/components/nvs_flash/include -I../freertos/vendors/espressif/esp-idf/components/pthread/include -I../freertos/vendors/espressif/esp-idf/components/newlib/platform_include -I../freertos/vendors/espressif/esp-idf/components/soc/src/esp32/include -I../freertos/vendors/espressif/esp-idf/components/soc/soc/esp32/include -I../freertos/vendors/espressif/esp-idf/components/soc/soc/include -I../freertos/vendors/espressif/esp-idf/components/soc/include -I../freertos/vendors/espressif/esp-idf/components/spi_flash/include -I../freertos/vendors/espressif/esp-idf/components/vfs/include -I../freertos/vendors/espressif/esp-idf/components/esp_ringbuf/include -I../freertos/vendors/espressif/boards/esp32/../components/freertos/include -I../freertos/vendors/espressif/esp-idf/components/mbedtls/port/include -I../freertos/vendors/espressif/esp-idf/components/xtensa/esp32/include/xtensa -I../freertos/vendors/espressif/esp-idf/components/xtensa/esp32/include -I../freertos/vendors/espressif/esp-idf/components/xtensa/include -I../freertos/vendors/espressif/esp-idf/components/esp_rom/include -I../freertos/vendors/espressif/esp-idf/components/esp_timer/include -I../freertos/vendors/espressif/esp-idf/components/esp_common/include -I../freertos/vendors/espressif/esp-idf/components/esp_system/include -I../freertos/vendors/espressif/boards/esp32/../components/freertos_tcpip/ethernet/include -I../freertos/vendors/espressif/boards/esp32/../components/freertos_tcpip/smartconfig_ack/include -I../freertos/vendors/espressif/boards/esp32/../components/freertos_tcpip/tcpip_adapter/include -I../freertos/libraries/freertos_plus/standard/freertos_plus_tcp/portable/Compiler/GCC -I../freertos/vendors/espressif/esp-idf/components/xtensa -I../freertos/vendors/espressif/esp-idf/components/esp_wifi/include -I../freertos/vendors/espressif/esp-idf/components/esp_netif/include -I../freertos/vendors/espressif/esp-idf/components/esp_eth/include -I../freertos/vendors/espressif/boards/esp32/aws_demos/config_files/default_pkcs11_config -I../freertos/freertos_kernel/portable/ThirdParty/GCC/Xtensa_ESP32/include -I../freertos/demos/include -I../freertos/vendors/espressif/boards/esp32/aws_demos/config_files -Iconfig -mlongcalls -std=gnu99 -Wno-old-style-declaration -Wno-stringop-overflow -Wno-bool-operation -Wno-sizeof-pointer-div -Wno-nonnull -ffunction-sections -fdata-sections -fstrict-volatile-bitfields -Wall -Werror=all -Wno-error=unused-function -Wno-error=unused-but-set-variable -Wno-error=unused-variable -Wno-error=deprecated-declarations -Wextra -Wno-unused-parameter -Wno-sign-compare -ggdb -Og -Wno-parentheses -Wno-sizeof-pointer-memaccess -Wno-clobbered -Wno-format-overflow -Wno-stringop-truncation -Wno-misleading-indentation -Wno-cast-function-type -Wno-implicit-fallthrough -Wno-unused-const-variable -Wno-switch-unreachable -Wno-format-truncation -Wno-memset-elt-size -Wno-int-in-bool-context -MD -MT freertos/CMakeFiles/afr_wifi.dir/libraries/freertos_plus/standard/freertos_plus_tcp/portable/NetworkInterface/esp32/NetworkInterface.c.obj -MF freertos/CMakeFiles/afr_wifi.dir/libraries/freertos_plus/standard/freertos_plus_tcp/portable/NetworkInterface/esp32/NetworkInterface.c.obj.d -o freertos/CMakeFiles/afr_wifi.dir/libraries/freertos_plus/standard/freertos_plus_tcp/portable/NetworkInterface/esp32/NetworkInterface.c.obj -c ../freertos/libraries/freertos_plus/standard/freertos_plus_tcp/portable/NetworkInterface/esp32/NetworkInterface.c
../freertos/libraries/freertos_plus/standard/freertos_plus_tcp/portable/NetworkInterface/esp32/NetworkInterface.c:36:10: fatal error: esp_wifi_internal.h: No such file or directory
 #include "esp_wifi_internal.h"
```

**To reproduce**
Steps to reproduce the behavior:

1. Upgrade freertos submodule.
2. Add wifi capabilities
3. Flash with idf.py

**Code to reproduce the bug**
The code should be wrapped in the cpp tag in order to be displayed clearly. For example:

```cpp
printf("Hello World")
```

If the code is longer than 30 lines, [GIST](https://gist.github.com) is preferred.

**Additional context**
As I said before, it happened after updating freertos submodule, I cannot find esp_wifi_internal.h file anywhere.
Thank you!

hackaprende, 2021-03-17:
I fixed it:

Instead of using 
#include esp_internal_wifi.h

I used
#include "esp_private/wifi.h"

ravibhagavandas, 2021-03-18:
Thanks for bringing this to our attention.
We are working on fixing this.

ravibhagavandas, 2021-03-22:
Closing this issue as its a duplicate of https://github.com/FreeRTOS/FreeRTOS-Plus-TCP/issues/212.

closed 2021-03-22 by ravibhagavandas