aws/amazon-freertos #3054: Lots of classes do not find included files or directories

hackaprende, 2021-03-18:
**Describe the bug**
After updating freertos submodule, a lot of classes give errors finding included files.

**System information**
- Which hardware board or part numbers? ESP32-DevkitC v4
- IDE used: Visual Studio Code
- Operating System MacOS
- Version of FreeRTOS (run ``git describe --tags`` to find it)
- Project Custom Application
- If your project is a Custom Application, please add the relevant code snippet in the section `Code to reproduce the bug`. 

**Expected behavior**
All included files in amazon-freertos should be found.

**Screenshots or console output**
If appropriate, please cut-and-paste the console output to explain the issue. If possible, include the call stack. 
This is an example, I am adding tcpip_adapter.h in my Main, which has #include "tcpip_adapter_types.h", and inside it has #include "lwip/ip_addr.h", if I do cmd + click it goes to ip_addr.h file, but when I do idy.py I get this error:

```
In file included from ../src/../freertos/vendors/espressif/esp-idf/components/tcpip_adapter/include/tcpip_adapter.h:24,
                 from ../src/main.c:24:
../src/../freertos/vendors/espressif/esp-idf/components/tcpip_adapter/include/tcpip_adapter_types.h:18:10: fatal error: lwip/ip_addr.h: No such file or directory
 #include "lwip/ip_addr.h"
          ^~~~~~~~~~~~~~~~
compilation terminated.
[12/823] Building C object esp-idf/wpa_supplicant/CMakeFiles/__idf_wpa_supplicant.dir/src/crypto/rc4.c.obj
ninja: build stopped: subcommand failed.
```

This happens with A LOT of other files around all amazon-freertos. Is there a way to fix this?

**To reproduce**
Steps to reproduce the behavior:

1. Setup a project with Freertos's latest version
2. Build and flash your code

**Code to reproduce the bug**
The code should be wrapped in the cpp tag in order to be displayed clearly. For example:

```cpp
printf("Hello World")
```

If the code is longer than 30 lines, [GIST](https://gist.github.com) is preferred.

**Additional context**
This was working well in the previous version.

Thank you!

hackaprende, 2021-03-18:
Another example is here, I do not even use this class touch_pad.h but It´s giving me this error:

```
In file included from ../freertos/vendors/espressif/esp-idf/components/esp32/include/esp_sleep.h:20,
                 from ../src/main.c:50:
../freertos/vendors/espressif/esp-idf/components/driver/include/driver/touch_pad.h:17:10: fatal error: driver/touch_sensor.h: No such file or directory
 #include "driver/touch_sensor.h"
```

[image]

hackaprende, 2021-03-19:
I used the branch you told me here #3052 and it seems it´s working, only my custom components are not working but I almost have it thanks for your answer.

shubhamkulkarni97, 2021-03-19:
@hackaprende Build system APIs for adding components have also changed.
Please check API to add a component: https://github.com/shubhamkulkarni97/amazon-freertos-examples/blob/75c7e2aaba9ffb29afb386c7e273836c896cd3a2/components/foo/CMakeLists.txt#L4

hackaprende, 2021-03-19:
I did that but no results, still my component is not found in my main:

[image]

[image]

[image]

hackaprende, 2021-03-19:
I solved this by adding the whole path to my component in main:

`#include "../components/adc_hx711/include/adc_hx711.h"
`

hackaprende, 2021-03-21:
Oh I feel bad now, the error was that in the bottom of the file I had to add idf::adc_hx711

target_link_libraries(
    my-project
    PRIVATE
    AFR::demo_core_mqtt
    ...
    idf::adc_hx711
)

Sorry for the trouble

closed 2021-03-21 by hackaprende