aws/amazon-freertos #3056: Literature for integrating ota, mqtt and wifi_provisioning over ble demos AWS FreeRTOS ESP32

Yoimer, 2021-03-18:
Hello there. Is there any documentation where I could check in order to integrate
demos from here https://github.com/aws/amazon-freertos/releases/tag/202012.00?

I want to integrate:
https://github.com/Yoimer/aws_freertos_esp32_components/blob/main/demos/coreMQTT/mqtt_demo_mutual_auth.c
https://github.com/Yoimer/aws_freertos_esp32_components/blob/main/demos/ota/aws_iot_ota_update_demo.c
https://github.com/Yoimer/aws_freertos_esp32_components/blob/main/demos/wifi_provisioning/aws_wifi_connect_task.c

What's in this code is not that clear to me https://github.com/Yoimer/aws_freertos_esp32_components/blob/main/demos/include/iot_demo_runner.h, in order to add another tasks as they mention.

Thanks in advance.

ravibhagavandas, 2021-03-18:
Hi @Yoimer

You could use one of the demo entry function  `vRunOTAUpdateDemo()` or `RunCoreMqttMutualAuthDemo()` and modify it to spawn a new task and run the other demo entry function in it. 
However it should be noted that the OTA update demo uses old MQTT library APIs and it works through a SHIM layer implementation. This means the network interface and connection type used is different between demos and could not be shared. The combined demo will create two separate MQTT connections to the cloud, so the client identifier used should be different for both connections otherwise AWS IoT will reject the older connection with duplicate client identifier.
For provisioning WiFi credentials using BLE you can enable the demo and follow the instructions to provision a WiFi network from app as shown in doc here: https://docs.aws.amazon.com/freertos/latest/userguide/ble-demo.html#ble-demo-wifi
The `aws_wifi_connect_task.c` will take care of reconnecting to a provisioned network when device is rebooted or current wifi connection is lost.

Please let us know if you have more questions.

Yoimer, 2021-03-18:
@ravibhagavandas Dear, thanks a lot for replying and assisting me. I will put in practice this valuable information. Please keep open this issue. I will update over here by tomorrow evening.
Have a wonderful day.

Yoimer, 2021-03-24:
@ravibhagavandas I truly hope you're having a great day. I found an ota + mqtt fully functional demo for AWS FreeRTOS ESP32 for version `V202002.00` which you kindly you can check here https://github.com/Yoimer/AFR_ESP32_FreeRTOS_V202002.00/blob/master/demos/ota/V202002.00_aws_iot_ota_update_demo_and_mqtt.c.

My goal is now integrate the `wifi_provisioning` example on this `ota + mqtt` integration. Specifically this https://github.com/Yoimer/AFR_ESP32_FreeRTOS_V202002.00/blob/master/demos/wifi_provisioning/aws_wifi_connect_task.c

I had to hard code my wifi credentials here https://github.com/Yoimer/AFR_ESP32_FreeRTOS_V202002.00/blob/master/demos/include/aws_clientcredential.h in order to test the mqtt + ota. I want to use ble over wifi demo to accomplish that goal.

Would you mind guiding me on this, please?

Thanks in advace.

ncohen17, 2021-03-26:
@Yoimer Did you also add the shadow example or are you only looking to use the MQTT ?

ravibhagavandas, 2021-03-26:
Hi @Yoimer

I am looking at your code and it seems you have all the pieces of code in your repo. The wifi connect task as given in demo is initialized through network manager component, when you enable WiFi network, code [here](https://github.com/Yoimer/AFR_ESP32_FreeRTOS_V202002.00/blob/master/demos/network_manager/aws_iot_network_manager.c#L485).  For the task to initialize,  `IOT_BLE_ENABLE_WIFI_PROVISIONING` flag should be enabled as in the comment in issue [#3048 ](https://github.com/aws/amazon-freertos/issues/3048#issuecomment-802282528). 
The demo will wait [here](https://github.com/Yoimer/AFR_ESP32_FreeRTOS_V202002.00/blob/master/demos/demo_runner/iot_demo_freertos.c#L255) for a  WiFi credential to be provisioned and connected before running the entry function.

Yoimer, 2021-03-26:
> @Yoimer Did you also add the shadow example or are you only looking to use the MQTT ?

Hello @ncohen17 I did not since I do not need it. In case you do, please download it from here https://github.com/aws/amazon-freertos/releases/download/202002.00/amazon-freertos.tar.gz. That tar has all the demos available for the `Version 202002.00` including all the vendors.

Yoimer, 2021-03-26:
> Hi @Yoimer
> 
> I am looking at your code and it seems you have all the pieces of code in your repo. The wifi connect task as given in demo is initialized through network manager component, when you enable WiFi network, code [here](https://github.com/Yoimer/AFR_ESP32_FreeRTOS_V202002.00/blob/master/demos/network_manager/aws_iot_network_manager.c#L485). For the task to initialize, `IOT_BLE_ENABLE_WIFI_PROVISIONING` flag should be enabled as in the comment in issue [#3048 ](https://github.com/aws/amazon-freertos/issues/3048#issuecomment-802282528).
> The demo will wait [here](https://github.com/Yoimer/AFR_ESP32_FreeRTOS_V202002.00/blob/master/demos/demo_runner/iot_demo_freertos.c#L255) until a WiFi credential is provisioned through a smartphone app and device gets connected to it, before running the entry function. When the flag is enabled you don't need to configure any static wifi credentials in https://github.com/Yoimer/AFR_ESP32_FreeRTOS_V202002.00/blob/master/demos/include/aws_clientcredential.h.

@ravibhagavandas Thanks a lot for your support. That's very kind from you. I will check this and update over here later. Have a great day.

ncohen17, 2021-03-26:
@Yoimer I am looking to make a similar firmware, are you available as a freelancer for my project?

Yoimer, 2021-03-27:
@ravibhagavandas Thanks a lot for your support. I managed to activate the `wifi_provisioning` on `V202002.00` as you explained. The rest of the code is waiting for the ssid and password to continue.
Now we will test the Android app based on this Workshop. https://teuteuguy.github.io/amazon-freertos-workshop-docs/labs/04-wifi-provisioning/. (Later we will create the app using the Amplify and Cognito services as mentioned on docs)

I will still, making efforts to accomplish the same goal on `V202012.00` as you explained here.
 https://github.com/aws/amazon-freertos/issues/3056#issuecomment-802265486

I will keep you posted here. Amazing piece of software. I am enjoying a lot from what you've been teaching over here.

Have a great weekend.

Yoimer, 2021-04-06:
> Hi @Yoimer
>
> You could use one of the demo entry function `vRunOTAUpdateDemo()` or `RunCoreMqttMutualAuthDemo()` and modify it to spawn a new task and run the other demo entry function in it.
> However [...]

@ravibhagavandas Hello. I've come across with an idea. What if I adapt the  `mqtt`  library present in [mqtt_demo_mutual_auth.c](https://github.com/Yoimer/aws_freertos_esp32_components/blob/main/demos/coreMQTT/mqtt_demo_mutual_auth.c) for the [aws_iot_ota_update_demo.c](https://github.com/Yoimer/aws_freertos_esp32_components/blob/main/demos/ota/aws_iot_ota_update_demo.c)?
Since the `ota` demo implements and mqtt version older than the current one running on [mqtt_demo_mutual_auth.c](https://github.com/Yoimer/aws_freertos_esp32_components/blob/main/demos/coreMQTT/mqtt_demo_mutual_auth.c), would make sense to add that mqtt libraries and make it compatible for both of them?

Would that way let me integrate ota + coreMQTT demos?

Looking forward for your comments. Thanks in advance.

ravibhagavandas, 2021-04-12:
Hi @Yoimer ,

To update the old OTA demo with the new coreMQTT library, it also requires changes in ota library specifically, you need to change the interface file [aws_iot_ota_mqtt.c](https://github.com/aws/amazon-freertos/blob/master/libraries/freertos_plus/aws/ota/src/mqtt/aws_iot_ota_mqtt.c) to adapt to new coreMQTT APIs. Also, you need to guarantee thread safety of the shared MQTT connection by implementing concurrency in your application.

We have a [labs repository](https://github.com/FreeRTOS/Lab-Project-coreMQTT-Agent) which uses a MQTT agent  that handles thread safety of a shared MQTT connection between different demo application. The file [ota_over_mqtt_demo.c](https://github.com/FreeRTOS/Lab-Project-coreMQTT-Agent/blob/main/source/ota_over_mqtt_demo.c) shows how the OTA demo task shares the MQTT connection with other demo tasks. More details can be found in documentation [here](https://freertos.org/ota/ota-mqtt-agent-demo.html).

We are working on upgrading this repository to use the new modules, but this gives a reference implementation  to get started.

Yoimer, 2021-04-12:
> Hi @Yoimer ,
>
> To update the old OTA demo with the new coreMQTT library, it also requires changes in ota library specifically, you need to change the interface file [...]

@ravibhagavandas Thanks a lot for replying and taking time to consider my idea. I really appreciate your support and the helpful information you're providing here. I am reading it already. Thanks again.

pvyawaha, 2021-05-17:
Hello , 

Do you still need help on this topic or this issue can be closed?

Yoimer, 2021-05-17:
> Hello ,
> 
> Do you still need help on this topic or this issue can be closed?

@pvyawaha Thanks a lot for keeping in touch. Still, we are doing QA tests. When we finish them all, I will close this issue.

Have a wonderful day.

muneebahmed10, 2021-07-21:
Hi @Yoimer,

>@pvyawaha Thanks a lot for keeping in touch. Still, we are doing QA tests. When we finish them all, I will close this issue.

Do you have an update on this? Can we close this issue?

Yoimer, 2021-07-21:
> Hi @Yoimer,
> 
> > @pvyawaha Thanks a lot for keeping in touch. Still, we are doing QA tests. When we finish them all, I will close this issue.
> 
> Do you have an update on this? Can we close this issue?

@ravibhagavandas Please, close it for now. Thanks

closed 2021-07-21 by muneebahmed10