aws/amazon-freertos #3057: mbedtls functions not linked correctly

SimionMogildea, 2021-03-19:
**Describe the bug**
I'm using the ESP32 OTA example in a standalone project but it seems to crash every time duning the linking phase (it probably won't work as expected because compiling is the goal for this issue). Being more specific, the `mbedtls_net_*` functions seems not to be linked correctly giving `undefined reference to 'mbedtls_net_*'` error.
It seems to be a Cmake problem in the library or maybe even in my example.

**System information**
- Which hardware board or part numbers? ESP32-DevkitC v4
- IDE used: VSCode
- Operating System: Windows
- Version of FreeRTOS: `202012.00-294-ga64c8555f`
- Project: OTA

**Expected behavior**
All functions (ex. `mbedtls_net_*`) shoud be linked correctly.

**Screenshots or console output**
```
C:\Users\user1\workspace\AmazonFreertosOta>python ./amazon-freertos/vendors/espressif/esp-idf/tools/idf.py erase_flash flash -p COM10 -b 115200 monitor
Setting IDF_PATH environment variable: c:\users\user1\workspace\amazonfreertosota\amazon-freertos\vendors\espressif\esp-idf
Executing action: erase_flash
Running esptool.py in directory c:\users\user1\workspace\amazonfreertosota\build
Executing "C:\Users\user1\.espressif\python_env\idf4.2_py3.7_env\Scripts\python.exe c:\users\user1\workspace\amazonfreertosota\amazon-freertos\vendors\espressif\esp-idf\components/esptool_py/esptool/esptool.py -p COM10 -b 115200 --before default_reset --after hard_reset --chip esp32 erase_flash"...
esptool.py v3.0
Serial port COM10
Connecting.....
Chip is ESP32-D0WD-V3 (revision 3)
Features: WiFi, BT, Dual Core, 240MHz, VRef calibration in efuse, Coding Scheme None
Crystal is 40MHz
MAC: 24:0a:c4:e7:fc:28
Uploading stub...
Running stub...
Stub running...
Erasing flash (this may take a while)...
Chip erase completed successfully in 16.4s
Hard resetting via RTS pin...
Executing action: flash
Running ninja in directory c:\users\user1\workspace\amazonfreertosota\build
Executing "ninja flash"...
[0/2] Re-checking globbed directories...
[1/7] Performing build step for 'bootloader'
ninja: no work to do.
[3/5] Linking CXX executable my_beautiful_app
FAILED: my_beautiful_app
cmd.exe /C "cd . && C:\Users\user1\.espressif\tools\xtensa-esp32-elf\esp-2020r3-8.4.0\xtensa-esp32-elf\bin\xtensa-esp32-elf-g++.exe  -mlongcalls  -Wl,--gc-sections -Wl,--cref -Wl,--Map=aws_demos.map -Wl,--undefined=uxTopUsedPriority @CMakeFiles\my_beautiful_app.rsp  -o my_beautiful_app  && cd ."
c:/users/user1/.espressif/tools/xtensa-esp32-elf/esp-2020r3-8.4.0/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld.exe: esp-idf/esp-tls/libesp-tls.a(esp_tls.c.obj):(.literal.esp_tls_low_level_conn+0x24): undefined reference to `mbedtls_net_init'
c:/users/user1/.espressif/tools/xtensa-esp32-elf/esp-2020r3-8.4.0/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld.exe: esp-idf/esp-tls/libesp-tls.a(esp_tls.c.obj): in function `esp_tls_low_level_conn':
C:\Users\user1\workspace\AmazonFreertosOta\build/../amazon-freertos/vendors/espressif/esp-idf/components/esp-tls/esp_tls.c:247: undefined reference to `mbedtls_net_init'
c:/users/user1/.espressif/tools/xtensa-esp32-elf/esp-2020r3-8.4.0/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld.exe: esp-idf/esp-tls/libesp-tls.a(esp_tls_mbedtls.c.obj):(.literal.esp_mbedtls_conn_delete+0x0): undefined reference to `mbedtls_net_free'
c:/users/user1/.espressif/tools/xtensa-esp32-elf/esp-2020r3-8.4.0/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld.exe: esp-idf/esp-tls/libesp-tls.a(esp_tls_mbedtls.c.obj):(.literal.esp_create_mbedtls_handle+0x24): undefined reference to `mbedtls_net_recv'
c:/users/user1/.espressif/tools/xtensa-esp32-elf/esp-2020r3-8.4.0/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld.exe: esp-idf/esp-tls/libesp-tls.a(esp_tls_mbedtls.c.obj):(.literal.esp_create_mbedtls_handle+0x28): undefined reference to `mbedtls_net_send'
c:/users/user1/.espressif/tools/xtensa-esp32-elf/esp-2020r3-8.4.0/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld.exe: esp-idf/esp-tls/libesp-tls.a(esp_tls_mbedtls.c.obj): in function `esp_mbedtls_conn_delete':
C:\Users\user1\workspace\AmazonFreertosOta\build/../amazon-freertos/vendors/espressif/esp-idf/components/esp-tls/esp_tls_mbedtls.c:198: undefined reference to `mbedtls_net_free'
collect2.exe: error: ld returned 1 exit status
ninja: build stopped: subcommand failed.
ninja failed with exit code 1
```

Those functions are linked as shown:
```
mbedtls_net_free                                  esp-idf/esp-tls/libesp-tls.a(esp_tls_mbedtls.c.obj)
mbedtls_net_init                                   esp-idf/esp-tls/libesp-tls.a(esp_tls.c.obj)
mbedtls_net_recv                                  esp-idf/esp-tls/libesp-tls.a(esp_tls_mbedtls.c.obj)
mbedtls_net_send                                 esp-idf/esp-tls/libesp-tls.a(esp_tls_mbedtls.c.obj)
```

**To reproduce**
Steps to reproduce the behavior:

1. Clone the example code:
     -  git clone https://github.com/SimionMogildea/AmazonFreertosOta.git --recurse-submodules
2. Configure project: `cmake -S . -B build  -DBOARD=esp32_devkitc -DCMAKE_TOOLCHAIN_FILE='amazon-freertos/tools/cmake/toolchains/xtensa-esp32.cmake' -G Ninja`
3. Build project: `cmake --build build`
4. Flash: `python ./amazon-freertos/vendors/espressif/esp-idf/tools/idf.py erase_flash flash -p COMxx -b 115200 monitor`

**Code to reproduce the bug**
I made an example project based on the `amazon-freertos\vendors\espressif\esp-idf\examples\system\ota\simple_ota_example`
- https://github.com/SimionMogildea/AmazonFreertosOta

**Additional context**
Same problems occur even when I include `mbedtls` component directly in my app.
See example: https://gist.github.com/SimionMogildea/0a0e05f6294a4c1a1200433fd6c802e3

Thank you!

ravibhagavandas, 2021-03-19:
Hi @SimionMogildea

I think to compile those ESP32 libraries ( esp_http_client, esp_https_server)  requires mbedtls dependency through ESP build system which is not available through the freertos for for ESP IDF used.

Freertos repo has its own implementations for OTA (https://github.com/aws/amazon-freertos/tree/master/libraries/freertos_plus/aws/ota) and HTTP Client (https://github.com/FreeRTOS/coreHTTP) which uses 3rdparty fork of mbedtls (https://github.com/aws/amazon-freertos/tree/master/libraries/3rdparty).
Could this satisfy your use case?

SimionMogildea, 2021-03-22:
Hi @ravibhagavandas,
Thank you for the quick reply!
Unfortunately the OTA implementation you mentioned, which is based on AWS Job system, doesn't suit our needs. That's why we chose to use the ESP_IDF's implementation of OTA.

For instance, we need to migrate to the lastest version of this repo due to ESP_IDF's new secure features in v4.2, but the OTA feature we require seems to be faulty and does't work anymore.
Could this be a bug in any internal CmakeLists of `amazon-freertos`?

ravibhagavandas, 2021-03-24:
Mbedtls dependency for the ESP OTA component get linked from within ESP cmake build system and not directly from amazon-freertos repository. I would suggest asking this in Espression section of the freertos forum [here](https://forums.freertos.org/c/partners-sponsors/espressif/14).

shubhamkulkarni97, 2021-04-22:
Hi @SimionMogildea,

Can you try using changes in PR https://github.com/aws/amazon-freertos/pull/3115?

SimionMogildea, 2021-04-23:
Thank you @shubhamkulkarni97, your PR soved our problem!

closed 2021-04-23 by SimionMogildea