aws/amazon-freertos #3058: ESP-IDF and Toolchain version for AWS FreeRTOS ESP32 V202002.00

Yoimer, 2021-03-21:
Hello there. I need to test the `AWS FreeRTOS ESP32 V202002.00` zip file automatically generated on AWS Console. However, I don't find which exact version of ESP-IDF and Toolchain is required to do so. I created a Github repo https://github.com/Yoimer/AFR_ESP32_FreeRTOS_V202002.00 which has all the files from the zip file. Even I got a version file, https://github.com/Yoimer/AFR_ESP32_FreeRTOS_V202002.00/blob/master/vendors/espressif/esp-idf/version.txt which says` v3.3-163-g601a03e`, however I don't find that tag release from espressif/esp-idf.
Could you please, help me out telling me the right tools I need to tests the demos? I plan to work on this using Windows.
Thanks in advance.

ravibhagavandas, 2021-03-22:
Hi,
The porting layer of ESP32 in amazon FreeRTOS uses a base repo [esp-afr-sdk](https://github.com/espressif/esp-afr-sdk/tree/release/v3.3) instead of vanilla espressif/esp-idf. Amazaon FreeRTOS v202002.00 supports ESP IDF 3.3 as mentioned in the version file so you should use toolchain associated with that IDF version. Getting started guide provides details on how to build using toolchain for esp-idf v3.3:
https://docs.aws.amazon.com/freertos/latest/userguide/getting_started_espressif.html

Yoimer, 2021-03-22:
@ravibhagavandas Thanks for replying. I will check and report over here my results. Have a great day.

Yoimer, 2021-03-23:
@ravibhagavandas Thanks for your support. It is working all good.

closed 2021-03-23 by Yoimer