aws/amazon-freertos #3059: Publishing to MQTT not calling callback in IDF V4

hackaprende, 2021-03-22:
**Describe the bug**
When you publish to a topic by MQTT, the prvEventCallback is not called in IDF V4, it works for subscribing but not for publishing

**System information**
- Which hardware board or part numbers? ESp32 V4
- IDE used Visual Studio Code
- Operating System [Windows|Linux|MacOS] MacOS
- Version of FreeRTOS (run ``git describe --tags`` to find it)
- Project [Custom Application|MQTT|Shadow|Greengrass|OTA|Device Defender|TCP Echo|Sub pub Demo|Simple TCP Server] Custom Application
- If your project is a Custom Application, please add the relevant code snippet in the section `Code to reproduce the bug`. 
I am using practically the code that comes with the MQTT mutual auth demo:
```
static BaseType_t publishMQTTMessage(MQTTContext_t * pxMQTTContext, const char * topic, char messagePayload[], 
    uint32_t timeoutMs) {

    MQTTStatus_t xResult;
    MQTTPublishInfo_t xMQTTPublishInfo;
    BaseType_t xStatus = pdPASS;

    /* Some fields are not used by this demo so start with everything at 0. */
    ( void ) memset( ( void * ) &xMQTTPublishInfo, 0x00, sizeof( xMQTTPublishInfo ) );

    xMQTTPublishInfo.qos = MQTTQoS1;
    xMQTTPublishInfo.retain = false;
    xMQTTPublishInfo.pTopicName = topic;
    xMQTTPublishInfo.topicNameLength = ( uint16_t ) strlen( topic );
    xMQTTPublishInfo.pPayload = messagePayload;
    xMQTTPublishInfo.payloadLength = strlen(xMQTTPublishInfo.pPayload);

    /* Get a unique packet id. */
    usPublishPacketIdentifier = MQTT_GetPacketId( pxMQTTContext );

    ESP_LOGI(TAG, "Publishing topic %s\nPayload: %s", topic, messagePayload);
    /* Send PUBLISH packet. Packet ID is not used for a QoS1 publish. */
    xResult = MQTT_Publish( pxMQTTContext, &xMQTTPublishInfo, usPublishPacketIdentifier );

    if( xResult != MQTTSuccess )
    {
        xStatus = pdFAIL;
        ESP_LOGI(TAG, "Failed to send PUBLISH message to broker: Topic=%s, Error=%s",
                    topic,
                    MQTT_Status_strerror( xResult ));
    }

    return xStatus;
}
```

**Expected behavior**
When publishing something, the prvEventCallback is not called.

**Screenshots or console output**
If appropriate, please cut-and-paste the console output to explain the issue. If possible, include the call stack. 

This is when subscribing:
[image]

But when publishing the callback is not called:
[image]

**To reproduce**
Steps to reproduce the behavior:

1. Setup MQTT with callback
2. Publish something to MQTT

**Code to reproduce the bug**
The code should be wrapped in the cpp tag in order to be displayed clearly. For example:

```cpp
printf("Hello World")
```

If the code is longer than 30 lines, [GIST](https://gist.github.com) is preferred.

**Additional context**
It was working with IDF V3.3.

Thank you!

ravibhagavandas, 2021-03-23:
Hi,

It is highly unlikely that IDF version change could have caused the issue you are observing. 

To help debug further could you send the full logs after the MQTT_Publish() ?  Are you receiving any events after MQTT_Publish()?
AWS docs suggest an empty message has to be sent to $aws/things/thingName/shadow/get. Is this API publishing an empty message?

Could you also paste the code snippet for `prvEventCallback` ?

hackaprende, 2021-03-23:
I am not receiving events after publishing, I am using an empty payload message. The weird thing is that when I subscribe I am receiving an event, this is my prvEventCallback, I have a log at the beggining to check when I am getting an event. When I publish it does not log anything.

```
static void prvEventCallback( MQTTContext_t * pxMQTTContext,
                              MQTTPacketInfo_t * pxPacketInfo,
                              MQTTDeserializedInfo_t * pxDeserializedInfo )
{
    ESP_LOGI(TAG, "prvEventCallback called");
    ShadowMessageType_t messageType = ShadowMessageTypeMaxNum;
    const char * pcThingName = NULL;
    uint16_t usThingNameLength = 0U;
    uint16_t usPacketIdentifier;

    ( void ) pxMQTTContext;

    assert( pxDeserializedInfo != NULL );
    assert( pxMQTTContext != NULL );
    assert( pxPacketInfo != NULL );

    usPacketIdentifier = pxDeserializedInfo->packetIdentifier;

    ESP_LOGI(TAG, "prvEventCallback type: %d", pxPacketInfo->type & 0xF0U);

    /* Handle incoming publish. The lower 4 bits of the publish packet
     * type is used for the dup, QoS, and retain flags. Hence masking
     * out the lower bits to check if the packet is publish. */
    if ((pxPacketInfo->type & 0xF0U) == MQTT_PACKET_TYPE_PUBLISH) {
        assert(pxDeserializedInfo->pPublishInfo != NULL);
        LogInfo(("pPublishInfo->pTopicName:%s.", pxDeserializedInfo->pPublishInfo->pTopicName));

        /* Let the Device Shadow library tell us whether this is a device shadow message. */
        if (SHADOW_SUCCESS == Shadow_MatchTopic(pxDeserializedInfo->pPublishInfo->pTopicName,
                                                pxDeserializedInfo->pPublishInfo->topicNameLength,
                                                &messageType,
                                                &pcThingName,
                                                &usThingNameLength)) {
            /* Upon successful return, the messageType has been filled in. */
            if (messageType == ShadowMessageTypeGetAccepted) {
                handleShadow(pxDeserializedInfo->pPublishInfo);
            } else if (messageType == ShadowMessageTypeGetRejected) {
                
            } else if (messageType == ShadowMessageTypeUpdateDelta) {
                /* Handler function to process payload. */
                prvUpdateDeltaHandler( pxDeserializedInfo->pPublishInfo );
            } else if (messageType == ShadowMessageTypeUpdateAccepted) {
                /* Handler function to process payload. */
                prvUpdateAcceptedHandler( pxDeserializedInfo->pPublishInfo );
            } else if (messageType == ShadowMessageTypeUpdateDocuments ) {
                LogInfo(("/update/documents json payload:%s.", (const char *) pxDeserializedInfo->pPublishInfo->pPayload ) );
            } else if (messageType == ShadowMessageTypeUpdateRejected ) {
                LogInfo(("/update/rejected json payload:%s.", (const char *) pxDeserializedInfo->pPublishInfo->pPayload ) );
            } else if (messageType == ShadowMessageTypeDeleteAccepted) {
                LogInfo(("Received an MQTT incoming publish on /delete/accepted topic."));
                xShadowDeleted = pdTRUE;
                xDeleteResponseReceived = pdTRUE;
            } else {
                LogInfo(("Other message type:%d !!", messageType));
            }
        } else {
            LogError( ( "Shadow_MatchTopic parse failed:%s !!", ( const char * ) pxDeserializedInfo->pPublishInfo->pTopicName ) );
        }
    } else {
        prvMQTTProcessResponse( pxPacketInfo, pxDeserializedInfo->packetIdentifier );
    }
}
```

ravibhagavandas, 2021-03-23:
I think the issue is because an `MQTT_ProcessLoop` is not called after a publish packet is sent out. In order to process any incoming packet such as PUBACK, coreMQTT library requires application to call `MQTT_ProcessLoop` at least once after the Qos1 publish is sent successfully. Please take a look at the sample demo function [PublishToTopic](https://github.com/aws/amazon-freertos/blob/master/demos/common/mqtt_demo_helpers/mqtt_demo_helpers.c#L1098) for further details.

hackaprende, 2021-03-23:
@ravibhagavandas Wow thank you very much!, when I updated my code to the latest I removed the process loop, this fixed my problem. Thanks again, I hope all developers were as helpful as you.

ravibhagavandas, 2021-03-23:
Glad to know it works. I am closing the issue. Please open an issue if you have further questions.

closed 2021-03-23 by ravibhagavandas