aws/amazon-freertos #3060: Errors Occurred While flashing demo projects

aadu999, 2021-03-23:
I was trying to build a demo project using AWS FreeRTOS, after build while trying to flash the device using cmake i'm getting the following error 

``` shell
cmake --build buildTRL4v3 --target flash
Scanning dependencies of target blank_ota_data
[  0%] Generating ../../ota_data_initial.bin
[  0%] Built target blank_ota_data
[  1%] Built target partition_table
[  1%] Built target idf_component_ulp
[  2%] Built target afr_kernel
[  2%] Building C object CMakeFiles/afr_wifi.dir/vendors/espressif/boards/esp32/ports/wifi/iot_wifi.c.obj
/home/adarsh/Documents/Work/cCabin/myExp/FreeRTOS/vendors/espressif/boards/esp32/ports/wifi/iot_wifi.c: In function 'WIFI_NetworkAdd':
/home/adarsh/Documents/Work/cCabin/myExp/FreeRTOS/vendors/espressif/boards/esp32/ports/wifi/iot_wifi.c:953:29: warning: initialization of 'nvs_handle' {aka 'unsigned int'} from 'void *' makes integer from pointer without a cast [-Wint-conversion]
     nvs_handle xNvsHandle = NULL;
                             ^~~~
/home/adarsh/Documents/Work/cCabin/myExp/FreeRTOS/vendors/espressif/boards/esp32/ports/wifi/iot_wifi.c: In function 'WIFI_NetworkDelete':
/home/adarsh/Documents/Work/cCabin/myExp/FreeRTOS/vendors/espressif/boards/esp32/ports/wifi/iot_wifi.c:1075:26: warning: initialization of 'nvs_handle' {aka 'unsigned int'} from 'void *' makes integer from pointer without a cast [-Wint-conversion]
  nvs_handle xNvsHandle = NULL;
                          ^~~~
/home/adarsh/Documents/Work/cCabin/myExp/FreeRTOS/vendors/espressif/boards/esp32/ports/wifi/iot_wifi.c: In function 'prvStoreWIFINetwork':
/home/adarsh/Documents/Work/cCabin/myExp/FreeRTOS/vendors/espressif/boards/esp32/ports/wifi/iot_wifi.c:883:45: error: 'snprintf' output may be truncated before the last format character [-Werror=format-truncation=]
   snprintf(cWifiKey, MAX_WIFI_KEY_WIDTH, "%d", usIndex );
                                             ^
/home/adarsh/Documents/Work/cCabin/myExp/FreeRTOS/vendors/espressif/boards/esp32/ports/wifi/iot_wifi.c:883:3: note: 'snprintf' output between 2 and 6 bytes into a destination of size 5
   snprintf(cWifiKey, MAX_WIFI_KEY_WIDTH, "%d", usIndex );
   ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
/home/adarsh/Documents/Work/cCabin/myExp/FreeRTOS/vendors/espressif/boards/esp32/ports/wifi/iot_wifi.c: In function 'prvGetWIFINetwork':
/home/adarsh/Documents/Work/cCabin/myExp/FreeRTOS/vendors/espressif/boards/esp32/ports/wifi/iot_wifi.c:901:45: error: 'snprintf' output may be truncated before the last format character [-Werror=format-truncation=]
   snprintf(cWifiKey, MAX_WIFI_KEY_WIDTH, "%d", usIndex);
                                             ^
/home/adarsh/Documents/Work/cCabin/myExp/FreeRTOS/vendors/espressif/boards/esp32/ports/wifi/iot_wifi.c:901:3: note: 'snprintf' output between 2 and 6 bytes into a destination of size 5
   snprintf(cWifiKey, MAX_WIFI_KEY_WIDTH, "%d", usIndex);
   ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
/home/adarsh/Documents/Work/cCabin/myExp/FreeRTOS/vendors/espressif/boards/esp32/ports/wifi/iot_wifi.c: In function 'WIFI_NetworkDelete':
/home/adarsh/Documents/Work/cCabin/myExp/FreeRTOS/vendors/espressif/boards/esp32/ports/wifi/iot_wifi.c:1094:46: error: 'snprintf' output may be truncated before the last format character [-Werror=format-truncation=]
    snprintf(cWifiKey, MAX_WIFI_KEY_WIDTH, "%d", xRegistry.usStorageIdx[ usIndex ]);
                                              ^
/home/adarsh/Documents/Work/cCabin/myExp/FreeRTOS/vendors/espressif/boards/esp32/ports/wifi/iot_wifi.c:1094:4: note: 'snprintf' output between 2 and 6 bytes into a destination of size 5
    snprintf(cWifiKey, MAX_WIFI_KEY_WIDTH, "%d", xRegistry.usStorageIdx[ usIndex ]);
    ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
cc1: some warnings being treated as errors
make[3]: *** [CMakeFiles/afr_wifi.dir/build.make:63: CMakeFiles/afr_wifi.dir/vendors/espressif/boards/esp32/ports/wifi/iot_wifi.c.obj] Error 1
make[2]: *** [CMakeFiles/Makefile2:2085: CMakeFiles/afr_wifi.dir/all] Error 2
make[1]: *** [CMakeFiles/Makefile2:2702: CMakeFiles/flash.dir/rule] Error 2
make: *** [Makefile:340: flash] Error 2
```

Thanks for your Help in Advance.

ravibhagavandas, 2021-03-24:
Hi @aadu999 

Could you provide more information on which toolchain are you using to build your code? Which version of this repo are you using?

AniruddhaKanhere, 2021-04-14:
Hello @aadu999, you can also try to increase the value of `MAX_WIFI_KEY_WIDTH` macro. Keep in mind that this must be less than `NVS_KEY_NAME_MAX_SIZE-1` (which is generally 16-1 = 15).
A safe bet would be to make `MAX_WIFI_KEY_WIDTH == 10`.

I am closing this issue since it has been quite a while since the last activity. Feel free to open this issue or a new one if you have any more questions or issues.

Thank you.

closed 2021-04-14 by AniruddhaKanhere