aws/amazon-freertos #3061: choosing between the "ESP_IDF + esp-aws-iot" and "amazon FreeRTOS"2

AhmedElsayedHamouda, 2021-03-24:
Hello, I am about to start building a system based on PUB/SUB / SHADOW / JOBS / FLEET PROVISIONING . I am having a difficulties to decide the way I am going to uses. I have two ways:

1- using ESP_IDF and use the esp-aws-iot SDK to use.
2- using aws FreeRTOS.

The question is : what makes me change from ESP_IDF to amazon FreeRTOS package? what is the difference between the FreeRTOS used in both IDF and aws FreeRTOS. knowing that I am going to use OTA. 

Thanks

muneebahmed10, 2021-03-25:
Hi @AhmedElsayedHamouda,

ESP-IDF and the esp-aws-iot repository are maintained by Espressif, while this one is updated by AWS.

Both use the FreeRTOS kernel, though the versions they use might differ. This repository submodules the latest V10.4.3 kernel release.

Looking at [esp-aws-iot](https://github.com/espressif/esp-aws-iot), it looks to be using a [deprecated version](https://github.com/espressif/aws-iot-device-sdk-embedded-C/tree/61f25f34712b1513bf1cb94771620e9b2b001970) of the [AWS IoT Embedded C SDK](https://github.com/aws/aws-iot-device-sdk-embedded-C). Since you want to use AWS IoT services, I would recommend using the [latest LTS releases](https://freertos.org/lts-libraries.html) of the libraries, which you can find in this repository and the [OTA](https://github.com/aws/ota-for-aws-iot-embedded-sdk/tree/main) repository.

Let me know if you have further questions.

ncohen17, 2021-03-25:
Does AWS FreeRTOS support multi-core processing on the ESP32 yet ?

AhmedElsayedHamouda, 2021-03-26:
Thanks for the replay @muneebahmed10 . I was saying the OTA as example, like is there any advantages that the [amazon FreeRTOS](https://github.com/aws/amazon-freertos) has and esp-aws-iot doesn't ?

muneebahmed10, 2021-03-26:
Hi @AhmedElsayedHamouda 

As I mentioned in the post above, the AWS IoT Embedded C SDK libraries submoduled by [esp-aws-iot](https://github.com/espressif/esp-aws-iot) are deprecated and over two years old. This includes AWS IoT libraries such as Device Shadow as well as the MQTT stack. Save for OTA, [amazon-freertos](https://github.com/aws/amazon-freertos) submodules the latest versions of these libraries and you can find them [here](https://github.com/aws/amazon-freertos/tree/master/libraries). Of course, you can also use [ESP-IDF](https://github.com/espressif/esp-idf) and manually import the latest AWS IoT libraries if you wish to do so.

@ncohen17
No, the FreeRTOS distribution currently does not include support for SMP. To use SMP on ESP32, you can use [ESP-IDF](https://github.com/espressif/esp-idf) and follow Espressif's [guide](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-guides/freertos-smp.html).

AhmedElsayedHamouda, 2021-03-28:
Hi @muneebahmed10 
Thanks a lot, I understand now the deprecation problem. Could I just replace the SDK with the new one, or this will cause different  building problems ?. I just prefer IDF because it is more organized than amazon freeRTOS. 
Thanks

AhmedElsayedHamouda, 2021-03-28:
That helped a lot, thanks @muneebahmed10 .

ncohen17, 2021-03-29:
Hi @muneebahmed10 ,

Is there any plan in the roadmap to make the SMP available on the FreeRTOS on the ESP32 ?

muneebahmed10, 2021-07-20:
@ncohen17,

You can read about recently added support for SMP in the FreeRTOS Kernel in our [blog post](https://www.freertos.org/2021/06/introducing-the-freertos-symmetric-multiprocessing-smp-github-branch.html).

Since the original question has been answered, I will close this issue. Please open a new issue for any further questions.

closed 2021-07-20 by muneebahmed10