aws/amazon-freertos #3063: ESP32 WIFI_ConnectAP hangs waiting for connect event

jonathanfoster, 2021-03-25:
**Describe the bug**

Calling WIFI_ConnectAP on the ESP32 hangs. The function call never returns because the station connected event never fires.

**System information**
- Hardware board: Espressif ESP32 DevKitC
- IDE: CLion
- Operating System: macOS
- Version of FreeRTOS: 202012.00
- Project: Custom Application

**Expected behavior**

After calling `WIFI_ConnectAP`, the Wi-Fi module should connect to the AP, emit a `SYSTEM_EVENT_STA_CONNECTED` event, and return `eWiFiSuccess`.

**Code to reproduce the bug**

```c
// Init Wi-Fi
nvs_flash_init();
tcpip_adapter_init();
WIFI_On();

// Connect to AP
WIFINetworkParams_t network_params;
// Set network params...
WIFI_ConnectAP(&network_params);
```

**Additional context**

The issue appears to be caused by not calling `esp_wifi_start` if the current mode is set to `WIFI_MODE_STA`. If the current mode is not set to `WIFI_MODE_STA`, then the mode is set and `esp_wifi_start` is called. The can be resolved by calling `esp_wifi_start` before calling `WIFI_ConnectAP`.

```c
// Connect to AP
WIFINetworkParams_t network_params;
// Set network params...
esp_wifi_start();
WIFI_ConnectAP(&network_params);
```

Moving the `esp_wifi_start` call outside of the if block may resolve this issue. https://github.com/aws/amazon-freertos/blob/master/vendors/espressif/boards/ports/wifi/iot_wifi.c#L496

shubhamkulkarni97, 2021-03-26:
@jonathanfoster,

Do you observe this issue in specific conditions? Can you share code to reproduce thus issue?

jonathanfoster, 2021-03-26:
@shubhamkulkarni97, I can reliably reproduce this issue with this code:

```c
// Init Wi-Fi
nvs_flash_init();
tcpip_adapter_init();
WIFI_On();
// Calling esp_wifi_start here will resolve the issue

// Connect to AP
WIFINetworkParams_t network_params;
// Set network params...
WIFI_ConnectAP(&network_params);
```

The call hangs waiting for the `SYSTEM_EVENT_STA_START` here, https://github.com/aws/amazon-freertos/blob/master/vendors/espressif/boards/ports/wifi/iot_wifi.c#L504. If I call `esp_wifi_start` before , the connection is made successfully.

It looks like `WIFI_StartAP` is responsible for calling `esp_wifi_start`. Should I be calling this before `WIFI_ConnectAP` instead of `esp_wifi_start`? I'm new to FreeRTOS and ESP32 so the problem very well could be that I'm holding it wrong. I'm running the latest release branch, which I believe is using ESP IDF v3.3.

jonathanfoster, 2021-03-26:
The issue also occurs if you use `AwsIotNetworkManager` without BLE Wi-Fi provisioning enabled. In this case, `WIFI_On` and `WIFI_ConnectAP` are called from `AwsIotNetworkManager_EnableNetwork` and there's no way to call `esp_wifi_start`.

```c
nvs_flash_init();
tcpip_adapter_init();
AwsIotNetworkManager_Init();
AwsIotNetworkManager_EnableNetwork(configSUPPORTED_NETWORKS);
```

shubhamkulkarni97, 2021-04-06:
Hi @jonathanfoster,

Can you try the patch attached below and check if it fixes your issue?

[0001-iot_wifi.c-Fix-wifi-start-stop-for-Espressif-boards.patch.zip](https://github.com/aws/amazon-freertos/files/6262964/0001-iot_wifi.c-Fix-wifi-start-stop-for-Espressif-boards.patch.zip)

jonathanfoster, 2021-04-10:
@ravibhagavandas Here's my network params init.

```c
    WIFINetworkParams_t network_params;

    const char *ssid = WIFI_SSID;
    size_t ssid_length = strlen(ssid);
    memcpy(network_params.ucSSID, ssid, ssid_length);
    network_params.ucSSIDLength = ssid_length;

    const char *password = WIFI_PASSWORD;
    size_t password_length = strlen(password);
    memcpy(network_params.xPassword.xWPA.cPassphrase, password, password_length);
    network_params.xPassword.xWPA.ucLength = password_length;
    network_params.xSecurity = eWiFiSecurityWPA2;
```

jonathanfoster, 2021-04-10:
@shubhamkulkarni97 The commits were rejected. Initially the path was incorrect, it was missing the `esp32` in `vendors/espressif/boards/esp32/ports/wifi/iot_wifi.c`. I updated the path and then I got the rejections.

I'm using the `202012.00` release. Here's `iot_wifi.c` from that release.  https://github.com/aws/amazon-freertos/blob/202012.00/vendors/espressif/boards/esp32/ports/wifi/iot_wifi.c

[iot_wifi.c.rej.zip](https://github.com/aws/amazon-freertos/files/6289165/iot_wifi.c.rej.zip)

shubhamkulkarni97, 2021-04-12:
@jonathanfoster Sorry, I forgot to mention that this patch is for master branch. Could you please try applying it on top of master branch?

jonathanfoster, 2021-05-08:
@shubhamkulkarni97 I updated my FreeRTOS submodule to commit 6b75bc2916658fc7ca097a591821a476a65517f2 on master and I still got rejections. It might be easier if I just create small repo to reproduce the issue and you can test out your changes there. I'll do that and post a link here.

jonathanfoster, 2021-05-08:
@shubhamkulkarni97 I created a [demo](https://github.com/jonathanfoster/freertos-esp32-wifi-connect-demo) using master and I wasn't able to reproduce the issue. [Here's where I was calling `esp_wifi_start`.](https://github.com/jonathanfoster/freertos-esp32-wifi-connect-demo/blob/master/src/wifi/demo_wifi.c#L109)

Looks good now. Thanks for your help.

closed 2021-05-08 by jonathanfoster