aws/amazon-freertos #3064: ESP32 CMake IDF component libraries output is empty

jonathanfoster, 2021-03-25:
**Describe the bug**

The ESP32 IDF component libraries list is empty in the CMake output. Without this list, it's difficult to determine the library name to include when adding target link libraries.

**System information**

- Hardware board: Espressif ESP32 DevKitC
- IDE: CLion
- Operating System: macOS
- Version of FreeRTOS: 202012.00
- Version of IDF: 3.3
- Project: Custom Application 

**Expected behavior**

When running CMake, the output should include a list of available IDF components.

```shell
-- Component libraries: idf_component_soc;idf_component_log;idf_component_heap;idf_component_freertos;...
```

Currently, the IDF components libraries list is empty.

```shell
-- Component libraries:
```

**Code to reproduce the bug**

```shell
cmake -S . -B "$BUILD_DIR" -DCMAKE_TOOLCHAIN_FILE=freertos/tools/cmake/toolchains/xtensa-esp32.cmake -GNinja
```

**Additional context**

The issue appears to be caused by referencing the incorrect library list variable. https://github.com/espressif/esp-afr-sdk/blob/1d2d93acd1dfd00a63309cdeb109a467efbcb327/tools/cmake/idf_functions.cmake#L302

```cmake
// message(STATUS "Component libraries: ${IDF_COMPONENT_LIBRARIES}")
message(STATUS "Component libraries: ${libraries}")
```

muneebahmed10, 2021-03-25:
Hi @jonathanfoster 

It looks like the CMake issues you are having are due to files in the [esp-afr-sdk](https://github.com/espressif/esp-afr-sdk) repository from Espressif, not from this repository.

@shubhamkulkarni97, are you able to take a look at this?

shubhamkulkarni97, 2021-03-26:
@jonathanfoster Thanks for reporting this issue. Will update after discussing internally.

shubhamkulkarni97, 2021-03-26:
@jonathanfoster,

Can you try building demo on AFR master branch?
master branch supports ESP-IDF v4.2

jonathanfoster, 2021-03-26:
@shubhamkulkarni97, looks good on master. Any timeline when this will be released? I'm also interested in the SoftAP Wi-Fi provisioning in ESP-IDF v4.2.

closed 2021-03-26 by jonathanfoster